Koenkk/zigbee2mqtt #7362: Multiple Zigbee coordinators (USB/EZSP) support

Masterz69, 2021-05-06:
## Describe the solution you'd like
Looking for possibility to use multiple Zigbee sticks and multiple EZSP devices at a same time on same Z2M instance.

My use case: locally connected USB Zigbee stick + remotely (same LAN) located Xiaomi Gateway 3 (one or more) with AlexxIT integration.

Hedda, 2021-05-06:
You will never be able to use two Zigbee coordinators in the same Zigbee network as only one Zigbee coordinator per network is allowed, but a feature could be implemented to support two more Zigbee coordinators which each have their on Zigbee network that know nothing of each other at the Zigbee level. The implementation would then be done only at the application level.

FYI, Jeedom new Zigbee plugin does that https://blog.jeedom.com/5183-tout-ce-quil-faut-savoir-sur-le-plugin-officiel-zigbee/

MattWestb, 2021-05-07:
From Zigbee point its possible but not with on trustcenter security.
If using distributed security and more "zigbe control bridges" you can have more "gate ways" in the system but you need writing all code for manage the network your self then no  open system have implanting it.

Known system using the same Principe is Philips HUE and IKEA GW and so on.

Masterz69, 2021-05-07:
As @Hedda wrote:
> feature could be implemented to support more Zigbee coordinators which each have their on Zigbee network.

@MattWestb 
As IP stack can work with multiple NIC, in "same manner" see no issues to work with multiple Zigbee coordinators (each in own Zigbee network).

MattWestb, 2021-05-07:
Distributed security is not having any coordinator = no trustcenter.

As sad coordinator it can only being one in the zigbee world = only one trustcenter.

Doing multiple zigbee network with coordinators is no problem in the zigbee level as long they is using different PAN-ID and extended pan-ID and also different network key can being good.

Masterz69, 2021-06-09:
IMHO multiple Zigbee networks support can be useful for some of the Z2M users.

closed 2021-07-18 by github-actions[bot]