Koenkk/zigbee2mqtt #7365: ROB_200-006-0 Zigbee LED dimmer

mrblwd-058, 2021-05-06:
## Information about the device + link
..
ROB_200-006-0 LED Driver

https://www.robbshop.nl/led-driver-zigbee-12-36v-robb-smarrt

## data/database.db entry of the device
..
Last seen
N/A
Device type
Router
Zigbee model
ROB_200-006-0
Support status
Unsupported

IEEE address
0x60a423fffec2a69c
Network address
0x158D
Firmware build date
NULL
Firmware version
2.5.3_r51
Power source
Interview completed
Yes

I read the article about to add new devices but for me that is 2 complex. Can you please add support for this device? I already bought it but didn't look in the supported devices list before, my bad. 

Thanks a lot!

I'm on the latest version of zigbee2mqtt and latest z-stack firmware.

Koenkk, 2021-05-06:
Added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

mrblwd-058, 2021-05-07:
Great, thanks for the quick response!