Koenkk/zigbee2mqtt #7371: Moes BHT-002-GCLZB is not showing me the running_state.

Marc11170, 2021-05-06:
Hello,

## What happened
I have a Moes BHT-002-GCLZB thermostat and Zigbee2mqtt is not showing me the running_state.
in the "Exposes" tab, The current running state is N/A

## What did you expect to happen
running_state: The current running state. Possible values are: idle, heat, cool.

## How to reproduce it (minimal and precise)
Send a message to zigbee2mqtt/FRIENDLY_NAME/get with payload {"running_state": ""}.

## Debug info
Zigbee2MQTT version: 1.18.3
Adapter hardware:  Conbee II
Adapter firmware version:

insipiens, 2021-05-14:
Odd that, right? 

Would a thermostat have a running state - maybe. 

The generic card in home assistant doesn't have that attribute - isn't it really for an HVAC configuration?

The device as configured does show a running state, "Heat":"ON"/"OFF"

 `payload '{"child_lock":null,"current_heating_setpoint":19,"deadzone_temperature":0,"heat":"OFF","last_seen":"2021-05-14T11:44:07+01:00","linkquality":108,"local_temperature":17.5,"local_temperature_calibration":0,"max_temperature_limit":30,"preset":"hold","preset_mode":"hold","sensor":"IN","system_mode":"off"}'
`

You could probably rename that or duplicate to correct this.

Marc11170, 2021-05-14:
Hello,
As the documentation describes:
[https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/BHT-002-GCLZB.md](url)

system_mode: allows you to see if the thermostat is on or off (Thermostat on or off)
system_mode: Mode of this device. To control publish a message to topic zigbee2mqtt/FRIENDLY_NAME/set with payload {"system_mode": VALUE} where VALUE is one of: off, heat. To read send a message to zigbee2mqtt/FRIENDLY_NAME/get with payload {"system_mode": ""}

running_state: allows you to see the state of the relay (activation of the boiler or not)
running_state: The current running state. Possible values are: idle, heat, cool. To read send a message to zigbee2mqtt/FRIENDLY_NAME/get with payload {"running_state": ""}.

The system_mode is displayed correctly, it is the running_sate which is not displayed.

insipiens, 2021-05-14:
ah, I was sort of greeing with you - the running state was a later addition to the original (which I created). 

I just tried it on my installation but without much success.

Lets see if we can get hold of the architect to see if its an easy fix.

hi @drzony I wonder if you might be able to help us out on this one. Some pointers would be ideal.

Cheers

drzony, 2021-05-14:
Missing running_state is probably a side effect of at least 2 different valves mixed up in this single model. Some of them report running state, some of them don't. Please see #5004

insipiens, 2021-05-17:
looking at it anew it seems this is related to the newer April version of Home Assistant reporting warnings such as:

`WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'off' when rendering '{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state] }}'`

This isn't a device specific issue per se.

see this issue: https://github.com/Koenkk/zigbee2mqtt/issues/6987

Marc11170, 2021-05-18:
Personally, I am under Jeedom 4.1.22

richsimpson, 2021-06-01:
I am using the same item but the GBLZB (16A switch current Electric Underfloor Heating) version. It is in through zigbee2mqtt as a GCLZB though.

For me I get thousands of these warnings:
Logger: homeassistant.helpers.template
Source: helpers/template.py:1367
First occurred: 9:48:58 (82 occurrences)
Last logged: 10:36:43

Template variable warning: 'mappingproxy object' has no attribute 'heat' when rendering '{{ states.climate.underfloor_heating.attributes.heat }}'
Template variable warning: dict object has no element Undefined when rendering '{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state] }}'

For me in HASS I see the "HVAC_MODES" as HEAT or OFF, the "HEAT" as ON or OFF. HVAC being unit on or not, the HEAT being actually calling for heat from the floor pads.
On the Z2M dash running state is a N/A
Seems the same issue. Function seems to be fine but the warning log is concerning to have such a high number of the warnings. Ok to ignore, will a fix follow?

insipiens, 2021-06-01:
I applied the fix suggested in #6987 manually which solved the running state for me (as updating z2m version hadn’t fixed it)

insipiens, 2021-06-01:
I should add I still get these warnings for some other devices - as it doesn’t seem to affect behaviour I just ignore them in Home Assistant

richsimpson, 2021-06-22:
I don't understand where I have to add that or change that can you point me in the right direction please?

Marc11170, 2021-06-22:
the running_sate is not displayed. (Heating relay engaged or not) I am under Jeedom V4.1.23 and zigbee2mqtt

richsimpson, 2021-07-18:
I just removed the warnings in HASS but would love a way to remove the issue in a better way.

closed 2021-08-25 by github-actions[bot]

KameDomotics, 2021-09-09:
Hi. I have the same problem. How can I solve it? Thanks