Koenkk/zigbee2mqtt #7380: Zigbee2mqtt on Proxmox (LXC)

Ramshield, 2021-05-07:
## What happened

Hi,

Last week I messed up my server and have to rebuild all my LXC containers.
So I decided to try this on Proxmox in stead of plain Debian. 

After some trouble getting the passthrough to LXC to work I got Zigbee2mqtt running.
However when trying to (re-)pair a device, nothing happens.
Messed around with it for couple more days, and I tried it on a Virtualbox VM, on my Windows PC, which works instantly. 
So I tried running Zigbee2mqtt on the Proxmox server itself.
Nothing happens again... So nothing is wrong with the USB, or Zigbee2mqtt, because I do everything exactly the same according to the guide.
The issue seems Proxmox. 

Has anyone have Zigbee2mqtt working on Proxmox?

Thank you! 

Logs: https://pastebin.com/vzjqcgDy

## Debug info
Zigbee2MQTT version: 1.18.3 (commit #f2e39af1)
Adapter hardware: CC2652R
Adapter firmware version: CC2652R_coordinator_20210120

juslex, 2021-05-09:
Try this:

https://www.martinellis.me/posts/2020/03/zig-a-zig-ah/

AlexKay88, 2021-05-28:
> Has anyone have Zigbee2mqtt working on Proxmox or in a Proxmox LXC container?

Runs fine as a unprivileged Proxmox LXC container with Conbee II Zigbee stick connected to Proxmox host here.
1) Debian 10 inside unprivileged LXC
2) Proxmox host: `/etc/udev/rules.d/99-ttyACM-rw.rules` Purpose: make persistant device name `/dev/ttyACMZigbee` and make sure its writable (mode=0666).
```
# ID 1cf1:0030 Dresden Elektronik ConBee II
# make Z-Wave stick RW for others, so that container can access it
SUBSYSTEMS=="usb", ATTRS{idVendor}=="1cf1", ATTRS{idProduct}=="0030", SYMLINK+="ttyACMZigbee", MODE="0666"
```
3) Proxmox host: add to LXC config `/etc/pve/nodes/pve1/lxc/111.conf`
```
lxc.cgroup.devices.allow: c 189:* rwm
lxc.mount.entry: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2414690-if00 dev/ttyACMZigbee none bind,optional,create=file
```
4) LXC container: Install Z2M as described here https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html
5) LXC container: Use `/dev/ttyACMZigbee` in Z2M config `/opt/zigbee2mqtt/data/configuration.yaml`
```
serial:
  adapter: deconz
  port: /dev/ttyACMZigbee
```

Ramshield, 2021-05-28:
Thanks @AlexKay88 I figured out the problem. If I have ANY other USB device in it, it won't work. 
No clue why or how, but removing the devices and I can learn new devices again.

closed 2021-05-28 by Ramshield

Ramshield, 2021-05-28:
> > If I have ANY other USB device in it, it won't work.
> 
> I have a Z-Wave USB stick connected and mapped to same LXC container without a problem.

I got 2 HDD's, a broadlink, RFlink and RIPE probe USB. This all on an Intel NUC. Worked before with the same devices though, and now all of the sudden it doesn't...

Ramshield, 2021-06-07:
@juslex @AlexKay88 so I found the actual problem. I recently bought a couple Philips Hue lights and was messing around.
Then I plugged my external HDD's back in, and encountered this error: `MAC channel access failure' (225))`
I searched this and found this Github issue: https://github.com/Koenkk/Z-Stack-firmware/issues/240
So somehow external HDD's cause interference. I placed my Zigbee USB device on a USB external and moved the external HDD's back a bit, and it works again!

seb49, 2022-08-18:
Hello,

Which template is the best to use Zigbee2Mqtt in LXC container ? I try with Alpine but I can't install. It's OK with Ubuntu template

Regards

juslex, 2022-08-18:
You can use this here

https://github.com/tteck/Proxmox