Koenkk/zigbee2mqtt #7383: Lonsonho dimmer does not turn on properly on a dimmed level

Vinndimon, 2021-05-07:
## What happened
I have the [Lonsonho QS-Zigbee-D02-TRIAC-L dimmer](https://www.zigbee2mqtt.io/devices/QS-Zigbee-D02-TRIAC-L.html#lonsonho-qs-zigbee-d02-triac-l). What I try to do is to turn on the dimmer on a dimmed level right away:

1. I set some dimmer brightness from the z2m "Exposes" section.
2. The dimmer says that its' state is "ON", but nothing actually happens on the device itself - the light stays off.
3. At once, the brightness level sets itself to 245. However, the light stays off.
4. If I set the dimmer to OFF, then to ON - it finally turns ON, at the maximum, brightness.

[Here is the video](https://drive.google.com/file/d/1rE51W6WjqrjquQMl67sU9yXFppyhyI1u/view?usp=sharing)

I need this dimmer as a wake-up light, but it requires the capability to turn on the light at the dimmed level.

## What did you expect to happen
When I change the brightness value, the dimmer turns on on this level right away.

## How to reproduce it (minimal and precise)
Try to set the level when the dimmer is off.

## Debug info
Zigbee2MQTT version: 1.18.3-1
Adapter hardware: JetHome ZB6UX v.1 (CC2652)
Adapter firmware version: CC1352P2_CC2652P_launchpad_coordinator_20210120

mercenaruss, 2021-05-09:
Having the same issue with Gledopto GL-C-009.
This come up after last update.

Drafteed, 2021-05-12:
Same with GL-C-007-2ID

`latest-dev:1.18.3-dev`

log:
```
2021-05-12 04:34:16 ERROR (MainThread) [homeassistant.util.logging] Exception in state_received when handling msg on 'zigbee2mqtt/gledopto_rgbw1/rgb': '{"brightness":254,"color":{"x":0.522,"y":0.218},"color_temp":19,"state":"ON"}'
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 32, in wrapper
msg_callback(msg)
File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 306, in state_received
self._update_color(values)
File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 245, in _update_color
color_mode = values["color_mode"]
KeyError: 'color_mode'
```

Koenkk, 2021-05-12:
After the update you need to set the color once (simply click a random color on the color wheel).

Vinndimon, 2021-05-13:
@Koenkk what about Lonsonho? Maybe it's a different problem?

mercenaruss, 2021-05-14:
@Koenkk Gledopto GL-C-009 the same issue.

{"state": "ON", "brightness": 213} - not working
{"state": "ON"} - is working

Koenkk, 2021-05-15:
@mercenaruss fixed it for the GL-C-009. 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

@Vinndimon do you have the exact same problem as @mercenaruss ? 👍 

> {"state": "ON", "brightness": 213} - not working
{"state": "ON"} - is working

If yes, this means your device does not support turning on at a brightness. (so it has to be turned on first at the brightness were it was turned off with, and then set the brightness).

Vinndimon, 2021-05-15:
@Koenkk "if yes"? Maybe "if no"?

Vinndimon, 2021-05-16:
@Koenkk even if this dimmer does not support turning on on some specific brightness. maybe it shouldn't report its state as ON when it is actually OFF?

Koenkk, 2021-05-16:
@Vinndimon does the dimmer turn on when off and sending : `{"brightness": 213}`

closed 2021-06-23 by github-actions[bot]