Koenkk/zigbee2mqtt #7388: Unsupported device failing interview

godzone, 2021-05-07:
I recently purchased a battery powered Zigbee Motion detector,  seems to be quite a popular device but is currently unknown to Zigbee2Mqtt. As its quite cheap, I'd love to get it to work and my efforts may help others.
I am using the Zigbee2Mqtt Add-on in HomeAssistant, it is version 1.18.1-2
I have the device in the office right beside the USB coordinator (slae.sh_cc2652rb_stick), I have attempted peering multiple times. The device joins but then the interview fails. I have turned on the herdsman debug but am having a hard time understanding where to go next as I dont understand the output. It appears that the device is inaccessible once joined

[xxx.txt](https://github.com/Koenkk/zigbee2mqtt/files/6444599/xxx.txt)

godzone, 2021-05-08:
Some progress, without any intervention, the device seems to be sorting itself out.  It has now at least identified itself.
model 'TS0202' manufacturer name '_TYZB01_dl7cejts'.  Still unsupported but at least I can make a start

godzone, 2021-05-16:
I thought this would be straight forward :-( I followed the various instructions for adding a new device, but when I add the external command to the config, z2m fails to start.  It looks like an include is failing but I have checked all the file names and everything looks ok.  I am using the HAS plugin and wonder if the instruction are not quite right or the paths in the examples don't match. WIth everything running in a container its a little more difficult to debug.

Schnittcher, 2021-05-23:
I have problems with the same device:
Warning Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

When i try to pair this device with the Sonoff Zigbee Bridge, i geht the Model and the manufacturer name.
Can anyone help to pair this device with Zigb22MQTT?

godzone, 2021-05-23:
I added the model and manufacturer to the tuya.js file as an experiment.  I then get the signal strength and battery level being returned but occupancy is still N/A. I have tried adding the device as an external but other than crashing z2m on startup have had no success.

Schnittcher, 2021-05-23:
Can you send me your external file?

godzone, 2021-05-23:
I included in email reply, did you get it?

Schnittcher, 2021-05-23:
No, i didn't get it.

godzone, 2021-05-23:
I didnt get very far due to z2m failing to start. I was starting simple but was prepared to try to debug and implement the apppropriate 'exposes' methods

/config/zigbee2mqtt/TYZB01_dl7cejts.js

# -------------

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0202', manufacturerName: '_TYZB01_dl7cejts'}],
    zigbeeModel: ['TS0202_Motion_Detector'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'TS0202', // Vendor model number, look on the device for a model number
    vendor: 'TuYa', // Vendor of the device (only used for documentation and startup logging)
    description: 'Infra Red Motion Detector', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery, fz.ignore_basic_report], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.occupancy(), e.battery_low(), e.tamper(), e.battery()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;

Schnittcher, 2021-05-24:
Thanks.
With your external converter i get this. too:

Warning Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

What can I doo :-(

Schnittcher, 2021-05-24:
Now I get this:

[image]

But the interview failed. So i think i have the same situation as you, right?
Can anyone help us?

Schnittcher, 2021-05-24:
Sooo, now i get this Debugs:

Debug No converter available for 'TS0202' with cluster 'ssIasZone' and type 'attributeReport' and data '{"zoneStatus":1}'
Debug Received Zigbee message from '0x00124b0005b4ca90', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0

Can anyone help?

closed 2021-05-25 by Koenkk