Koenkk/zigbee2mqtt #7389: TRADFRI bulb E14 CWS 470lm

zebar88, 2021-05-07:
## Information about the device + link
Hello, 

I'm trying to add the new device TRADFRI bulb E14 CWS 470lm to zigbee2mqqt. I don't know if its possible or not.
Someone could help? the model is the LED1925G6.

warning Device '0x680ae2fffe597482' with Zigbee model 'TRADFRI bulb E14 CWS 470lm' and manufacturer name 'IKEA of Sweden' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

I don't know if I understood correctly but I only need to create a file <something>.js like the one bellow on the zignbee2mqtt path (the same as the database.db) ?

############

const exposes = require('../lib/exposes');
const fz = {...require('../converters/fromZigbee'), legacy: require('../lib/legacy').fromZigbee};
const tz = require('../converters/toZigbee');
const ota = require('../lib/ota');
const constants = require('../lib/constants');
const reporting = require('../lib/reporting');
const extend = require('../lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const bulbOnEvent = async (type, data, device) => {
    /**
     * IKEA bulbs lose their configured reportings when losing power.
     * A deviceAnnounce indicates they are powered on again.
     * Reconfigure the configured reoprting here.
     * NOTE: binds are not lost so rebinding is not needed!
     */
    if (type === 'deviceAnnounce') {
        for (const endpoint of device.endpoints) {
            for (const c of endpoint.configuredReportings) {
                await endpoint.configureReporting(c.cluster.name, [{
                    attribute: c.attribute.name, minimumReportInterval: c.minimumReportInterval,
                    maximumReportInterval: c.maximumReportInterval, reportableChange: c.reportableChange,
                }]);
            }
        }
    }
};

module.exports = [
    {
        zigbeeModel: ['TRADFRI bulb E14 CWS 470lm',
        model: 'TRADFRI bulb E14 CWS 470lm',
        vendor: 'IKEA',
        description: 'TRADFRI LED bulb E14 470 lumen, dimmable, color, opal white',
        extend: extend.light_onoff_brightness_color(),
        ota: ota.tradfri,
        meta: {supportsHueAndSaturation: false},
        onEvent: bulbOnEvent,
    },
];

pro2call-nl, 2021-05-15:
how can i add this bulb to my system?

Using HA with Zibee2mqtt version 1.18.3-1.

Its detected as unsuported

grigio, 2021-05-26:
+1 I also get unsupported, which zigbee2mqtt supports this bulb ?

pro2call-nl, 2021-05-27:
> +1 I also get unsupported, which zigbee2mqtt supports this bulb ?

It's in the new release. Which will be released in the next couple of days