Koenkk/zigbee2mqtt #7392: CC2531 - Failing to interview devices

thebencourt, 2021-05-08:
I've been using the CC2531 for a few months, all was working fine but after reinstalling (after updating my pi to Ubuntu 21.04) and pulling the latest docker image I can't get anything to pair.

I've started from scratch to make sure it wasn't old config files causing the issue, I've also tried running it via docker and on bare metal on two different computers (raspberry pi 4 and an x86 intel machine), both giving the exact same errors.

Trying to pair a single hue bulb which is less than 1m from the zigbee adapter I get the following:

```
Zigbee2MQTT:info  2021-05-08 10:35:53: Device '0x0017880108338bc5' joined
Zigbee2MQTT:info  2021-05-08 10:35:53: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0017880108338bc5","ieee_address":"0x0017880108338bc5"},"type":"device_joined"}'
Zigbee2MQTT:info  2021-05-08 10:35:53: Starting interview of '0x0017880108338bc5'
Zigbee2MQTT:info  2021-05-08 10:35:53: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0017880108338bc5","ieee_address":"0x0017880108338bc5","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2021-05-08 10:35:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x0017880108338bc5"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-05-08 10:35:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x0017880108338bc5"},"type":"pairing"}'
(node:7) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:7) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
Zigbee2MQTT:error 2021-05-08 10:37:40: Failed to interview '0x0017880108338bc5', device has not successfully been paired
```

If I then try to remove the device form the UI it fails with this error:
```
Zigbee2MQTT:error 2021-05-08 10:38:03: Request 'zigbee2mqtt/bridge/request/device/remove' failed with error: 'Failed to remove device '0x0017880108338bc5' (block: false, force: false) (Error: SRSP - ZDO - mgmtLeaveReq after 6000ms)'
```

Let me know if you need any more logs or to try anything else

thebencourt, 2021-05-08:
If I leave the docker container running (no devices paired) I also get this error repeating constantly
```
(node:7) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
(node:7) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
```

Koenkk, 2021-05-08:
Can you try flashing this firmware instead: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing ?

Koenkk, 2021-05-10:
Not that I'm aware of

thebencourt, 2021-05-21:
Finally got the jumper wires to flash the new firmware, back working perfectly. Thanks!

closed 2021-05-21 by thebencourt