Koenkk/zigbee2mqtt #7393: OTA Updates are specified as presence sensors

Patrick1610, 2021-05-08:
## What happened
When a device which is added to an area has an available OTA update, the magic areas integration sees this as activity because the update sensor is specified as 'presence'. This feels strange since a device that has an available update has nothing to do with presence in a room. 

## What did you expect to happen
That the device did not activate a presence sensor when it has an OTA update available. 

## How to reproduce it (minimal and precise)
Install magic areas, add a device with an OTA update available to an area. Select presence as sensor in the magic area. Reboot Home Assistant. 

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

rindlerblabla, 2021-05-09:
Instead of configuring area of the unit in Home assistant, can't you just configure area of the occupancy entity?

Patrick1610, 2021-05-09:
Thanks for your reply, I don't really understand what your suggestion means, can you elaborate?

rindlerblabla, 2021-05-09:
When you find your device in Configure > Devices in Home assistant, instead of setting an area of the device itself (by  clicking on the pencil in the upper right corner and then choose area), click on the occupancy entity > Advanced settings and choose an area for the entity.

frenck, 2021-06-03:
Could you provide a state dump of that entity?

(developer tools -> states tab)

closed 2021-07-11 by github-actions[bot]