Koenkk/zigbee2mqtt #7405: Device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_cphmq0q7'

r52alex, 2021-05-09:
Hi,

I have this log in mqtt add-on. Could you support my device? Thank you.

Device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_cphmq0q7'

Zigbee2MQTT:warn  2021-05-09 13:39:43: Device '0x847127fffe8d9764' left the network
Zigbee2MQTT:warn  2021-05-09 13:39:48: Device '0x847127fffe8d9764' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_cphmq0q7' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Zigbee2MQTT:warn  2021-05-09 13:40:30: Device '0x588e81fffedb8848' left the network
Zigbee2MQTT:warn  2021-05-09 13:40:31: Device '0x588e81fffedb8848' left the network
Zigbee2MQTT:error 2021-05-09 13:41:59: Failed to execute LQI for 'projectors'
Zigbee2MQTT:warn  2021-05-09 13:44:58: Device '0x847127fffe8d9764' left the network
Zigbee2MQTT:warn  2021-05-09 13:44:59: Device '0x847127fffe8d9764' left the network
Zigbee2MQTT:warn  2021-05-09 13:45:36: Device '0x847127fffe8d9764' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_cphmq0q7' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Bought this device here

trashographer, 2021-05-09:
Same problem, bought 2 items, same link
warning Device '0x847127fffe8d9783' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_cphmq0q7' is NOT supported

shvirik, 2021-05-13:
Dear,

I bought 3, and two works properly and the third one gave me the same error :( Different seller, but the same item (apparently).

Regards,

shvirik, 2021-05-13:
one additional info. 2/3 devices that work properly are recognized by zigbee2mqtt as TS0121_plug.

razmoket89, 2021-05-13:
Indeed @shvirik , I bought from the same seller and am having the same problem. The item is recognised as TS011F, but was sold as TS0121 which is working with z2m.

shvirik, 2021-05-15:
> 
> 
> Indeed @shvirik , I bought from the same seller and am having the same problem. The item is recognised as TS011F, but was sold as TS0121 which is working with z2m.

@razmoket89  do you have an operative TS0121_plug?. Because, I have two working, but they dont  expose the energy sum of consumed energy:

[image]

Do you have this issue too?

r52alex, 2021-05-15:
I have one TS0121_plug.it work fine. Exposes  Kwh
[image]

razmoket89, 2021-05-15:
> @razmoket89 do you have an operative TS0121_plug?. Because, I have two working, but they dont expose the energy sum of consumed energy:
>
> [image]
>
> Do you have this issue too?

I dont own any TS0121_plug, sorry.

shvirik, 2021-05-15:
> I have one TS0121_plug.it work fine. Exposes Kwh
> Uploading Screenshot_2021-05-15-13-04-28-562_io.homeassistant.companion.android.jpg…

Thanks. I don't know why mine ones dont give me the sum values.

shvirik, 2021-05-15:
> I dont own any TS0121_plug, sorry.

Thanks

Koenkk, 2021-05-16:
Added now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-05-16 by Koenkk

r52alex, 2021-05-16:
Thank you very much!

Koenkk, 2021-05-18:
Does somebody have the TuYa gateway? If yes please help with this issue: https://github.com/Koenkk/zigbee-herdsman-converters/pull/2587#issuecomment-842416346

manJerky, 2021-05-18:
Just tried and got unsupported message on dev branch (1.18.3-dev):

`Device '0x847127fffe8d9689' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_cphmq0q7' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
`

Shouldn't there be no issue when the model and manufacturer name are the same or am I misunderstanding?

pszafer, 2021-05-26:
hi, I got new plugs and unsupported as well:
```
2021-05-26 13:15:05: Device '0x847127fffe8d8686' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_ew3ldmgx' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
```

should I create PR against - https://github.com/Koenkk/zigbee-herdsman-converters/blob/ba9e37c01e493df1d51e960df7aa26b39faa316e/devices/tuya.js#L649-L670 or just wait?

Koenkk, 2021-05-26:
@pszafer is it this plug https://zigbee.blakadder.com/Tuya_TS011F.html ?

pszafer, 2021-05-26:
manufacter name is different but looks very similar. https://a.aliexpress.com/_uzbEDB

Koenkk, 2021-05-27:
@pszafer added, 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

kotyla, 2021-07-06:
Hello,

I've got another TS011F plug not supported. I've tried to use instruction how to support new devices but I failed (seems too complicated for me). I've also tried dev branch.

Is it possible to add it ?
Thank you.

`Device '0x60a423fffeff96db' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_jvzvulen' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
`
[image]
[image]

david08154711, 2021-08-03:
I have a problem too
Warning device '0x84fd27fffece9981' with Zigbee model 'TS011F' and manufacturer name '_TZ3000_mraovvmm' is NOT supported

I've already switched to dev branch.

(description: 'Smart plug (with power monitoring)')

the manufacturerName '_TZ3000_mraovvmm' is not contained in the tuya.js

Koenkk, 2021-08-03:
@david08154711 added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

miit86, 2021-10-01:
David,

Did you manage to get your "After Power Outage" option set-up?

KMDobrev, 2021-10-30:
> Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name **'_TZ3000_w0qqde0g**

I also have one of these devices and Energy measurements are not displayed.

dominikkarall, 2021-11-16:
I also got an unsupported device
TS0011F
_TZ3000_nfnmi125

Can you please add this one? Thx!

KMDobrev, 2021-11-16:
Hallo,

_TZ3000_mraovvmm

This doesn't work either.

[image]
[image]

wkryzel, 2021-11-17:
Hallo,

_TZ3000_x2ddiwsv

This doesn't work either.

n8schicht17, 2021-11-24:
Hi,
could you please also support this one?

Model: TS011F
Manufacturer: _TZ3000_cphmq0q7

It's not reporting power, current, voltage, energy

Thanks a lot!

Koenkk, 2021-11-24:
@n8schicht17 that issue has just been fixed in the dev branch.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

KMDobrev, 2021-11-24:
> @n8schicht17 that issue has just been fixed in the dev branch.
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Thank you🙏

yukeiw, 2021-11-26:
Hi there,

Do you think you could add the one below as well?

Model: TS011F
Manufacturer:_TZ3000_cymsnfvf

power, current, voltage, energy are missing.

Much appreciated!

simie7, 2021-12-01:
> I also got an unsupported device TS0011F _TZ3000_nfnmi125
> 
> Can you please add this one? Thx!

@dominikkarall you can get it working reading [this](https://github.com/Koenkk/zigbee2mqtt/issues/9493#issuecomment-981153732) issue.

The problem left is energy monitoring, it's not yet supported, as uses new undeclared cluster for this.

KMDobrev, 2021-12-16:
In my case, these two models still do not show energy measurement in the Edge version.

_TZ3000_ew3ldmgx

_TZ3000_mraovvmm

Koenkk, 2021-12-16:
@KMDobrev can you post the data/database.db entry of your devices?

KMDobrev, 2021-12-16:
{"id":26,"type":"Router","ieeeAddr":"0x540f57fffeddebbc","nwkAddr":5067,"manufId":4098,"manufName":"_TZ3000_ew3ldmgx","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_ew3ldmgx","powerSource":1,"zclVersion":3,"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00238db4ae","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b00238db4ae","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b00238db4ae","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":680236614},"lastSeen":1639678653689,"useImplicitCheckin":true}

Koenkk, 2021-12-16:
@KMDobrev what model is displayed in the z2m frontend, `TS011F_plug_3` or `TS011F_plug_1`?

KMDobrev, 2021-12-16:
> @KMDobrev what model is displayed in the z2m frontend, `TS011F_plug_3` or `TS011F_plug_1`?

This is the other

{"id":40,"type":"Router","ieeeAddr":"0x84fd27fffece7f84","nwkAddr":15042,"manufId":4098,"manufName":"_TZ3000_mraovvmm","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_mraovvmm","powerSource":1,"zclVersion":3,"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00238db4ae","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b00238db4ae","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b00238db4ae","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":69,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":680236614},"lastSeen":1639679399977,"useImplicitCheckin":true}

KMDobrev, 2021-12-16:
> `TS011F_plug_1`?

Both the same

TS011F_plug_1

Koenkk, 2021-12-16:
@KMDobrev then you are not on the latest edge, uninstall and reinstall to upgrade.

KMDobrev, 2021-12-16:
> @KMDobrev then you are not on the latest edge, uninstall and reinstall to upgrade.

Sorry

TS011F_plug_3

ferropolis, 2021-12-20:
> @KMDobrev then you are not on the latest edge, uninstall and reinstall to upgrade.

Hey @Koenkk any information or timetable when will it be available in vanilla mode (normal Zigbee2MQTT without Edge)?
Thanks

Koenkk, 2021-12-20:
@ferropolis release is on every 1st of the month (so 1 January next release)

KMDobrev, 2021-12-20:
> @ferropolis release is on every 1st of the month (so 1 January next release)

I uninstalled and reinstalled Edge but the problem persists.

In the Edge it say TS011F_plug_3
In the Normal stable version it say TS011F_plug_1

frederikbove, 2021-12-24:
I'm trying to add the same device: 
Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_v1pdxuqq'

mkotek, 2021-12-24:
In my case, just got 4 of SHP15 marked as _TZ3000_u5u4cakc

Koenkk, 2021-12-25:
@frederikbove added `https://github.com/Koenkk/zigbee2mqtt/issues/7405` (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

@mkotek will work in stable with the next release (1 January)

mortelil, 2021-12-26:
Hello!
I have a XH-002P plug, identified as TS011F_plug_1 - _TZ3000_v1pdxuqq in the latest edge version. Power, voltage, current and energy is marked as 'Null' in the z2m interface, power switch works. Hope for support for this. This is the info from database.db:

{"id":38,"type":"Router","ieeeAddr":"0xa4c138aeff547348","nwkAddr":7829,"manufId":4417,"manufName":"_TZ3000_v1pdxuqq","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_v1pdxuqq","powerSource":1,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"20480":1,"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2}},"manuSpecificTuya_3":{"attributes":{"53264":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff079473","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1640547141776,"useImplicitCheckin":true}

tansuka, 2022-01-01:
Any luck for same device but with the manufacturer _TZ3000_typdpbpg 
Thank you very much

mortelil, 2022-01-02:
I updated my Zigbee2MQTT edge today, and the _TZ3000_v1pdxuqq still wont work. Is this supposed to be available in the next stable release? 
I get the following errors in my log when I plug it in:

Failed to configure '0xa4c138aeff547348', attempt 2 (Error: Bind 0xa4c138aeff547348/1 haElectricalMeasurement from '0x00212effff079473/1' failed (Error: status: 130) at DeconzAdapter.bind (/app/node_modules/zigbee-herdsman/src/adapter/deconz/adapter/deconzAdapter.ts:753:19) at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:517:13) at Object.bind (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:33:9) at Object.configure (/app/node_modules/zigbee-herdsman-converters/devices/tuya.js:1084:13) at Configure.configure (/app/lib/extension/configure.ts:115:13))

Failed to ping '0xa4c138aeff547348' (attempt 1/1, Read 0xa4c138aeff547348/1 genBasic(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (no response received))

WouterJN, 2022-01-03:
After the latest update my plug (_TZ3000_u5u4cakc) works fine, thnx!

KochergaAlex, 2022-01-17:
Hi,
I have this log in mqtt add-on. Could you support my device? 
Thank you.

Zigbee model 'TS011F' and manufacturer name '_TZ3000_cfnprab5' is NOT supported

KochergaAlex, 2022-01-17:
Zigbee model 'TS011F' and manufacturer name '_TZ3000_cfnprab5' is NOT supported

Smart Power Strip  -  Model: SM-SO306E-1Z

kkossev, 2022-01-27:
@frederikbove is your **_TZ3000_v1pdxuqq** model reporting any power or energy readings?

> inClusterList":[3,4,5,6,57344,57345,0]
There are no power or energy clusters there, nor the usual Tuya EF00 cluster ..

Koenkk, 2022-01-28:
@kkossev based on the cluster list, it doesn't support this indeed. Moved it to TS011F_plug_2 (Smart plug (without power monitoring))

JJussi, 2022-02-03:
I just received multiple TS011F, but with different  manufacturer name '_TZ3000_typdpbpg'. These are with power monitoring.  Could this be added to the supported models and deployed to HA side?

Koenkk, 2022-02-03:
@JJussi `_TZ3000_typdpbpg` is already supported in z2m 1.23.0

JJussi, 2022-02-03:
OK. 
My HA z2m add-on is 1.20.0-1, so... I need to wait that a newer version is published.  Or is there an easy way to install a newer version to my system?

Koenkk, 2022-02-04:
@JJussi HA should already show an update, not sure how to force it but uninstall -> install again may fix it.

SAMUD, 2022-02-04:
I also just received an TS011F, this time with model ID TZ3000_gjnozsaz.

`Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_gjnozsaz'`

It is[ this one](https://www.ebay.de/itm/334215428684), so it should also support power metering.

Would be cool If you could add it. Thanks :)

sermayoral, 2022-02-04:
Hi! I need an smart plug and i have seen this interesting model. I have read in https://www.zigbee2mqtt.io/devices/TS011F_plug_3.html a broken reporting functionality. Is it compatible right now or i should buy the shp13 old model?

Thanks!!

JJussi, 2022-02-04:
I have those power plugs and they work fine...
{
    "current": 0.05,
    "energy": 0.27,
    "linkquality": 0,
    "power": 10,
    "power_outage_memory": "restore",
    "state": "ON",
    "voltage": 232
}

SAMUD, 2022-02-07:
> I also just received an TS011F, this time with model ID TZ3000_gjnozsaz.
>
> `Received message from unsupported device with Zigbee model 'TS011F' and manufacturer name '_TZ3000_gjnozsaz'`
>
> It is[ this [...]

I made a pull request and found out that this device was already added. It was just my Docker container which was a "little bit" outdatted :D

[67 more replies]