Koenkk/zigbee2mqtt #7407: Device NAS-AB02B0 pairs but does not work

BerndK, 2021-05-09:
Was not able to use Zigbee Siren https://www.zigbee2mqtt.io/devices/NAS-AB02B0.html

## What happened
I pressed the button in the back of the siren (close to the batteries) for 3 secs, a beep and Device joined and paired.
I paired the siren more or less successfully. System reports successful pairing and reports proper description (see log below).
Using Assistant GUI also reports the correct picture, so something seems to work

## What did you expect to happen
I would expect to receive some messages with payload like humidity, temperature and other values (similar to the last line in log which is another device).
Next thing I expect is that I will be able to send data to the siren like
zigbee2mqtt/0x804b50fffe0482ae/set { "alarm": true }
nothing happens.
While playing around for hours once I got this error:
`Zigbee2MQTT:error 2021-05-09 12:53:18: Publish 'set' 'alarm' to '0x804b50fffe0482ae' failed: 'Error: Command 0x804b50fffe0482ae/1 manuSpecificTuya.setData({"status":0,"transid":10,"dp":104,"datatype":1,"length_hi":0,"length_lo":1,"data":[1]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))'`
Was not able to reproduce the error. After normal sending data just nothing happens. I'm able to send data to other devices like e.g an OSRAM Plug switch without problems.

## How to reproduce it (minimal and precise)
I pressed button 3 secs, beep, got the log below. Siren is constantly blinking when powered up with USB. Blinking continues after pairing and after repower...

## Debug info
Device: NAS-AB02B0
Zigbee Id is 0x804b50fffe0482ae.
Address on Zigbee Network is 0x6474.
Link quality is 102/255.
Model is TS0601 by _TZE200_d0yu2xgi.
Hardware version is 1.
Firmware version is unknown.
This device type is EndDevice

Version of Zigbee2Mqtt: 1.18.1
Coordinator version: 20190425
Coordinator Zigbee Id: 0x00124b0018df4239
Adapter hardware: CC25xx USB Dongle Rev 1.0
Adapter firmware version: ?

This is the log I see:
All items belong to the siren (0x804b50fffe0482ae) except the last lane that shoudl show that ohter items works well, so the overall setup is not broken.

```
Zigbee2MQTT:info  2021-05-09 13:21:54: Device '0x804b50fffe0482ae' joined
Zigbee2MQTT:info  2021-05-09 13:21:54: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x804b50fffe0482ae","ieee_address":"0x804b50fffe0482ae"},"type":"device_joined"}'
Zigbee2MQTT:info  2021-05-09 13:21:54: Starting interview of '0x804b50fffe0482ae'
Zigbee2MQTT:info  2021-05-09 13:21:54: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x804b50fffe0482ae","ieee_address":"0x804b50fffe0482ae","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2021-05-09 13:21:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x804b50fffe0482ae"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-05-09 13:21:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x804b50fffe0482ae"},"type":"pairing"}'
Zigbee2MQTT:info  2021-05-09 13:22:09: Successfully interviewed '0x804b50fffe0482ae', device has successfully been paired
Zigbee2MQTT:info  2021-05-09 13:22:09: Device '0x804b50fffe0482ae' is supported, identified as: Neo Temperature & humidity sensor and alarm (NAS-AB02B0)
Zigbee2MQTT:info  2021-05-09 13:22:09: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Temperature & humidity sensor and alarm","exposes":[{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"°C"},{"access":1,"description":"Measured relative humidity","name":"humidity","property":"humidity","type":"numeric","unit":"%"},{"access":3,"name":"humidity_alarm","property":"humidity_alarm","type":"binary","value_off":false,"value_on":true},{"access":1,"description":"Indicates if the battery of this device is almost empty","name":"battery_low","property":"battery_low","type":"binary","value_off":false,"value_on":true},{"access":3,"name":"temperature_alarm","property":"temperature_alarm","type":"binary","value_off":false,"value_on":true},{"access":3,"name":"alarm","property":"alarm","type":"binary","value_off":false,"value_on":true},{"access":3,"name":"melody","property":"melody","type":"enum","values":["1","2","3","4","5","6","7","8","9","10","11","12","13","14","15","16","17","18"]},{"access":3,"name":"duration","property":"duration","type":"numeric","unit":"second"},{"access":3,"name":"temperature_min","property":"temperature_min","type":"numeric","unit":"°C"},{"access":3,"name":"temperature_max","property":"temperature_max","type":"numeric","unit":"°C"},{"access":3,"name":"humidity_min","property":"humidity_min","type":"numeric","unit":"%"},{"access":3,"name":"humidity_max","property":"humidity_max","type":"numeric","unit":"%"},{"access":3,"name":"volume","property":"volume","type":"enum","values":["low","medium","high"]},{"access":1,"name":"power_type","property":"power_type","type":"enum","values":["battery_full","battery_high","battery_medium","battery_low","usb"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"NAS-AB02B0","supports_ota":false,"vendor":"Neo"},"friendly_name":"0x804b50fffe0482ae","ieee_address":"0x804b50fffe0482ae","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:info  2021-05-09 13:22:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Temperature & humidity sensor and alarm","friendly_name":"0x804b50fffe0482ae","model":"NAS-AB02B0","supported":true,"vendor":"Neo"},"type":"pairing"}'
Zigbee2MQTT:info  2021-05-09 13:22:30: MQTT publish: topic 'zigbee2mqtt/0x00158d0002559476', payload '{"battery":80,"humidity":46.24,"linkquality":115,"temperature":23.13,"voltage":2965}'
```

dfrome, 2021-05-11:
Hi Bernd.
Do you have a native Tuya gateway ? It seems nobody can prepare a working HA integration as long as we can't find anybody to sniff conversations between this siren and a tuya gateway.

BerndK, 2021-05-11:
No I don't have a Tuya Gateway. There is an App for mobile phones (iOS, Android), but I did not use it. Alos not sure if this app talks WiFi or ZigBee!? I'm not a real expert in ZigBee. I just realized that the pairing works great and the report of the available values looks promising.
Perhaps we have to use another firmware like Tasmota?
If there is someone out there, that can do the sniffing - I'm willing to send the device to Europe locations!

BerndK, 2021-06-02:
Now I bought an Tuya coordinator and an additional CC2531 and were able to create some sniffing files.
I’m not an expert in Zigbee, but am a SW developer, so I tried to analyze the communication from tuya to the Siren NAS-AB02B0.
I found that the datapoints seems to be OK.
To Device:
```
Alarm ON : 00076801000101 DP 0x68 (104) Bool (01) fn:00, Data 0101 (01=length=1???, Bits: 1 = ON)
Alarm OFF: 00086801000100
From Device
Humidity? : 00196a0200040000004f DP 0x6a (106) Value (04) fn:00, Data 04 length?, 0000004f = 79%?
Temp?     : 002369020004000000f6 DP 0x69 (105) Value (04) fn:00, Data 04 length?, 000000f6 = 246 24,6?
```
This looks as expected (from my point of view)!

But please remember, that the initial problem is that the device seems not to pair completely. Not sure about that statement, but the siren keeps blinking when pairing with zigbee2mqtt.
When I pair it to the tuya hub, the blinking stops.
To dig deeper, I will include a part of the zigbee2mqtt logs, a wireshark file while pairing to tuya and one pairing to zigbee2mqtt. There is also a file that shows alarm on/off via Tuya app.
Without having more info, to me it looks like the device is waiting for a „warm welcome tot he network message“ to complete pairing and stop blinking. Perhaps it waits for some time settings (see the unknown datapoint debug message) and will then be happy.
**Perhaps it waits for the result of command 0x24 with data 0006 (pairing with zigbee2mqtt - packet 233) wich is answered by the tuya hub (pairing with tuya - packets 406, 412) with 0x24 Data 003d60b0cbbf60b0e7df ???**
Not being a zigbee expert, I don’t know what that command is for.
Please let me know, if I can do any further sniffs or tests.

Keys (to encrypt the Wireshark files, see attached zip file):
5A:69:67:42:65:65:41:6C:6C:69:61:6E:63:65:30:39
29c640fe2398d435b9e208bafc2f9440
01030507090b0d0f00020406080a0c0d

ZigBeeToMQTT log: (Plese note the last few messages)
Attaching to zigbee2mqtt
```
zigbee2mqtt             | Using '/app/data' as data directory
zigbee2mqtt             | Zigbee2MQTT:debug 2021-05-29 10:43:16: Loaded state from file /app/data/state.json
zigbee2mqtt             | Zigbee2MQTT:info  2021-05-29 10:43:16: Logging to console and directory: '/app/data/log/2021-05-29.10-43-13' filename: log.txt
zigbee2mqtt             | Zigbee2MQTT:debug 2021-05-29 10:43:16: Removing old log directory '/app/data/log/2021-05-28.09-24-09'
zigbee2mqtt             | Zigbee2MQTT:info  2021-05-29 10:43:16: Starting Zigbee2MQTT version 1.18.3 (commit #f2e39af)
zigbee2mqtt             | Zigbee2MQTT:info  2021-05-29 10:43:16: Starting zigbee-herdsman (0.13.92)
zigbee2mqtt             | Zigbee2MQTT:debug 2021-05-29 10:43:16: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
zigbee2mqtt             | Zigbee2MQTT:info  2021-05-29 10:43:19: zigbee-herdsman started
zigbee2mqtt             | Zigbee2MQTT:info  2021-05-29 10:43:19: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20201127,"transportrev":2},"type":"zStack12"}'
zigbee2mqtt             | Zigbee2MQTT:debug 2021-05-29 10:43:19: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754}
zigbee2mqtt             | Zigbee2MQTT:info  2021-05-29 10:43:19: Currently 3 devices are joined:
zigbee2mqtt             | Zigbee2MQTT:info  2021-05-29 10:43:19: AquaraTempSensor_1 (0x00158d0002559476): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)
zigbee2mqtt             | Zigbee2MQTT:info  2021-05-29 10:43:19: XiaomiSwitch_1 (0x00158d00029233d9): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
zigbee2mqtt             | Zigbee2MQTT:info  2021-05-29 10:43:19: 0x804b50fffe0482ae (0x804b50fffe0482ae): NAS-AB02B0 - Neo Temperature & humidity sensor and alarm (EndDevice)
zigbee2mqtt             | Zigbee2MQTT:warn  2021-05-29 10:43:19: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt             | Zigbee2MQTT:warn  2021-05-29 10:43:19: Allowing new devices to join.
zigbee2mqtt             | Zigbee2MQTT:warn  2021-05-29 10:43:19: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt             | Zigbee2MQTT:info  2021-05-29 10:43:19: Zigbee: allowing new devices to join.
zigbee2mqtt             | Zigbee2MQTT:info  2021-05-29 10:43:19: Connecting to MQTT server at mqtt://127.0.0.1:1883
zigbee2mqtt             | Zigbee2MQTT:debug 2021-05-29 10:43:19: Using MQTT anonymous login
zigbee2mqtt             | Zigbee2MQTT:info  2021-05-29 10:43:19: Connected to MQTT server
zigbee2mqtt             | Zigbee2MQTT:info  2021-05-29 10:43:19: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt             | Zigbee2MQTT:info  2021-05-29 10:43:19: MQTT publish: topic 'zigbee2mqtt/AquaraTempSensor_1', payload '{"battery":80,"humidity":45.67,"linkquality":102,"temperature":20.7,"voltage":2965}'
zigbee2mqtt             | Zigbee2MQTT:info  2021-05-29 10:43:19: MQTT publish: topic 'zigbee2mqtt/XiaomiSwitch_1', payload '{"battery":100,"linkquality":42,"voltage":3042}'
[... cut ...]
zigbee2mqtt             | Zigbee2MQTT:debug 2021-05-29 10:44:13: Received Zigbee message from '0x804b50fffe0482ae', type 'raw', cluster 'manuSpecificTuya', data '{"data":[9,2,17,0,7,68],"type":"Buffer"}' from endpoint 1 with groupID 0
zigbee2mqtt             | Zigbee2MQTT:debug 2021-05-29 10:44:13: No converter available for 'NAS-AB02B0' with cluster 'manuSpecificTuya' and type 'raw' and data '{"data":[9,2,17,0,7,68],"type":"Buffer"}'
```

[TuyaSiren.zip](https://github.com/Koenkk/zigbee2mqtt/files/6585650/TuyaSiren.zip)

GrimDeclaw, 2021-06-24:
> Perhaps it waits for the result of command 0x24 with data 0006 (pairing with zigbee2mqtt - packet 233) wich is answered by the tuya hub (pairing with tuya - packets 406, 412) with 0x24 Data 003d60b0cbbf60b0e7df ???

Hi,
I'm not a ZIgbee expert neither, but I also own this v3 device, and I'd love to have it work!
According to this documentation : https://developer.tuya.com/en/docs/iot/tuuya-zigbee-door-lock-docking-access-standard?id=K9ik5898uzqrk

`TUYA_MCU_SYNC_TIME | 0x24 | Time synchronization (two-way)
`
which strangly refer to time as the log just before the first "no converter" message

>zigbee2mqtt             | Zigbee2MQTT:debug 2021-05-29 10:43:53: Received Zigbee message from '0x804b50fffe0482ae', type >'commandSetTimeRequest', cluster 'manuSpecificTuya', data '{"payloadSize":1536}' from endpoint 1 with groupID 0

>zigbee2mqtt             | Zigbee2MQTT:debug 2021-05-29 10:43:53: No converter available for 'NAS-AB02B0' with cluster >manuSpecificTuya' and type 'commandSetTimeRequest' and data '{"payloadSize":1536}

So this looks like a time issue (or not... one again, not a Zigbee expert).
I wonder if a "fz.ignore_tuya_set_time" in the fromZigbee device definition would help (the neo.js one doesn't have it)

IMHO, the issue is in the converter, not in zigbee2mqtt

GrimDeclaw, 2021-06-25:
The more I try to figure this out, the more the "raw" data triggers me!

@BerndK this might be usefull reading : https://medium.com/@dzegarra/zigbee2mqtt-how-to-add-support-for-a-new-tuya-based-device-part-2-5492707e882d
This post is also interesting, it's the one that lead to the writing of the previous link : https://community.home-assistant.io/t/i-am-totally-noob-with-zigbee-and-tuya-and-i-need-advice-for-smart-thermostatic-radiator-valves/163607/18 and describe all the way from the sniffing to the addition of the device to zigbee2mqtt (the converter actually)

I'll try to understand the zigbee sniffing you did but not exactly used to!

GrimDeclaw, 2021-06-25:
I really have hard time understanding that Zigbee protocol!

I thought that the Sequence number was used to identify request/response but in the Tuya gateway sniffing, for the 0x24 command, the "request" Seq is 99 (packet 406), and the response one is "71" (packet 412)

Another "weird" thing (for me... pretty sure this is not so weird) is the "never responded" 0x10 command (packet 459)
And we also have 0x0b command (OTA? https://developer.tuya.com/en/docs/iot/tuya-zigbee-module-uart-communication-protocol?id=K9ear5khsqoty#title-23-Data%20format%20of%20requesting%20OTA%20version) that is just responded with "ZCL: Default Response" (Seq 98 packet 372 / 378 and Seq 100 packet 463 / 467)

So ... I'm a bit lost ... my guess is that the 0x24 can be ignored (like fz.ignore_tuya_set_time might), is it the OTA command missing (I think I tried to add it, but got error message something like "type of undefined") or is it something else completly different... I don't know ....

BTW, still trying to figure out the "raw data [9,2,17,0,7,68]" which I wasn't able to find in the z2mqtt sniff

GrimDeclaw, 2021-06-25:
I might have something for that "raw" thing!!
My guess is that this is the reading of our missing data!

Let's look at packet 244 for exemple:
`09 54 02 00 55 72 01 00 01 00
`
Here is our 9 of our raw data (0x09). 0x54 is the sequence Number (0x54=>84). 0x02 is the command and we have 7 bytes of "data" : 00 55 72 01 00 01 00

So, according to @dzegarra in is article part 2 that I posted earlier:
0x00 => Status
0x55 => transID
Here is a little change, DP seems to be only uint8 (maybe because it is what I need to see :D )
0x72=> Corresponding to DP 114, which if we look at the lib/tuya.js file is neoHumidityAlarm

With the same reasoning, packet 260 might be related to DP 0x71 => 113 : neoTempAlarm

Now, even with this findings... I don't know how to try to implement that behaviour in the neo.js device file (ie using the 0x09 "I don't know what" to read measured data instead of "I don't know what")

If this might help someone....

GrimDeclaw, 2021-06-25:
Do go deeper, if reading are done on the 0x09 "I don't know what", it looks like the writing (settings) are done in the 0x11 "I don't know what"

Looking at the "Alarm On/off" sniffing:
Packet 5 : Command 0x00, 0x11 Frame Control Field, Data : 00 07 68 01 00 01 01
So, we write 01 (ON) on the 0x68 DP (104 => neoAlarm)
Packet 21 : Command 0x02, 0x09 FCF, Data : 00 17 68 01 00 01 01
So, we read 01 (ON) on the 0x68 DP

Packet 32 : Same as Packet 5, but with Data : 00 08 68 01 00 01 00
This time, we set Alarm (0x68) off (last 0x00)
Packet 36: Same as Packet 21, Data: 00 08 68 01 00 01 00
Read 0FF (last 0x00)

All others 0x02 commands  are the readings of the other values:
57,68,78,95,105 : 0x6A => 106 => neoHumidity
118,133 : 0x69 => 105 => neoTemp

GrimDeclaw, 2021-06-27:
Well... I tried various things, but none gives results... :(

I think that I have successfully added the "Writes" from packet 174

```
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Basic (0x0000)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 19
ZigBee Cluster Library Frame, Command: Write Attributes, Seq: 63
    Frame Control Field: Profile-wide (0x10)
        .... ..00 = Frame Type: Profile-wide (0x0)
        .... .0.. = Manufacturer Specific: False
        .... 0... = Direction: Client to Server
        ...1 .... = Disable Default Response: True
    Sequence Number: 63
    Command: Write Attributes (0x02)
    Attribute Field, Uint8: 19
        Attribute: Unknown (0xffde)
        Data Type: 8-Bit Unsigned Integer (0x20)
        Uint8: 19 (0x13)
```
with a configure block:
```
configure: async (device, coordinatorEndpoint, logger) => {
            logger.debug("Grim configure Neo");
            const options = {
                srcEndpoint:1
            };
            const payload = {0xffde: {value:0x13,type:0x20}}
            logger.debug("Grim configure Neo await Write");
            await device.getEndpoint(1).write(0x0000, payload,options);
            logger.debug("configure Neo await Write done");
}
```
But this doesn't seems to change anything (device still keeps blinking) ...
I found another write in your sniffing (packet 210), but can't find how to implement this, as it is "Unknown" (don't know how to write the payload)

```
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Unknown (0xef00)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 184
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x09)
        .... ..01 = Frame Type: Cluster-specific (0x1)
        .... .0.. = Manufacturer Specific: False
        .... 1... = Direction: Server to Client
        ...0 .... = Disable Default Response: False
    Sequence Number: 82
    Command: Unknown (0x02)
Data (7 bytes)
    Data: 00537404000102
    [Length: 7]

```

I tried 
```
 const optionstwo={
        srcEndpoint:1,
        direction:1,
        manufacturerCode:4098,
        disableDefaultResponse:false
}
  const paytwo={
      0x5300: {
              value:[0x74,0x04,0x00,0x01,0x02],
              type:0x41
      }
  }
await device.getEndpoint(1).write(0xef00,paytwo,optionstwo);
```
But only get "Timeout", and looking at the others look-a-like packet (232 for exemple) 0x00, 0x53 become 0x00 0x54 so i must be some kind of counter, not identifier....

Enjoy-Combi, 2021-06-29:
Hi,

I don't have a tuya hub, did someone tried the procedure described here ?: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

esion-net, 2021-06-29:
@Enjoy-Combi 
I have tried but did not succeed. I've tried to debug the tutorial also but it seems like it's outdated (or I totally misunderstood something).

Enjoy-Combi, 2021-06-29:
Pity, I would enjoy to try but have to find a gateway.

GrimDeclaw, 2021-06-30:
Forget about packet 210, it's the device sending this packet, not the hub... I'm trying to implement a "defaultResponse" to this, but still no luck

BerndK, 2021-07-31:
this workaround worked for me (see also discussion here #7051):
1 hold button, blinking
2 pairing with the tuya gateway (wireless one) (stopped blinking)
3 remove siren from gateway app (started blinking)
4 it automatically paired with my z2m stick (stopped blinking)
and it worked as expected: reported data and was able to activate the alarm

BerndK, 2021-07-31:
Closing as is duplicate of #7051

closed 2021-07-31 by BerndK

zedmaster, 2021-08-09:
Can anyone give me the way to learn how to develop the solution to this problem?

I have this siren standing here, and I want to help fix the problem.

Any idea how to get started?

mrTuomoK, 2021-08-15:
I'd like to buy a good Tuya gateway to debug stuff (and get more devices to work with zigbee2mqtt) so could someone here point me one?

joreung, 2021-08-21:
@mrTuomoK , as this thread is closed, I suggest you have a look at the openned one [7051](https://github.com/Koenkk/zigbee2mqtt/issues/7051#issuecomment-844870519) where you will find a proposition to your question. It will be a good help if you can solve our problem connecting to this device NAS-AB02B0