Koenkk/zigbee2mqtt #7412: Support for Sunricher "HK-ZD-CCT-A" aka LED-Trading.de "Driver 9128"

tjwinter75, 2021-05-10:
Hi together

I am actually at beginners level with zigbee2mqtt, unfortuneately i had some "bad luck" choosing an unsupported constant current driver for my CCT panel, even it's basicly a very interesting device.

Device is a Sunricher OEM, sold by led-trading.de as "Driver 9128" 
https://www.led-trading.de/zigbee-cct-dimmer-constant-current-konstantstrom-panel

They confirmed it's a Sunricher product. Probably a 'SRP-ZG9105-50CC' (unconfirmed, but all data are matching).
https://www.sunricher.com/50w-constant-current-zigbee-led-dimmable-driver-srp-zg9105-50w-cc.html
Manual, with cluster description:
https://www.sunricher.com/media/resources/manual/SRP-ZG9105-50CC%20instruction%20V1.pdf

Output at iobroker.zigbee is following:
(Sorry me that i use that project fordisplaying output; how do i same simple way at zigbee2mqtt?)

**Pairing:**
Pairing time left: 36
Error: Device 0xec1bbdfffe9d7a5c "HK-ZD-CCT-A" not described in statesMapping.. undefined
Pairing time left: 37
New device joined '0xec1bbdfffe9d7a5c' model HK-ZD-CCT-A
Interview successful: [object Object]
Pairing time left: 38
Pairing time left: 39
Pairing time left: 40
Pairing time left: 41
Pairing time left: 42
Pairing time left: 43
Pairing time left: 44
Pairing time left: 45
Pairing time left: 46
Pairing time left: 47
Pairing time left: 48
Pairing time left: 49
Pairing time left: 50
Pairing time left: 51
Device '0xec1bbdfffe9d7a5c' announced itself
Interview started: 0xec1bbdfffe9d7a5c
Pairing time left: 52

**Device:**

    HK-ZD-CCT-A
    ieee:0xec1bbdfffe9d7a5c
    nwk:2101 (0x835)
    model:HK-ZD-CCT-A
    groups:

**Device informations:**

    modelZigbee:HK-ZD-CCT-A
    type:Router
    nwk:2101
    manuf id:4107
    manufacturer:Sunricher
    power:Mains (single phase)
    app version:0
    hard version:1
    zcl version:3
    stack version:0
    date code:NULL
    build:2.5.3_r51
    interviewed:true
    configured:false

    endpoint:1
    profile:260
    input clusters:genBasic (0)
    genIdentify (3)
    genGroups (4)
    genScenes (5)
    genOnOff (6)
    genLevelCtrl (8)
    lightingColorCtrl (768)
    haDiagnostic (2821)
    touchlink (4096)
    output clusters:genOta (25)

    endpoint:242
    profile:41440
    input clusters:greenPower (33)
    output clusters:greenPower (33)

Thank you very much for your support. Do i have to give any additional informations?

tjwinter75, 2021-05-14:
& here database.db

{"id":4,"type":"Router","ieeeAddr":"0xec1bbdfffe9d7a5c","nwkAddr":9007,"manufId":4107,"manufName":"Sunricher","powerSource":"Mains (single phase)","modelId":"HK-ZD-CCT-A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"HK-ZD-CCT-A","manufacturerName":"Sunricher","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.5.3_r51"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.5.3_r51","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1621023561512}

tjwinter75, 2021-05-23:
Even i had troubles to adapt "How to support new devices", follow code start to work as my HK-ZD-CCT-A.js

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
        zigbeeModel: ['HK-ZD-CCT-A'],
        model: 'HK-ZD-CCT-A',
        vendor: 'Sunricher',
        description: '50W ZigBee CCT LED Driver(constant current)',
        extend: extend.light_onoff_brightness_colortemp(),
    },
];`

Koenkk, 2021-05-24:
@tjwinter75 if you provide the color temp range I can add this device: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#31-retrieving-color-temperature-range-only-required-for-lights-which-support-color-temperature

Koenkk, 2021-05-26:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-05-26 by Koenkk