Koenkk/zigbee2mqtt #7423: last_seen attribute is not updated in topic for some devices

Enzokot, 2021-05-10:
## What happened
Last_seen attribute is not updated in topic zigbee2mqtt for some devices, only updates in database.db.
For example: [TuYa TS0601_thermostat](https://www.zigbee2mqtt.io/devices/TS0601_thermostat.html) and [Moes MS-104Z](https://www.zigbee2mqtt.io/devices/MS-104Z.html)
[image]
## What did you expect to happen
Last_seen attribute in topic zigbee2mqtt is equal data from database.db
## How to reproduce it (minimal and precise)
Do not send any commands to device, just idle
## Debug info
Zigbee2MQTT version: 1.18.3
Adapter hardware: CC26X2R1
Adapter firmware version: zStack3x0 20210218

Enzokot, 2021-05-16:
and same with [Moes AM43-0.45/40-ES-EB](https://www.zigbee2mqtt.io/devices/TS0601_cover.html)

RedShuhart88, 2021-05-21:
https://www.zigbee2mqtt.io/devices/TS0115.html and https://www.zigbee2mqtt.io/devices/TS0601_dimmer.html

Enzokot, 2021-06-02:
The same with 1.19.0

Enzokot, 2021-07-01:
The same with 1.20.0

Koenkk, 2021-07-02:
The `last_seen` is only updated when a message MQTT is published. (it's not updated standalone)

Enzokot, 2021-07-02:
> The `last_seen` is only updated when a message MQTT is published. (it's not updated standalone)

But why another devices publish last_seen, minimal one time at hour. For example, Xiaomi MCCGQ01LM, but TuYa TS0601_thermostat save and update last only in database.db
If it is possible to make this device also report like some others. That would be great.

Koenkk, 2021-07-03:
Because a MQTT is published for this Zigbee message, if you can catch a `Received zigbee message..` without  a `MQTT publish` from this device I can add it. (requires debug log)

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

Enzokot, 2021-07-03:
I catch. Device name 'Left shades drive motor kitchen' and 'Bedroom valve with thermostat'
[example.txt](https://github.com/Koenkk/zigbee2mqtt/files/6758509/example.txt)

Koenkk, 2021-07-03:
Should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Enzokot, 2021-07-04:
The fix work for me. Thank you very much!

Enzokot, 2021-10-02:
@Koenkk the fix has been broken in the release `1.21.2`. Debug log in attachment. For example - `Right shades drive motor kitchen`
[example.txt](https://github.com/Koenkk/zigbee2mqtt/files/7271067/example.txt)

Koenkk, 2021-10-02:
Should be fixed now and be better than before. It will now rely on the internal last seen changed events which is also updated when e.g. device announces, changed network address or when a command to it is successfully send. 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

tteck, 2021-10-02:
groups don't change last seen.

Koenkk, 2021-10-03:
That is expected, with a group command you don't know if a device actually received it.

Enzokot, 2021-12-02:
@Koenkk after 1.22.1 release it's broken again. For example, TuYa TS0601_cover

Enzokot, 2021-12-02:
@Koenkk sorry, I forget about it.
For example `Right shades drive motor kitchen`
Debug log:
[example.txt](https://github.com/Koenkk/zigbee2mqtt/files/7643726/example.txt)

Koenkk, 2021-12-03:
Should be fixed again.

closed 2022-01-01 by Enzokot