Koenkk/zigbee2mqtt #7444: Livolo Zigbee Blind Motor

grinco, 2021-05-12:
I'd like to reopen https://github.com/Koenkk/zigbee2mqtt/issues/5180. The fact that the device is hardcoded to certain pan id and channel doesn't prevent me from using it in z2m. I've set up a second instance and paired the device with the following settings:
advanced:
```
  pan_id: 6756
  ext_pan_id:
    - 33
    - 117
    - 141
    - 25
    - 0
    - 75
    - 18
    - 0
  channel: 26
```
Unfortunately, every time I'm trying to send a request to device, it times out with the following error:
`error Publish 'set' 'state' to 'Curtain Motor' failed: 'Error: WriteStructured 0x00124b0022e13a39/6 genPowerCfg([{"attrId":0,"selector":null,"elementData":[4,0,0,0,0,0,0,0]}], {"timeout":10000,"disableResponse":true,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":3,"manufacturerCode":6866,"transactionSequenceNumber":233,"writeUndiv":true,"frameType":0}) failed (Data request failed with error: 'MAC transaction expired' (240))'`

According to [documentation](https://www.zigbee2mqtt.io/devices/TI0001-cover.html) the device is supposed to be fully supported, therefore I'm confused whether I should open a new device request or a bug report.

closed 2021-06-19 by github-actions[bot]

sean-lamont, 2022-09-18:
Did you manage to get this device working?

grinco, 2022-09-18:
No, its dead weight. There's something wrong with pairing, the device won't report states or receive commands once the pairing mode is disabled. Folks are saying that it needs to be paired to an official bridge first, which I don't have ...

grinco, 2023-02-21:
anyone managed to get this working without a livolo gateway?