Koenkk/zigbee2mqtt #7445: 'Data request failed...' after migrating to an ssd drive

alexryd, 2021-05-12:
## What happened
I've been using zigbee2mqtt on my Raspberry pi 4b with a zzh! dongle for a while now and it's been working great. Today I decided to migrate from the SD card to an external SSD drive connected to a usb3 port (which the rpi is now booting from). I've installed the exact same versions of Raspbian, z2m, mosquitto etc. and I've copied the `zigbee2mqtt/data` directory to the new drive. The zzh! is still connected to the other usb3 port (through an extension cable). Everything is working except that z2m can't control my devices. Whenever I try to e.g. set the state of a light the request fails with either `Data request failed with error: 'No network route' (205)` or `Data request failed with error: 'MAC channel access failure' (225)` (see [logs](https://pastebin.com/rcsfR2EC)). Updates from my motion sensor and switches are still coming in though and I can see announcements from my lights if I physically turn them off and then on again.

How can I debug this further?

## What did you expect to happen
My lights to turn on or off.

## How to reproduce it (minimal and precise)
I'm not sure.

## Debug info
Zigbee2MQTT version: 1.18.3
Adapter hardware: CC2652R
Adapter firmware version: 20210120

alexryd, 2021-05-12:
#6652 seems related

rihoattila, 2021-05-13:
Same config same problem.
Had a stable config with pi2B SD card zzh! dongle and BLE dongle.
Moved to pi4b USB3 SSD and extension cable on USB2 with zzh! dongle and BLE dongle.
Now I have problem with zigbee and BLE too.

Koenkk, 2021-05-13:
I've seen this previously, that person fixed it by connecting the SSD through a powered USB hub (and connect the ZZH via an extension cable and position it away from the pi/ssd)

rihoattila, 2021-05-14:
Made some experiments with external power supply, but no success.
However found some interesting topic on USB3 and 2.4GHz interference.
[forum](https://www.indilib.org/forum/general/6576-pi4-usb3-and-wireless-2-4ghz-interference.html?start=0)
[intel](https://www.intel.es/content/www/es/es/products/docs/io/universal-serial-bus/usb3-frequency-interference-paper.html)

alexryd, 2021-05-16:
Thanks @Koenkk and @rihoattila. I'm not sure why, but everything is working for me now. I reinstalled the OS, moved the zzh! to a usb2 port and moved my ssd to the other usb3 port, and suddenly I don't see these errors anymore 🥳 I agree @rihoattila that it seems more like some kind of interference with usb3. Hopefully you'll find a solution too.

rihoattila, 2021-05-16:
Made a modification too.
Plugged the SSD to USB2 and the error gone.

closed 2021-05-17 by alexryd