Koenkk/zigbee2mqtt #7446: Woox R7049 Smoke Alarm (Tuya)

mattleh, 2021-05-12:
since #6044 is closed

## Information about the device + link
Manufactuirer: WOOX
Model: R7049
Battery 2x AAA
Photoelectric smoke detector.
Alarm sound: 85dB
https://wooxhome.com/products-c10/security-c6/r7049-smoke-alarm-single-unit-p70

i used:
modelID: 'TS0601',
manufacturerName: '_TZE200_aycxwiau'
as fingerprint

i got following dump data (removed some duplicate dp) :
1620817683836 0xbc33acfffedb71f2 00 3d 01 04 00 00
1620817857303 0xbc33acfffedb71f2 00 3e 01 04 00 01
1620818043114 0xbc33acfffedb71f2 00 3f 08 01 00 01
1620820225750 0xbc33acfffedb71f2 00 02 0b 05 00 00
1620820225965 0xbc33acfffedb71f2 00 03 0e 04 00 02
1620822863992 0xbc33acfffedb71f2 00 03 0e 04 00 01
1620823078037 0xbc33acfffedb71f2 00 03 0e 04 00 00

with the python script i got:
2021-05-12T13:08:03 0xbc33acfffedb71f2 (tid: 61, s: 0) [unknown ] => 0 0
2021-05-12T13:10:57 0xbc33acfffedb71f2 (tid: 62, s: 0) [unknown ] => 0 1
2021-05-12T13:14:03 0xbc33acfffedb71f2 (tid: 63, s: 0) [wndow detect] => 0 True
2021-05-12T13:50:25 0xbc33acfffedb71f2 (tid: 2, s: 0) [unknown ] => 0 00000000
2021-05-12T13:50:25 0xbc33acfffedb71f2 (tid: 3, s: 0) [unknown ] => 0 2
2021-05-12T14:34:23 0xbc33acfffedb71f2 (tid: 3, s: 0) [unknown ] => 0 1
2021-05-12T14:37:58 0xbc33acfffedb71f2 (tid: 3, s: 0) [unknown ] => 0 0

the wndow detect is when the test button is pressed
the first unknown with value 0 1 (i think dp=1) means no smoke
the second unknown with value 0 0 (i think dp=1) is smoke detected
following messages come after removing inserting batteries
the unknown(dp=14?) with value is probably related to battery status it shows:
0 2 above 2,8v
0 1 above 2.6v
0 0 at 2.6v and below (beeps ocassionally)
The unknown with 0 00000000 (dp=11) hasnt changed value yet

can someone help me implement this device? is there any more data needed?

i cant see node_modules/zigbee-herdsman-converters/lib/tuya.js in the docker volume :(

mattleh, 2021-05-12:
update some additional received data:
2021-05-12T18:55:33.858338339Z Zigbee2MQTT:debug 2021-05-12 20:55:33: Received Zigbee message from '0xbc33acfffedb71f2', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID null
2021-05-12T18:55:39.263525732Z Zigbee2MQTT:debug 2021-05-12 20:55:39: Received Zigbee message from '0xbc33acfffedb71f2', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":1,"fn":0,"status":0,"transid":1}' from endpoint 1 with groupID null
2021-05-12T18:55:39.521503969Z Zigbee2MQTT:debug 2021-05-12 20:55:39: Received Zigbee message from '0xbc33acfffedb71f2', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0],"type":"Buffer"},"datatype":5,"dp":11,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID null
2021-05-12T18:55:39.731243271Z Zigbee2MQTT:debug 2021-05-12 20:55:39: Received Zigbee message from '0xbc33acfffedb71f2', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[2],"type":"Buffer"},"datatype":4,"dp":14,"fn":0,"status":0,"transid":3}' from endpoint 1 with groupID null
2021-05-12T18:55:39.849628213Z Zigbee2MQTT:debug 2021-05-12 20:55:39: Received Zigbee message from '0xbc33acfffedb71f2', type 'commandSetTimeRequest', cluster 'manuSpecificTuya', data '{"payloadSize":1024}' from endpoint 1 with groupID null
2021-05-12T18:55:59.975754931Z Zigbee2MQTT:debug 2021-05-12 20:55:59: Received Zigbee message from '0xbc33acfffedb71f2', type 'raw', cluster 'manuSpecificTuya', data '{"data":[9,5,17,0,6,128],"type":"Buffer"}' from endpoint 1 with groupID null

mattleh, 2021-05-12:
fz.tuya_smoke works and delivers smoke sensor data

Miswu13, 2021-05-15:
> fz.tuya_smoke works and delivers smoke sensor data
This means that now, zigbee2mqtt recognises it?
Can i make it work in Home Assistant with zigbeetomqtt?

mattleh, 2021-05-15:
> > fz.tuya_smoke works and delivers smoke sensor data
> 
> This means that now, zigbee2mqtt recognises it?
> Can i make it work in Home Assistant with zigbeetomqtt?

no what i meant is that it uses the same datapoint and value for smoke detection as https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/smart9.js smoke detector.

So just for the smoke detecting it would be easy to write a working converter.
But i havent found anything about the other datapoints and have no expirience with zigbee devices so i hope someone can help a little.

Bartole, 2021-06-03:
I have done some sniffing, but yet missing the skills to bring this to fruition, so maybe it inspires someone who can ;-)

Result in attached libreoffice calc doc with some assumptions. Hopefully someone can extend, verify of falsify this!
[sniffing_r7049.ods](https://github.com/Koenkk/zigbee2mqtt/files/6590674/sniffing_r7049.ods)

mattleh, 2021-06-07:
i created a basic converter (battery_low and smoke) and opened a pull request. 
[https://github.com/Koenkk/zigbee-herdsman-converters/pull/2669](url)
[https://github.com/Koenkk/zigbee2mqtt.io/pull/786](url)
lets see if it gets accepted

ankostis, 2021-06-07:
Thank you!

Shouldn't there be a sirene actuator somehow?

mattleh, 2021-06-07:
> Thank you!
> 
> Shouldn't there be a sirene actuator somehow?

I'm not sure but many Smoke alarms don't have the option to trigger the alarm remotely. And since I don't have the woox gateway and my experience is lacking my debugging options are limited...

if someone can provide the data point and payload to trigger the siren I think I could implement it

Bartole, 2021-06-07:
@mattleh, thanks for your efforts! I will give it a try and hopefully it get's accepted!

mattleh, 2021-06-08:
@ankostis  I have the smoke detector but i dont have the woox gateway. im using the deconz stick with the iobroker.zigbee module

Bartole, 2021-06-08:
The pull request is accepted, so that's great news :-)

Miswu13, 2021-06-08:
> @ankostis I have the smoke detector but i dont have the woox gateway. im using the deconz stick with the iobroker.zigbee module

I have woox R7049 Smoke alarm, and Woox R7070 gateway - if that can help in some way (also a conbee II stick with HA). What i don't have is knoledge :), only, if you tell me what to do with it, and how to extract the data that we need.

ankostis, 2021-06-08:
What happens when the smoke alarm goes off?
Is it possible to switch it off, remotely?

Bartole, 2021-06-09:
Also interesting if it is possible to remotely set off a detector so you can alarm all (or selected) detectors through domotica software when one detects smoke. I have not been able to find an option to do that yet. It is possible to remotely start a self-test.

ankostis, 2021-06-09:
What is the difference is a self-test from starting the siren?
Does it wear the hardware?

Bartole, 2021-06-09:
Apparantly the self-test stops the siren in matter of seconds and of course replies to the coordinator with some data. Wear is limited mostly to battery is my guess. I am not sure there even is an option to turn on the siren of a specific detector. There is no option in the Woox app to do that, but never tried what happens if one smoke detector is activated, maybe it does activate all other smoke detectors registered with the app, although I would like to have a choice about that.

mattleh, 2021-06-09:
I will close this issue since the device is now supported.

@Bartole @ankostis I don't think that it is possible to trigger the siren remotely (that would include smoke detectors triggering each other. if they can trigger each other it would be possible to trigger remotely ;) )
@Miswu13 thanks for the effort but I'm not very good at programming either and for me the implementation of the smoke and battery_low status is enough.

closed 2021-06-09 by mattleh