Koenkk/zigbee2mqtt #7451: [CC1352P-2] Coordinator failed to start, probably the panID is already in use after several attempts at re-flashing

Rathorian, 2021-05-13:
## What happened
Good morning,

For several days, I have been trying to set up my "Texas Instruments LAUNCHXL-CC1352P-2" coordinator but I keep getting the following error when launching Zigbee2MQTT:

`Error: Coordinator failed to start, probably the panID is already in use, try a different panID or channel`

I give you my configuration:

- Intel NUC with Proxmox with the coordinator plugged into a USB3 port (No USB2 port)
- A Debian Buster VM with the latest version of Docker + Docker-compose and the coordinator mounted on it.

**I would like to point out that this is a brand new installation and that I do not have any Zigbee hardware at home or any configuration file.**

I flashed the coordinator several times with the latest firmware found [at this address](https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_launchpad_coordinator_20210120.zip) using [this documentation](https://www.zigbee2mqtt.io/information/flashing_via_uniflash.html) on Windows by flashing through a USB2 and USB3 port, with no errors in return.

The coordinator is well mounted on my VM and I have access to it with the user who launches the docker-compose:
```
test -w /dev/ttyACM0 && echo success || echo failure
success
```

- [My docker-compose](https://pastebin.com/uGWG1jSv)
- [My configuration.yaml](https://pastebin.com/xG9RFbmV)
- [Output log](https://pastebin.com/yrhEBdMC)

I also tried to put in the "configuration.yaml" file:
```
network_key: GENERATE
```

But also by changing channel 15 and 20 but no change and still the error.

In my persistent "zigbee2mqtt/data" folder, I have no files at each launch since I delete them before launching the container except for the "configuration.yaml" file.

## What did you expect to happen

Zigbee2MQTT should launch without error

## How to reproduce it (minimal and precise)

By relaunching the command each time:
`docker-compose up -d zigbee2mqtt && docker-compose logs -f zigbee2mqtt`

## Debug info
Zigbee2MQTT version: 1.18.3
Adapter hardware: CC1352P-2
Adapter firmware version: CC1352P2_CC2652P_launchpad_coordinator_20210120.zip

sjorge, 2021-05-13:
Have you tried actually using a different panID? In my old apartment there were 2 networks because the neighbors were using philips hue setups. 

You can let z2m generate one by setting:
```
advanced:
  pan_id: GENERATE
```

Rathorian, 2021-05-13:
> 
> 
> Have you tried actually using a different panID? In my old apartment there were 2 networks because the neighbors were using philips hue setups.
> 
> You can let z2m generate one by setting:
> 
> ```
> advanced:
>   pan_id: GENERATE
> ```

Hello,

If you look at the "configuration.yaml" file that I linked, you can see that I am already using:
```
advanced:
  pan_id: GENERATE
```

:)

Rathorian, 2021-05-13:
It turns out that I did a new test going directly through a manual installation on a Raspberry and it works perfectly without errors.

I have to say that the problem is that using Proxmox + VM + Docker the adapter does not like.

Sorry for the unnecessary issue, in any case, it may be able to help people.

closed 2021-05-13 by Rathorian