Koenkk/zigbee2mqtt #7452: Failed to call 'Configure' 'onZigbeeEvent'

dexif, 2021-05-13:
## What happened
```
Zigbee2MQTT:debug 2021-05-13 10:49:01: Received Zigbee message from 'Kitchen wall switch', type 'read', cluster 'genTime', data '["time"]' from endpoint 1 with groupID 0
Zigbee2MQTT:error 2021-05-13 10:49:01: Failed to call 'Configure' 'onZigbeeEvent' (TypeError: zhc.getConfigureKey is not a function
    at Configure.shouldConfigure (/app/lib/extension/configure.js:46:59)
    at Configure.onZigbeeEvent (/app/lib/extension/configure.js:118:18)
    at Controller.callExtensionMethod (/app/lib/controller.js:384:44))
```
## Debug info
Zigbee2MQTT version: dev
Adapter hardware: CC2538
Adapter firmware version: latest

fludstank, 2021-05-15:
Getting the same error every few seconds.
From the debug logs:
`Failed to call 'Configure' 'onZigbeeEvent' (TypeError: zhc.getConfigureKey is not a function at Configure.shouldConfigure (/app/lib/extension/configure.js:46:59) at Configure.onZigbeeEvent (/app/lib/extension/configure.js:118:18) at Controller.callExtensionMethod (/app/lib/controller.js:384:44))
`

Koenkk, 2021-05-16:
Make sure you do not override the zigbee herdsman converters. (e.g. for the bare metal installation execute `npm ci` in the zigbee2mqtt folder). This is caused by using an incompatible zigbee-hersdman-converters with the latest zigbee2mqtt.

dexif, 2021-05-16:
Ahh yes!.. You are right. Sorry.

closed 2021-05-16 by dexif