Koenkk/zigbee2mqtt #7454: ZNLDP12LM power_outage_memory fails

grandchamp, 2021-05-13:
## What happened
I've got 3 Xiaomi ZNLDP12LM bulbs and checking the documentation I saw that the attribute `power_outage_memory` is available for them. I've tried to set this to `true` in the UI but I receive the following error:

`Publish 'set' 'power_outage_memory' to 'kitchen_light' failed: 'Error: Write 0x00158d0003e7f603/1 genBasic({"65305":{"value":1,"type":16}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4447,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

## What did you expect to happen
I expect the command to succeed.

## How to reproduce it (minimal and precise)
Go to Zigbee2Mqtt UI, click on the bulb, go to `Exposes` then click the `true` link on `power_outage_memory`.

## Debug info
Zigbee2MQTT version: 1.18.3 commit: f2e39af
Adapter hardware: zzh! cc2652r
Adapter firmware version: CC2652R_coordinator_20210120

polkovnichenko, 2021-05-18:
I have same problem.
In MQTT client I try to set power outage, publish in topic "zigbee2mqtt/Light Zigbee/set" json { "power_outage_memory": true } but see message in log: "No converter available for 'power_outage_memory' (true)" and power outage not work.  I have two lamp and also power outage not working.

**Device:** 
Zigbee model: lumi.light.aqcn02
Description: Aqara smart LED bulb
Firmware build date: 11-22-2018
Firmware version: 1.23
Vendor: Xiaomi
Model: ZNLDP12LM

**Z2M:**
Zigbee2MQTT version: 1.18.1 commit: 1.18.1
Coordinator type: zStack30x
Coordinator revision: 20190425
Frontend version: 0.3.76

And I also have question: attribute "effect" not working, maybe firmware too old or I don`t know how use this.

grandchamp, 2021-05-26:
@polkovnichenko and to whoever this can be useful.
The issue is because of the firmware version of the bulbs. Mine are version 1.23.
I've went to `OTA` in the web ui and updated one bulb to the latest version 1.32 and the property started working.
I'm closing the issue as the problem is solved by updating bulb's firmware.

closed 2021-05-26 by grandchamp