Koenkk/zigbee2mqtt #7462: Failed to interview '0x00124b0008cc9429', device has not successfully been paired

userX32, 2021-05-13:
# Bug Report
I get error message "Failed to interview '0x00124b0008cc9429', device has not successfully been paired" when try to pair new device.
Trouble only with this specific device. All other devices works fine.

Tried to pair the device again for multiple times.
Tried to replace battery with a new one.

After 'Filed to interview', the device sends status updates, which are logged like this:
`debug 2021-05-13 21:53:51: Received Zigbee message from '0x00124b0008cc9429', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0`
`warn  2021-05-13 21:53:51: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'`
`warn  2021-05-13 21:53:51: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.`

## Debug Info
Zigbee2MQTT version: 1.18.1-1
Adapter hardware: CC2531
Adapter firmware version: 20190425

Log with debug on:
https://pastebin.com/PsyrvKc9

russdan, 2021-05-15:
From my experience these are "range" issues, in that the device is too far away from the coordinator - certainly with a CC2531 you won't have a huge range - I notice that depending on how many devices also it can take a minute or more to go through the entire handshake from "device announced" until it completes.  Also some devices are **really** painful to get working - I have two Sonoff SNB-Z01 button switches and they take **forever** to get to properly pair - it can take 5 or 6 attempts - my conclusion is it must be the device as like you everything else pairs without problem.

What is the device?  Is it actually a supported device?

userX32, 2021-05-16:
> From my experience these are "range" issues, in that the device is too far away from the coordinator - certainly with a CC2531 you won't have a huge range - I notice that depending on how many [...]

Tried to pair at a very small distance (a few centimeters).

It's a door sensor, not supported. I need model ID to make external converter, but it doesn't show up in log because interview fails.

Koenkk, 2021-05-17:
@userX32 make sure to try on z2m 1.18.3, interviewing might succeed there.

userX32, 2021-05-17:
> @userX32 make sure to try on z2m 1.18.3, interviewing might succeed there.

Same problem with 1.18.3-1.

 Log (filtered by this device): https://pastebin.com/PJ5YFdVa

closed 2021-06-24 by github-actions[bot]

KPTitan, 2022-08-10:
Hi Guys,
Is this issue occurring because the sensor that is commissioned a sleepy device? After it commissioned, it may go into sleep mode and the coordinator expects it to respond. 

Just a though.