Koenkk/zigbee2mqtt #7463: Tuya plug TS011F unsupported

razmoket89, 2021-05-13:
Just bought a tuya plug, it's pairing fine but is unsupported. I read thé howto add à New device but it's way over my competencies.

Do you need anything to make it work ?

[image]

shvirik, 2021-05-13:
I think there are a lot of different device with the same external case. In this thread:

https://github.com/Koenkk/zigbee2mqtt/issues/7405 

we have similar issue. I bought 3, and 2 works as TS0121 plug and one as Unsupported TS011F. The boxes had the same device label code: A5C-16B7-08.

Koenkk, 2021-05-15:
Let's continue in #7405

closed 2021-05-15 by Koenkk