Koenkk/zigbee2mqtt #7468: UnhandledPromiseRejectionWarning showing up in logs

eriklogtenberg, 2021-05-14:
## What happened
After updating to latest version of Z2M, I am seeing these messages showing up in the logs:

    (node:8324) UnhandledPromiseRejectionWarning: Error: SREQ '--> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)')

    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
    (node:8324) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 3)

## What did you expect to happen
No errors

## How to reproduce it (minimal and precise)
Happens every 5-10 mins.

## Debug info
Zigbee2MQTT version: 1.18.3 commit: f2e39af1
Adapter hardware: CC2538+CC2592 made by H4nc
Adapter firmware version: zStack30x, 20201010

jrmax, 2021-05-19:
I have the same issue in the logs
```
(node:17) UnhandledPromiseRejectionWarning: Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":136,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,152,2,11,254,0]}}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)')
    at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
```

ov1d1u, 2021-06-22:
I'm having a similar issue with a CC2652RB stick, seems to only happens if I leave `permit_join` to `true`:

```
(node:329) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
(node:329) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 24)
```

It seems to happen after a few hours and the only way to fix it is to restart Zigbee2MQTT from Home Assistant's supervisor.

MattWestb, 2021-06-22:
`permit_join` is not allowed being stetted to true permanent in Zigbee 3 so if you is having one Zigbee 3 firmware and doing it it shall not doing it and making errors or the firmware is not behaving OK.

If running on HA1.2 firmware its OK but you dont have any security in your system then all can joining your network 24/7.

Koenkk, 2021-06-22:
@ov1d1u @MattWestb thats is not the reason why this happens.

`(node:329) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms` means the stick has crashed. If you are running an older firmware, there is a newer firmware available which fixes stability issues after a lot of uptime. https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin

ov1d1u, 2021-06-24:
I'm already on 20210120, should I try the version from the develop branch?

Koenkk, 2021-06-26:
@ov1d1u yes please

ov1d1u, 2021-07-04:
@Koenkk i know this is not related to this issue, but I just wanted to let you know that the firmware from the develop branch works great. Never ever my Zigbee network was so stable like it is now.

closed 2021-08-11 by github-actions[bot]