Koenkk/zigbee2mqtt #7470: Over zealous replacement of key containing endpointName more than once

russdan, 2021-05-14:
The Hive SLR2 device has an endpoint `occupied_heating_setpoint_heat` with the endpoint name `_heat` in the key - I noticed when accessing this endpoint an error is rasied showing `No converter available for 'occupieding_setpoint_heat' ()`.

I expected to see this instead as `occupied_heating_setpoint` and realised that the stripping of the `_heat `in` publish.js `was replacing the first `_heat`, not the last one - as issue which only occurs when the `endpointNameMatch `occurs more than once in the key.

Having spent some time debugging I propose the following code changes:

```
diff --git a/lib/extension/publish.js b/lib/extension/publish.js
index 59f6834e..52827de4 100644
--- a/lib/extension/publish.js
+++ b/lib/extension/publish.js
@@ -169,7 +169,8 @@ class EntityPublish extends Extension {
                 const endpointNameMatch = utils.getEndpointNames().find((n) => key.endsWith(`_${n}`));
                 if (endpointNameMatch) {
                     endpointName = endpointNameMatch;
-                    key = key.replace(`_${endpointNameMatch}`, '');
+                    var regex = new RegExp("(\_" + endpointNameMatch + ")$", "gm");
+                    key = key.replace(regex, ``);

                     const device = target.getDevice();
                     actualTarget = device.getEndpoint(definition.endpoint(device)[endpointName]);
```

this changes the regex to match the "end" occurrence - I've tested it with a couple of examples (my debug code now removed from the diff above):
```
error 2021-05-14 11:37:13: key: occupied_heating_setpoint_heat
error 2021-05-14 11:37:13: endpointNameMatch 'heat' (heat)
error 2021-05-14 11:37:13: Before replace 'occupied_heating_setpoint_heat' (occupied_heating_setpoint_heat)
error 2021-05-14 11:37:13: After replace 'occupied_heating_setpoint' (occupied_heating_setpoint)
error 2021-05-14 11:37:26: key: occupied_heating_setpoint_water
error 2021-05-14 11:37:26: endpointNameMatch 'water' (water)
error 2021-05-14 11:37:26: Before replace 'occupied_heating_setpoint_water' (occupied_heating_setpoint_water)
error 2021-05-14 11:37:26: After replace 'occupied_heating_setpoint' (occupied_heating_setpoint)
```

## Debug info
Zigbee2MQTT version: v1.18.3
Adapter hardware: CC26X2R1
Adapter firmware version: 20210128

Koenkk, 2021-05-15:
Lol nice bug, could you make a PR? (click the pencil button here to do so: https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/publish.js)

Koenkk, 2021-05-16:
Merged, thanks!

closed 2021-05-16 by Koenkk