Koenkk/zigbee2mqtt #7473: Philips Dimmer RWL021 - actions using '_' (underscore)

Geoff571, 2021-05-14:
## What happened
Action string from button presses were being returned with an underscore instead of a hyphen, e.g. expected to be 'on-press', actual 'on_press'

## What did you expect to happen
Expected strings to be as documented, using '-'

## How to reproduce it (minimal and precise)
The string being sent depends on different dimmer units, I have 2, one of them works as per specification ('-'), the other returned '_'. It appears the may be variations in the firmware handling, however on my dashboard, both are showing the same versions. 

## Debug info
Zigbee2MQTT version: 1.18-3
Adapter hardware: CC26X2R1 - zzh stick
Adapter firmware version: 

I can provide more info if required.

Geoff571, 2021-05-17:
I have done some further testing and out of the 3 different RWL21 dimmers that I have, 2 or the 3 are sending actions with '_' (underscore) in the string rather than the '-' (hyphen) as documented. They're all the same model and firmware.

gdave321, 2021-05-19:
See issue #6235

Geoff571, 2021-05-19:
Thank you guys, I checked my devices.yaml and I'd missed `legacy: false` off one of the devices, I never checked that as I thought the setting only affected the `click` exposure.

Geoff571, 2021-05-19:
Reopening this ticket as I guess the Z2B documentation needs to be updated to highlight the different separating characters being used depending on the `legacy:` support setting, currently states the legacy action strings only.

closed 2021-06-26 by github-actions[bot]