Koenkk/zigbee2mqtt #7474: Luminosity | Humidity | Temperature sensor (Moes House)

DavidFFerreira, 2021-05-14:
[image]

Zigbee model 'TS0222' and manufacturer name '_TYZB01_kvwjujy9'

debug Received Zigbee message from '0x5c0272fffe0a4792' type 'attributeReport' cluster 'genPowerCfg' data '{"batteryPercentageRemaining":200}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x5c0272fffe0a4792' type 'attributeReport' cluster 'msIlluminanceMeasurement' data '{"measuredValue":22254}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x5c0272fffe0a4792' type 'attributeReport' cluster 'msRelativeHumidity' data '{"measuredValue":4132}' from endpoint 2 with groupID 0
debug Received Zigbee message from '0x5c0272fffe0a4792' type 'attributeReport' cluster 'msTemperatureMeasurement' data '{"measuredValue":2741}' from endpoint 2 with groupID 0

DavidFFerreira, 2021-05-18:
up

DavidFFerreira, 2021-05-19:
up

Koenkk, 2021-05-20:
Same as  https://github.com/Koenkk/zigbee2mqtt/issues/7535, supported in the dev branch now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-05-20 by Koenkk

DavidFFerreira, 2021-05-24:
it stills dont work, I have the edge in my HA. I did a force remote. Take out the battery and wait a few minuts. After try to pair again, I got this log:
`Info Successfully interviewed '0x5c0272fffe0a438d', device has successfully been paired
Warning Device '0x5c0272fffe0a438d' with Zigbee model 'TS0222' and manufacturer name '_TYZB01_kvwjujy9' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x5c0272fffe0a438d","ieee_address":"0x5c0272fffe0a438d","status":"successful","supported":false},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x5c0272fffe0a438d","supported":false},"type":"pairing"}'
Warning Received message from unsupported device with Zigbee model 'TS0222' and manufacturer name '_TYZB01_kvwjujy9'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.`

Koenkk, 2021-05-25:
@DavidFFerreira then you are probably using the wrong repo, has been changed to https://github.com/zigbee2mqtt/hassio-zigbee2mqtt#installation