Koenkk/zigbee2mqtt #7484: Can't drive the curtain motor

alex1iam, 2021-05-15:
Hello
Can't drive the curtain motor
https://aliexpress.ru/item/1005001698400295.html
Called AM43-0.45/40-ES-EZ(TY), manufactured by Moes
There is a similar one in the repository under the name TuYa TS0601_cover, but it looks a little different, like this
https://zigbee.blakadder.com/Moes_AM43-0_45_40-ES-EB.html

Pairing was successful:
MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Curtain motor/roller blind motor/window pusher/tubular motor","exposes":[{"features":[{"access":3,"name":"state","property":"state","type":"binary","value_off":"CLOSE","value_on":"OPEN"},{"access":3,"description":"Position of this cover","name":"position","property":"position","type":"numeric","value_max":100,"value_min":0}],"type":"cover"},{"features":[{"access":3,"description":"Motor speed","name":"motor_speed","property":"motor_speed","type":"numeric","value_max":255,"value_min":0}],"name":"options","property":"options","type":"composite"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0601_cover","supports_ota":false,"vendor":"TuYa"},"friendly_name":"0x847127fffed08e8a_curtain_motor1","ieee_address":"0x847127fffed08e8a","status":"successful","supported":true},"type":"device_interview"}'
May 16 01:30:10 alex1rasp npm[2146]: #033[32mZigbee2MQTT:info #033[39m 2021-05-16 01:30:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Curtain motor/roller blind motor/window pusher/tubular motor","friendly_name":"0x847127fffed08e8a_curtain_motor1","model":"TS0601_cover","supported":true,"vendor":"TuYa"},"type":"pairing"}'

when you click on the buttons, it gives only this:
#033[32mZigbee2MQTT:info #033[39m 2021-05-16 01:53:13: MQTT publish: topic 'zigbee2mqtt/0x847127fffed08e8a_curtain_motor1', payload '{"linkquality":49,"motor_speed":40}'

when trying to use mqtt:
mosquitto_pub -d -h localhost -t zigbee2mqtt/0x847127fffed08e8a/set -m {"motor_speed":100} -u al***** -P *****

#033[31mZigbee2MQTT:error#033: Invalid JSON '{motor_speed:100}', skipping...

What am I doing wrong and what mqtt topics for the device should I use?

robertalexa, 2021-05-16:
Motor Speed only sets the motor speed but does not move the motor. Not sure if that is what you are trying to do.

Set the Motor Speed to 40
Set Position to whatever. The blind should move with speed 40

Set the Motor Speed to 100
Set Position to whatever. The blind should now move faster.

Hope this helps

alex1iam, 2021-05-17:
what structure of the mqtt topic should be in order to start the motor?
mosquitto_pub -d -h localhost -t zigbee2mqtt/0x847127fffed08e8a/set ????????????????

Faustino-ai, 2021-05-17:
> what structure of the mqtt topic should be in order to start the motor?
> mosquitto_pub -d -h localhost -t zigbee2mqtt/0x847127fffed08e8a/set ????????????????

{"state": "OPEN"} or {"state": "CLOSE"}

robertalexa, 2021-05-17:
https://www.zigbee2mqtt.io/devices/TS0601_cover.html

Cover #
The current state of this cover is in the published state under the state property (value is OPEN or CLOSE). To control this cover publish a message to topic zigbee2mqtt/FRIENDLY_NAME/set with payload {"state": "OPEN"} or {"state": "CLOSE"}. To read the current state of this cover publish a message to topic zigbee2mqtt/FRIENDLY_NAME/get with payload {"state": ""}. To change the position publish a message to topic zigbee2mqtt/FRIENDLY_NAME/set with payload {"position": VALUE} where VALUE is a number between 0 and 100.

alex1iam, 2021-05-17:
got it right. full command is:
mosquitto_pub -d -h localhost -t zigbee2mqtt/0x847127fffed08e8a/set -m '{"state":"CLOSE"}' -u al**** -P ****
be sure to put ' on both sides of payload

alex1iam, 2021-05-17:
In short, by defaults, the motor works with commands {"state":"OPEN/STOP/CLOSE"}. After the upper and lower limits are set on the device itself, the commands will become available {"position":VALUE}.

robertalexa, 2021-05-17:
That is all correct :)

I think you have rushed to pairing this device before actually setting the device on your window. Doing the limits first ensures the devices works as it is, with or without wireless data transmission.

After that, you pair it and everything is good :)

Here is a screenshot of my device in Z2M frontend
[image]

closed 2021-06-24 by github-actions[bot]

bachoo786, 2021-07-09:
@robertalexa how do you pair it? there are no instructions on z2m page.

alex1iam, 2021-07-09:
what exactly interests? how did you pair or what commands did you use to control?

bachoo786, 2021-07-09:
> what exactly interests? how did you pair or what commands did you use to control?

I am not sure if that is for me? but like I asked how did you pair the motor to z2m?

bachoo786, 2021-07-12:
the open and close buttons on z2m gui dont work. The only way to close and open the blind motor is by setting the value. Is that correct? 

on homeassistant, there isn't a switch for open and close, when I paired the blind motor.

Can you guys please confirm?

Faustino-ai, 2021-07-19:
You probably need to calibrate your blind first before you can use the open and close button

bachoo786, 2021-07-19:
> You probably need to calibrate your blind first before you can use the open and close button

I have calibrated it before adding it to z2m. However all I get under action in z2m gui is _empty_string_

[image]

alex1iam, 2021-07-28:
By the way, I found out that it is not necessary to set a long value {"state": "OPEN"} in the topic. You can just send OPEN or CLOSE

bachoo786, 2021-08-24:
Hello all

do you guys experiencing any drop outs with the blind? My blind works fine for 3 days and then it just stops working until I repair with z2m again.

Any one experiencing this?

alex1iam, 2021-08-31:
By the way, really. What equipment are you using? I have RPi 4, RPiOS + usb whistle cc2531. I noticed that the parallel launch of the second whistle on the same channel greatly affects the operation of the main

robertalexa, 2021-08-31:
RPi 4, RPiOS. I used to have CC2531 and it was very unstable as I added more and more devices. I changed the CC2531 for https://www.tindie.com/products/electrolama/zzh-cc2652r-multiprotocol-rf-stick/#product-reviews and network is much more stable now.