Koenkk/zigbee2mqtt #7486: Zigbee (Tuya) Door Sensors

jeremysharp, 2021-05-15:
Trying to get these door sensors to work, following the howto for adding unsupported devices:

https://www.aliexpress.com/item/1005002348194512.html

But it doesn't give me any IDs to use in the logging:

`Zigbee2MQTT:warn 2021-05-04 13:46:33: Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'`

DavidFFerreira, 2021-05-17:
you need to debug to get some clusters info, without that, devs can do anything =)

So871, 2021-05-17:
I'm trying to understand this procedure, but is there a video or an "simplified" how to on how to add Unsupported Tuya devices ?

jeremysharp, 2021-05-17:
> you need to debug to get some clusters info, without that, devs can do anything =)

I'm just following https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html which suggests the output will include IDs, like this:

```
Zigbee2MQTT:info  2019-11-09T12:19:56: Successfully interviewed '0x00158d0001dc126a', device has successfully been paired
Zigbee2MQTT:warn  2019-11-09T12:19:56: Device '0x00158d0001dc126a' with Zigbee model 'lumi.sens' and manufacturer name 'some_name' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
```

...eg '0x00158d0001dc126a' where mine shows 'defined'. There's no mention of additional debug settings on that link?

jeremysharp, 2021-05-20:
What should I do given the unpopulated ID fields?

So871, 2021-05-23:
this is what i get in debug 

Debug Received Zigbee message from 'ZM-CG206', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":0}' from endpoint 1 with groupID null
Debug Received Zigbee message from 'ZM-CG206', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID null

robert-pajak, 2021-06-07:
I think this issue is a duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/7548 (these issues could be probably be merged)

mastertape, 2021-07-09:
Is there  a working Solution for this device ? Have it also here....

closed 2021-07-17 by github-actions[bot]