Koenkk/zigbee2mqtt #7489: Battery stays at 100% with Philipps Hue Motion Sensor

Jeoffreybauvin, 2021-05-15:
## What happened

I've been using a [Hue Motion Sensor](https://www.zigbee2mqtt.io/devices/9290012607.html) and the battery has been staying at 100% for 6 months of daily use.

## What did you expect to happen

I expected the battery indicator to be accurate.

## How to reproduce it (minimal and precise)

Just pair a Hue Motion Sensor, and wait.

## Debug info
Zigbee2MQTT version: 1.18.3
Adapter hardware: LaunchXL
Adapter firmware version: zStack3x0 20200805

See some debug.
[debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/6488144/debug.txt)

McGiverGim, 2021-05-16:
I've some presence sensors from other brands at 100% after more than one year. Others are at 91%. Are you sure that yours must be with a lower value? The variation in the battery is not lineal so maybe yours is ok.

sjorge, 2021-05-17:
I have 3 of these, the lowest is around 89% last I looked. After about 1.5y of usage.

The 2xAAA provide way more juice than the little cell batteries use by some other brands like Aqara.

closed 2021-06-17 by Jeoffreybauvin