Koenkk/zigbee2mqtt #7491: Philips Outdoor Motion Sensor no longer sending messages but does announce

Coder2012, 2021-05-16:
## What happened
The Philips Hue Outdoor sensor stopped sending messages

## What did you expect to happen
Device to function as it has for last 2 months

## How to reproduce it (minimal and precise)
Using my CC2531 all my devices work fine except this one. I can delete entry in configuration.yaml and it will pair without problem, says announced by doesn't send any messages.
Device has latest update installed.

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531
Adapter firmware version: 1.18.3

## Logs
https://pastebin.com/s8PKYWtf

sjorge, 2021-05-16:
Probably the same as https://github.com/Koenkk/zigbee2mqtt/issues/4694

once mine drop of the network all I see from them is occasional device announcements too.

Coder2012, 2021-05-16:
Hey so after a lot of reading in other issues. If I manually send a config message to the device, it starts working.

This will have to do for now.

panachoi, 2021-05-26:
After a failed attempt to migrate from my CC2531 to a CC2652, I'm seeing this issue with my Hue Motion sensor. Its rejoined the network, and I get messages:

```

info  2021-05-26 10:18:56: MQTT publish: topic 'zigbee2mqtt/HueMotion8/battery', payload '100'
info  2021-05-26 10:18:56: MQTT publish: topic 'zigbee2mqtt/HueMotion8/occupancy_timeout', payload '10'
info  2021-05-26 10:18:56: MQTT publish: topic 'zigbee2mqtt/HueMotion8/motion_sensitivity', payload 'high'
info  2021-05-26 10:18:56: MQTT publish: topic 'zigbee2mqtt/HueMotion8/update-state', payload 'idle'
info  2021-05-26 10:18:56: MQTT publish: topic 'zigbee2mqtt/HueMotion8/update_available', payload 'false'
info  2021-05-26 10:18:56: MQTT publish: topic 'zigbee2mqtt/HueMotion8/led_indication', payload 'true'
info  2021-05-26 10:18:56: MQTT publish: topic 'zigbee2mqtt/HueMotion8/last_seen', payload '2021-05-26T10:18:56+02:00'
info  2021-05-26 10:18:56: MQTT publish: topic `'zigbee2mqtt/HueMotion8/linkquality',` payload '68'

```

But its "sensors" never report any values. What config message did you send to it to get it reporting again ?

Coder2012, 2021-06-10:
Sorry for the delay I sent this topic:

`zigbee2mqtt/bridge/request/device/configure` 

with payload:

`{"id": "outdoor-motion-sensor"}` this is my friendly name in my config

closed 2021-07-18 by github-actions[bot]