Koenkk/zigbee2mqtt #7497: Add log file retention

deosrc, 2021-05-17:
## Is your feature request related to a problem? Please describe

I use Zigbee2mqtt in a container using a volume for the data directory. When the volume becomes full, zigbee2mqtt errors and crashes. This causes the container to enter a restart loop. Most of the storage seems to be used by log files with older files never being cleared and there doesn't seem to be any settings for removing old log files.

## Describe the solution you'd like

It would be great if there were an option to retain log files for x days and remove files from the log directory which are older than that. 

## Describe alternatives you've considered

For the moment I have turned off file logging but this means that I have no logs if the container restarts.

AlexKay88, 2021-05-21:
How many space is allocated by log files and how many log files do you see?
Pls. check https://www.zigbee2mqtt.io/information/configuration.html for 
```
advanced:
  log_rotation: true #(default: true)
```
By default log file rotates at 10MB and last 3 files are kept (#3280), so that you should have at least 40 MB available for logs.
You might also consider changing log_level to warn or error.
```
advanced:
 log_level: info #debug, info, warn, error (default: info)
```

deosrc, 2021-05-21:
I seem to have more than 3 files. Perhaps it is keeping 3 files per directory?

I was seeing directories going back to the beginning of the year. I'm just using default settings I think.

[image]

```bash
/app/data/log # ls -lah **
2021-05-09.19-49-42:
total 16M    
drwxr-xr-x    2 root     root        1.0K May 12 11:28 .
drwxr-xr-x    6 node     node        1.0K May 16 22:34 ..
-rw-r--r--    1 root     root        6.2M May 15 09:07 log.txt
-rw-r--r--    1 root     root        9.5M May 12 11:28 log1.txt

2021-05-15.09-08-33:
total 3M     
drwxr-xr-x    2 root     root        1.0K May 15 09:08 .
drwxr-xr-x    6 node     node        1.0K May 16 22:34 ..
-rw-r--r--    1 root     root        2.5M May 16 09:06 log.txt

2021-05-16.09-07-28:
total 638K   
drwxr-xr-x    2 root     root        1.0K May 16 09:07 .
drwxr-xr-x    6 node     node        1.0K May 16 22:34 ..
-rw-r--r--    1 root     root      634.9K May 16 13:52 log.txt

2021-05-16.22-30-20:
total 10M    
drwxr-xr-x    2 root     root        1.0K May 21 02:53 .
drwxr-xr-x    6 node     node        1.0K May 16 22:34 ..
-rw-r--r--    1 root     root      585.6K May 21 08:53 log.txt
-rw-r--r--    1 root     root        9.5M May 21 02:53 log1.txt
```

AlexKay88, 2021-05-21:
Right, looks like it keeps 3 files inside dir `/app/data/log/<Current_TimeStamp_At_Start_Of_Z2M>` when %TIMESTAMP% is resolved to current time during start, so that log dirs from previous starts are kept forever.
I am not the expert in that area, but maybe it's worth to try `/app/data/log` (without %TIMESTAMP%) and check if Z2M appends to old logs from previous starts?

deosrc, 2021-05-21:
I agree. I've updated my settings as you suggest and will check it later.

deosrc, 2021-06-04:
It's been running for a while now with the log directory set to `/app/data/log` and it seems to be working as expected. There are 3 log files. The older log files are at ~10MB.

Definitely seems that the setting to retain 3 files is only taking effect per directory.

closed 2021-07-12 by github-actions[bot]