Koenkk/zigbee2mqtt #7500: TS0601 - support for "windows_detection_params"

pawol, 2021-05-17:
## Information about the device + link
TS0601 with the latest firmware + CC2531 (the latest zstack12)

## data/database.db entry of the device
A lot of features of TS0601 is implemented now. I see that "window_detection_params" are not supported yet by Z2M

Sample of logs (it is possible to read params):
zigbee2mqtt/bridge/logging {"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/termostat_w_pokoju', payload '{\"auto_lock\":\"MANUAL\",\"away_mode\":\"OFF\",\"away_preset_days\":0,\"away_preset_temperature\":17,\"battery_low\":false,\"boost_time\":300,\"child_lock\":\"UNLOCK\",\"comfort_temperature\":21,\"current_heating_setpoint\":5,\"eco_temperature\":19,\"force\":\"normal\",\"holidays\":[{\"hour\":6,\"minute\":0,\"temperature\":20},{\"hour\":8,\"minute\":0,\"temperature\":21},{\"hour\":11,\"minute\":30,\"temperature\":21},{\"hour\":12,\"minute\":30,\"temperature\":21},{\"hour\":17,\"minute\":30,\"temperature\":21},{\"hour\":21,\"minute\":0,\"temperature\":19}],\"linkquality\":39,\"local_temperature\":18.5,\"local_temperature_calibration\":-2,\"max_temperature\":25,\"min_temperature\":5,\"position\":0,\"preset\":\"manual\",\"system_mode\":\"auto\",\"update\":{\"state\":\"idle\"},\"update_available\":false,\"week\":\"7\",\"window_detection\":\"ON\",\"window_detection_params\":{\"minutes\":115,\"temperature\":94},\"workdays\":[{\"hour\":6,\"minute\":0,\"temperature\":20},{\"hour\":8,\"minute\":0,\"temperature\":21},{\"hour\":11,\"minute\":30,\"temperature\":21},{\"hour\":12,\"minute\":30,\"temperature\":21},{\"hour\":17,\"minute\":30,\"temperature\":21},{\"hour\":21,\"minute\":0,\"temperature\":19}]}'"}
zigbee2mqtt/termostat_w_pokoju {"auto_lock":"MANUAL","away_mode":"OFF","away_preset_days":0,"away_preset_temperature":17,"battery_low":false,"boost_time":300,"child_lock":"UNLOCK","comfort_temperature":21,"current_heating_setpoint":5,"eco_temperature":19,"force":"normal","holidays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":21},{"hour":11,"minute":30,"temperature":21},{"hour":12,"minute":30,"temperature":21},{"hour":17,"minute":30,"temperature":21},{"hour":21,"minute":0,"temperature":19}],"linkquality":39,"local_temperature":18.5,"local_temperature_calibration":-2,"max_temperature":25,"min_temperature":5,"position":0,"preset":"manual","system_mode":"auto","update":{"state":"idle"},"update_available":false,"week":"7","window_detection":"ON","window_detection_params":{"minutes":115,"temperature":94},"workdays":[{"hour":6,"minute":0,"temperature":20},{"hour":8,"minute":0,"temperature":21},{"hour":11,"minute":30,"temperature":21},{"hour":12,"minute":30,"temperature":21},{"hour":17,"minute":30,"temperature":21},{"hour":21,"minute":0,"temperature":19}]}

wizu, 2021-05-18:
#6073

pawol, 2021-05-18:
https://github.com/Koenkk/zigbee2mqtt/issues/6073

closed 2021-06-26 by github-actions[bot]