Koenkk/zigbee2mqtt #7503: ZM-CG205 - Door/window sensor

elgerg, 2021-05-17:
Ive been to:
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

and am failing at the first step. I cant see where to get the device name and manufacturer IE:
with Zigbee model 'lumi.sens' and manufacturer name 'some_name'

can anyone point me in the right direction?

## Information about the device + link
Device is a door/window sensor (ZM-CG205)
https://www.aliexpress.com/item/1005002511620629.html?spm=a2g0s.9042311.0.0.2cbd4c4djUC20p

## data/database.db entry of the device
{"id":23,"type":"EndDevice","ieeeAddr":"0x00124b000f741021","nwkAddr":19910,"manufId":0,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":200}},"ssIasZone":{"attributes":{"zoneStatus":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1621268864026}

kimet83, 2021-06-07:
{"id":53,"type":"EndDevice","ieeeAddr":"0x00124b0014869b84","nwkAddr":44083,"manufId":0,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":200}},"ssIasZone":{"attributes":{"zoneStatus":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1623043428974}

maybe same device
https://ko.aliexpress.com/item/1005002565778355.html?spm=a2g0s.9042311.0.0.65554c4dhlgVCe

elgerg, 2021-06-14:
Anyone?

closed 2021-07-23 by github-actions[bot]

elgerg, 2021-07-28:
Has anyone else tried one of these?