Koenkk/zigbee2mqtt #7505: False Hue bulb

Tonio16, 2021-05-17:
## What happened
I paired a hue bulb tyme 9260018216 and it is regognized as a 8718699673147. I  an switch On or Off or used the effect. But the brightness or the binding don’t work.

## What did you expect to happen
To be regognized as 92y0018216 since it is a supported bulb.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version 1.18.3 commit: f2e39af1
Coordinator type EZSP v8
Coordinator revision 6.9.1.0 build 214
Frontend version 0.4.0

Antoine

Tonio16, 2021-05-24:
update: I paired the buld with a zigate instead of a EZSP.  I have the same behavior, ie. bulb is recognized as  8718699673147 though is a known problem which should be already solved.

Antoine

Koenkk, 2021-05-24:
Can you provide the data/database.db entry of your device?

Tonio16, 2021-05-25:
Hello

Here the database https://pastebin.com/L29jmjrL  (I hope that it’s ok to share it this way)

And the device info:
> Modèle 8718699673147
Documentation https://www.zigbee2mqtt.io/devices/8718699673147.html
IEEE address 0x0017880108217905
Fabricant 
Type Router
Alimentation Mains (single phase)
Topic zigbee2mqtt/0x0017880108217905

Antoine

Koenkk, 2021-05-26:
Device with address `0x0017880108217905` is a `RWL021` (Hue dimmer switch), not a bulb. I don't see another Philips device in your database.db

Tonio16, 2021-05-26:
Sorry, I forgot that the bulb was deleted from this database. Here the good one: 
https://pastebin.com/LbWJJaaH

Regards

Antoine

Koenkk, 2021-05-27:
The zigbeeModel of your bulb is `LWA001`, which we gave the model `8718699673147` to. (https://github.com/Koenkk/zigbee-herdsman-converters/blob/4af924199d6e6673012faf43c3882dee9f5de6c4/devices/philips.js#L408). 

Looks good because if I search for this model I get results: https://www.google.com/search?q=8718699673147 , for the 9260018216 I don't https://www.google.com/search?q=9260018216 . Note that this model is purely cosmetically and does not affect the functionality in any way.

Tonio16, 2021-05-27:
Ok. Thanks for the time your spent on. I tried today and except the bindings with the Hue dimmer, the other points look to be ok.

Tx

Antoine

closed 2021-05-27 by Tonio16