Koenkk/zigbee2mqtt #7507: Cant re-pair Ikea bulb after resetting.

S-tef, 2021-05-17:
## What happened

I switched to HomeAssistant OS after using the docker version for a while. 
I have used this with 4 Xiaomi battery powered switches, and an Ikea bulb (LED1842G3) and an CC2531 as coordinator.

I did not know I could migrate from another installation. So in HomeAssistant OS I tried to re-pair them all. The Xiaomi devices instantly re-paired. But the bulb did not.

I have tried multiple times resetting the bulb with different timeslots for on/off, I updated the CC2531 stick. But still nothing, I tried the TouchLink but that gave an empty adress. 

After switching my SD-card to the previous docker configuration. The bulb was gone, so trying to re-pair also did not work.
So I think it is problably a faulty bulb after resetting so many times. But maybe there is another thing I could try.

## What did you expect to happen

That the ikea bulb would re-pair after a few tries.

## How to reproduce it (minimal and precise)

Resetting the bulb as prescribed gives the same log results everytime.

## Debug info
Zigbee2MQTT version: latest
Adapter hardware: CC2531
Adapter firmware version: latest

LOGFILE: [https://pastebin.com/61wTzzau](url)

mcmatti, 2021-05-19:
If the light bulb does not give the feedback it is not reseted. I tried it a lot until it works, but now I have always success. Touchlink did not work for me too.

S-tef, 2021-05-24:
Thats kind off the problem, it is resetting
The bulb starts dimming bright/less bright a couple of times. And in that sequence I can see it communicating with the zigbee stick.

alexolivier, 2021-06-05:
I'm also struggling to re-pair ikea bulbs that were working before

S-tef, 2021-06-08:
I'm still having issues re-pairing that specific bulb. I ordered the Electrolama zig-a-zig-ah! (zzh!) on Tindie. Let's see if a brand new stick does the trick. Else the bulb is heading to recycling and im getting some new ones. 

But not knowing why is kind of frustrating.

closed 2021-07-17 by github-actions[bot]