Koenkk/zigbee2mqtt #7508: Remote Adapter connection dies after a while

zombielinux, 2021-05-17:
## What happened
After some unknown and variable time (days to hours), zigbee2mqtt loses connection with the remote adapter connected via ser2net. Restarting the zigbee2mqtt docker container causes the connection to begin working again with no changes on the remote side. 

I am running AdaptiveLighting on HomeAssistant, and at times, it can create substantial numbers of messages. 

## What did you expect to happen
Zigbee2mqtt to maintain the connection for the long term. 

## How to reproduce it (minimal and precise)
Set up ser2net with the slaesh stick as described in the documentation. Connect a sizable network (48 nodes). Wait.

## Debug info
Zigbee2MQTT version: Docker Latest
Adapter hardware: Slaesh
Adapter firmware version: Latest (Feb2021)

mulyonost, 2021-05-18:
I have similar issue, my CC2351 can only receive data (as I see in the history of home assistant), but can't send data (if I click to turn on light or whatever my automation is, it won't receive data). And after a while it shows error like this :
`Zigbee2MQTT:error 2021-05-18 16:03:58: Publish 'set' 'state' to '0x00158d000374b3c9' failed: 'Error: Command 0x00158d000374b3c9/3 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))'`

Sometimes, unplug the stick and restarting the Zigbee2Mqtt can temporarily solve the problem, but now I tried restarting and unplug the stick, even reflashing the stick, nothing solve the problem.

mulyonost, 2021-05-19:
Ok, I tried to reset everything including the home assistant (fresh install), and luckily it's working again, I don't know for how long, but it work.

zombielinux, 2021-05-24:
I found the ser2net options net-to-dev-bufsize and dev-to-net-bufsize, I suspect they operate as a buffer between the network connections and the stick on the ser2net host. 

Is there an outbound buffer somewhere that I can match in size? The default is 64 bytes, so I wonder if my large network is filling up the buffer, and getting cut off before the whole message can be transferred to the stick (limited to 115200baud after all, and the ser2net connection is 1gbit). I've set my net-to-dev-bufsize to 512 for now, we'll see if the problem reproduces.

closed 2021-07-01 by github-actions[bot]