Koenkk/zigbee2mqtt #7509: Hue Dimmer Switch drains battery badly

Wolk9, 2021-05-17:
## What happened

Changed batteries in two Hue Dimmer Switches only 5 days ago after they were empty pretty fast for the first time. The batteries drained to less than 3% (blinking in the overview). I was out of home for 4 days. So no extensive usage of the dimmer switches.
This happend after I detached the Dimmer Switches from the Hue Bridge and added them to my Zigbee2MQTT on Hassio.

## What did you expect to happen

I expected to have the same everlasting battery performance of the Hue Dimmer Switches as they are worked connected to the original Hue Bridge.

## How to reproduce it (minimal and precise)

I use brand new (not expired) Duracel 2450 battery cells, and this is now the second pair of batteries that are drained extensively.

## Debug info
Zigbee2MQTT version: 1.18.3-1
Adapter hardware: CC2652RB stick (Sleah) 
Adapter firmware version: latest

Wolk9, 2021-05-20:
Anyone else with this problem?

MattWestb, 2021-05-21:
My HUE Dimmer Switch is on the original battery for 3 years and was in deCONZ and now ZHA.
Sonoff ZBB with tasmota is having battery braining problems also with  the HUE switch but i think its coming from bad hardware (the SM-011 Zigbee module) in the ZBB.

My Philips HUE end devices is normally jumping around and shushing very bad parents but they is not draining batteries so fast but i have pity much routers they can using.

closed 2021-06-29 by github-actions[bot]

markmghali, 2021-08-10:
Yes I am also having this issue! I just connected my HUE dimmer yesterday and was at 78% and now is at 44% in less than 24 hours. This was not the case with deconz or the original Hue Hub.

Using Conbee II

Any fix for this?

MattWestb, 2021-08-10:
I think DE is trying making some changes in the Cor/RaspBee firmware but its not released yet that can helping keeping stable contact with direct connected sleeping end device (default response and Aps Ack in the firmware and not in the host system that is to slow for doing it good).

markmghali, 2021-08-10:
Ah Ok! So this would be a firmware update on the conbee II? Thank you for the helpful info

> I think DE is trying making some changes in the Cor/RaspBee firmware but its not released yet that can helping keeping stable contact with direct connected sleeping end device (default response and Aps Ack in the firmware and not in the host system that is to slow for doing it good).
> 
> Edit: deCONZ Enable APS ACK s for ZCL Default Response [#5173](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5173)

JRJSmith, 2022-05-04:
I've just seen the battery level for my testing Hue Dimmer switch drop from 65% to 1.2% in less than 24 hours.

Has anyone actually seen the resolution to this issue, please? If so, please advise how to resolve the issue.

Many thanks
Joe.

jan-gerard, 2023-03-10:
I seem to have the same problem. However, my dimmer switch is connected to the Hue hub. It suddenly started to drain a couple of weeks ago, and I have replaced the battery now 3 times. Before, it ran almost 2 years on a battery. 
I modified my switch a bit (2 years ago, so this was not related to the problem) but taking the PCB from the enclosure, cutting the PCB below the circuitry and soldering a single coin cell holder to it. Then it fitted into a wall switch socket (with 4 pulse switches) so it nicely integrated into the rest of the wall switches. Works excellent for 3 other dimmer switches, which still run on the 1st battery. Only this one suddenly went crazy and depleted.
I suspect that it's caused by too regular communication. However, I didn't change anything on my Hue config. And in principle, it shouldn't interfere with my other Zigbee network with all kinds of sensors and generic Zigbee dimmers. But maybe one of them switched channel by itself and now they do interfere?

Krabmania, 2023-10-16:
@jan-gerard I have the same problem. Did you find any solution?

popy2k14, 2023-11-03:
Exact same issue on one of 24x RWL021 hue dimmer switches.
At first i also thought the batterys are bad.
But it's the same on this particular switch with any battery.

Would be nice if there is an solution/workaround.

ppmg77, 2023-11-08:
Same here. I have the older version of the Hue Dimmer Switch.
Replaced the battery now 3x and again after the last time, 3 day ago, the battery is 0%.
I hope there will be a solution for this. :/

ikonia, 2023-11-10:
Also having this problem, original battery expired after multiple years, multiple battery replacements, different brands are lasting 1-2 days before hitting %1 battery warning

popy2k14, 2023-11-10:
I have now replaced the Noname batteries with Duracell ones 4 days ago and reseted the switch by pressing the setup button 10s and re-paired it.

Until now 100%, fingers crossed...

popy2k14, 2023-11-13:
7 days now and still @ 100%... will kepp you updated...

Wolk9, 2023-11-16:
> I have now replaced the Noname batteries with Duracell ones 4 days ago and reseted the switch by pressing the setup button 10s and re-paired it near the coordinator.
> 
> Until now 100%, fingers crossed...

> 7 days now and still @ 100%... will kepp you updated...

That sounds hopefull. I've drained (again) sinds my OP my 12th set of expensive Duracell 2450's. (i've 2 Hue dimmer switches).

popy2k14, 2023-11-16:
10 days now and still @ 100%...

My assumption is, that re-pairing it near the coordinator has done the trick!?

Will keep you updated...

popy2k14, 2023-11-20:
2 weeks and still @ 100%
So re-pairing it near the coordinator or the duracell did the trick.

ikonia, 2023-11-23:
I don't believe that repairing near the bridge is a solution, I've tried this twice now after your post, once with a Duracell battery and once with a Panasonic battery, on both occasions reset the switch, repaired and renamed/programmed with the button presses, put it back in the room it controls and 36 hours later battery is dead or close to dead

popy2k14, 2023-11-23:
Sorry that this doesn't work for you.
In my case, one of the two (re-pairing it near the coordinator or the duracell), did the trick.
Before that i have changed the battery multiple times in a few weeks, but with noname batterys.

popy2k14, 2023-11-23:
@Koenkk anything we can do to help identify the issue

popy2k14, 2023-12-05:
Sadly i did not looked at the parent (map) when the issue happened and now it's gone.

@ikonia @Wolk9 @ppmg77 @Krabmania and all others where the issue is still there:
Can you please tell us which device is the parent in your mesh?

PS.: in my case i have 99% hue devices

MattWestb, 2023-12-05:
First gen HUE sensors is also jumping around and using bad routers as parent in HUE systems but they is not complaining of it so old hard and software that is not working OK in all situations (still not Zigbee 3).

Krabmania, 2023-12-06:
Had same problem for more than six months. 
Tried everything, until I saw this post. I was inches from buying new contacts/bottoms. 

The solution is 100% the batteries.
I’ve have tried many different batteries, but there were all some cheap batteries from IKEA and others. 

Now I’m using ENERGIZER ULTIMATE LITHIUM CR2 (2032). 
100% battery for a week now. Before, with the other batteries, the battery level bounced up and down all the time, and the contact didn’t always work. 
Now they are back to normal!🥳

Chillzwinter, 2024-03-19:
My hue dimmer switch battery only seems to last 3 months.  I've changed 2 now. I measured the current pulled when the button is pressed which is less than 1 microamp. Maybe it chatters to the phillips hue hub too frequently (and draws more current while chatting - which I haven't measured). The battery housing says use a cr2032 (not cr2450). The cr2032 is designed to provide 210-230 mAh whereas the larger cr2450 can provide 620mAh (so maybe I'd get 9 months if the cr2450 at a diameter of 24.5mm could fit into the cr2032 hole at diameter of 20mm - but this will never happen.  I only use the switch as a 'toy' ( I bought for fun to see if it was useful in my office) and I only press the button about once or twice a month (as I usually use voice control to turn these lights on/off. The dimmer switch is located 3 meters from the philips hue hub. Not very happy that this switch 'presumably' draws so much current everyday. Maybe I'll solder it up to a 3volt regulator (as it sits next to my study desk power socket anyway). Pretty crap design though.

LukasdeBoer, 2024-12-11:
I have had issues for like the last year too where it would drain the battery extremely fast. I'm using the same batteries as previously, but I have: 
- removed the dimmer from my network
- allowed new device joining zigbee2mqtt, but only to the coordinator!
- it immediately rejoined my network

It now is connected directly to the coordinator and is still working. Seems a lot more snappy too. I will report back about how long it'll survive.

R-ens, 2024-12-16:
Any update already @LukasdeBoer? 

My problems started after migrating all my Hue devices to z2m. I changed my remotes so many times with cheaper cells (because I had them already). Which lasts with a maximum of 2 weeks.

On advice of one of the GitHub posts I changed mine into the expensive Varta cell. Conclusion... It died again within 2 weeks. 

What the heck is causing this?

LukasdeBoer, 2024-12-17:
> Any update already @LukasdeBoer?
>
> My problems started after migrating all my Hue devices to z2m. I changed my remotes so many times with cheaper cells (because I had them already). Which lasts with [...]

So everything was working completely fine until my weekly backup and update run yesterday morning. Now the Hue dimmer has dropped from the network and opening the network for new devices doesn't add it again, I'll need to reset it..

LukasdeBoer, 2024-12-23:
Update: the zigbee2mqtt restart dropped the Hue dimmer from the network. I tried to reset it this weekend but the battery was completely drained. Replaced it and it was just working again without reset/rejoin. I've now disabled the weekly restart of zigbee2mqtt to see if that'll make the situation a bit more stable.

Or perhaps the restart has nothing to do with the battery draining and the battery drained quickly and was just empty coincidentally during the weekly restart.

LukasdeBoer, 2025-01-04:
Aaaand it's gone again as of 2 days ago. It suddenly loses connection with the network it seems, as then you can still click the button but nothing happens but a red led blinks inside the remote. Perhaps it could also be that the battery is too low for it to function properly.

Daafip, 2025-01-08:
Having this issue too, will try better batteries and repairing.

nicopeet911, 2025-02-01:
Same here, still no solution anyone?
Using V1 and V2 hue dimmers, both are draining fast, I used the SONOFF Zigbee 3.0 USB Dongle Plus-P and switched to SLZB-06 this week but the battery drain remains :-( I only added these two hue dimmers over from the Hue bridges and both have this issue while being connected to the hue bridge they both lasted me for mutiple years, so it does seem to be a hardware issue. Both V1 and V2 never had any isssues with hue both drain on zigbee2mqtt with different co-ordinators?

blib-blub, 2025-02-28:
I also have the same problem. I am using 8 Hue Wall Switch modules. All in use for 2 years on DeConz then I migrated to Z2M in December. 

1 is still working perfectly fine
2 drain batteries in less than a week and are marked offline
5 have simply dissappeared from Z2M without any trace I can find. Only the broken HA Automations referencing them of course remain.

Is there anything I can check whats the difference between the working and draining ones?

smumriak, 2025-05-09:
One of mine started battery drain after its firmware was updated to 67.115.5

robvdl, 2026-04-25:
> why is this issue marked as 'stale' and 'closed'

Maybe it auto closes them after inactivity.

I found that mine eventually had actually updated on their own the next morning, but at the cost of completely draining the batteries in the process.