Koenkk/zigbee2mqtt #7510: Zigbee network nightmare

GSzabados, 2021-05-17:
## What happened

This is the weird thing, I really don't know. I have about 50 devices on my Zigbee network another 10-15 on Hue, and about another 10-15 on another Zigbee network (SmartThings). All on different channels. I have changed my previous CC2531 dongle to a CC2562P dongle recently to have a better and more reliable dongle, but the issues seems to start at that time. On the day, when I started to use the new dongle everything went haywire. Flashing lights, sockets turning on and off, etc... I couldn't really pinpoint to anywhere but the dongle. I have rebuilt the whole network with a set of completely new settings. Everything was working well for a while, then during the last few days weird things started to happen again.
A few IKEA bulbs, all of them white color or only dimmable bulbs started to turn on at the same time, plus a LIDL Zigbee LED strip. I though, I might have some power outages in the house, as they are all running old firmware, and they do turn on when the shortest power outage happens. The same happened today as well, but some of them didn't reconnect to the network. So I turned them on and off, but not all of them came back. So I reset those. And then started again, what has happened on the first day. The IKEA sockets are turning on and off, in the log they are announcing themselves multiple times, the bulbs are turning and going quick flashing like I would like to try to reset it, and indeed they reset themselves.
I tried to rule out any firmware issue, so I updated the sockets and the bulbs, but the sockets time to time were doing the same thing.  There is a pair of bulbs, what I couldn't even re-pair since the last issue. Every pairing ends up with either unable to pair at all, because not showing up, or the following error, interview failed. I can try anything, but the bulbs are not joining correctly. Indeed I tried to pair one of them to the SmartThings network, and joined without any issue.

Info Device '0x680ae2fffeb7961c' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x680ae2fffeb7961c","ieee_address":"0x680ae2fffeb7961c"},"type":"device_joined"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x680ae2fffeb7961c","ieee_address":"0x680ae2fffeb7961c","status":"started"},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x680ae2fffeb7961c"},"type":"pairing"}'
Error Failed to interview '0x680ae2fffeb7961c', device has not successfully been paired
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x680ae2fffeb7961c","ieee_address":"0x680ae2fffeb7961c","status":"failed"},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x680ae2fffeb7961c"},"type":"pairing"}'

## What did you expect to happen

Things just work without any flashing and resetting lights. The WAF is really high at the moment. The situation is like sitting on a barrel of gunpowder and lighting up a cigar meanwhile... No good... 

I cannot confirm what causing the lights to turn on, and what causing the sockets to start to change states back and forth. It hasn't happened before the C2652P dongle.

The bulbs not able to re-join or re-pair is another weird story. I have had troubles with IKEA bulbs before, but this is different, there is a repeater just less than 50 cm away from the bulb.

## How to reproduce it (minimal and precise)

No idea.

## Debug info
Zigbee2MQTT version: 1.18.3
Adapter hardware:  CC2652P Egony Stick V4 (Ebyte ver.)
Adapter firmware version: 20210319 (Egony firmware)

Krazy998, 2021-05-18:
While I cant directly help - I have over 45 devices of various mix and do not have the issues described.

Couple things to consider:

1. have you looked to ensure there is no overlapping wifi channels with your zigbee channel used?
2.  Are you using the version of firmware here: 
https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC1352P2_CC2652P_other_coordinator_20210120.zip

It seems to be a different version of firmware that you posted.

GSzabados, 2021-05-20:
Hi,

Thanks for the advice. I will have to look at it. Because I think, I have some interference issue, but the channels should not overlap as I can remember, but it looks like, that is really not the issue. It looks like that I have micro outages, really short ones. The IKEA plugs are really sensitive those.

I will need to find someone who can detect these and log them. One of the BlitzWolf plugs logs time to 0 for voltage. And there is another device as well, which reported some weird outage (not Zigbee).

Thanks for the advise!

The firmware from Egony has the amplifier enabled. The two bulbs has been re-paired since at another location, so interference definitely in the cards.

closed 2021-06-27 by github-actions[bot]

GSzabados, 2021-06-27:
Just a closing comment. At the end I've reflashed the dongle with Koenkk's firmware version and cleared the NVMe. Re-paired everything and all problem was solved after that.