Koenkk/zigbee2mqtt #7512: Eco.Dim-07 Zigbee Smart Dimmer Switch 200W LED

Krolik1988, 2021-05-18:
Eco.Dim-07 Zigbee Smart Dimmer Switch 200W LED (Router)
Type dimmer: R,C  Phase cut
Operation: Push / turn button

Input Cluster:
     - 0x0000: Basic
     - 0x0003: Identify
     - 0x0004: Groups
     - 0x0005: Scenes
     - 0x0006: On/off
     - 0x0008: Level Control
     - 0x0b05: Diagnostics
Output Cluster:
     - 0x0019: OTA
     
 [Manual](https://www.ecobright.nl/wms/fm/userfiles/catalogus/attachments/ECO-DIM-07-Zigbee-Basic-handleiding.pdf)

Hello, I would like to ask you for help. I moved from Domoticz to Home Assistant (HASS) and now can't understand how to add new devices. In logs I can see:
 
Device '0x086bd7fffe5f149d' with Zigbee model 'Dimmer-Switch-ZB3.0' and manufacturer name 'EcoDim BV' is NOT supported.
10x Received message from unsupported device.

It was [#2342](https://github.com/Koenkk/zigbee2mqtt/issues/2342) @17.11.2019, Strange it not work now.

Thanks you, and have a good day.

huizebruin, 2021-05-18:
same here.

Friendly name
    0xccccccfffe3a1530
Last seen
    N/A
Device type
    Router
Zigbee model
    Dimmer-Switch-ZB3.0
Support status

    Unsupported
IEEE address
    0xccccccfffe3a1530
Network address
    0xE628
Firmware build date
    20190803
Firmware version
    3.02

Koenkk, 2021-05-18:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/7325

closed 2021-05-18 by Koenkk