Koenkk/zigbee2mqtt #7519: Support for Tuya Smart Air Box

TheOriginal92, 2021-05-18:
Hello, 
there seems to be a quite new ZigBee based Tuxa Air Sensor. 

Measuring a few interesting parameters like CO2, Formaldehyde, VOC, Temperature and Humidity.

Example: Tuya Air Box
https://a.aliexpress.com/_v0gA0X

Does Z2M already beta-support this device as I can’t find it on the Devices List?

If not so I will help integrating it as soon as it arrives by mail :)

Thank all of you very much!

TheOriginal92, 2021-05-19:
Finally the device arrived. I took it apart and found.. not much.

The board looks pretty empty and uses minimal parts count. 
Its equipped with an DHT11 for Temperature and Humidity. The second sensor is unknown to me. 

[image]
[image]
[image]

Next up is adding the converter.

meisterlampe, 2021-05-22:
You can find a WIP-Converter here: https://github.com/Koenkk/zigbee2mqtt/discussions/7299#discussioncomment-769443

artemzaharov, 2021-05-25:
Hello guys! Thanks for your work, so finally this device will be add to zigbee2mqtt and work from box?

closed 2021-07-03 by github-actions[bot]