Koenkk/zigbee2mqtt #7520: installing converters for unsupported zigbee devices

jmarfer, 2021-05-18:
By linking a thermostatic valve TS0601 indicates that this device is not supported. In the error messages it gives me a github address where I access a zip file to use some converters. I download the zip file, but what do I do with it?

Thanks

afoeder, 2021-05-19:
what github address? usually you would follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

jmarfer, 2021-05-19:
Yes, I follow that direction, but I don't know how to configure the valve:
Zigbee model 'TS0601' and manufacturer name '_TZE200_b6wax7g0'

jmarfer, 2021-05-19:
When I get to this address: https://github.com/Koenkk/zigbee-herdsman-converters, I can download the zip file, but what do I do with it?

Thank you very much for the help

i-000, 2021-05-20:
Hello, I have the same doubt, please can help us?

Thanks!!

jmarfer, 2021-05-20:
I have created a file called Thermostatic Valves.js and I have added the following configuration:

const exposes = require ('../ lib / exposes');
const fz = {... require ('../ converters / fromZigbee'), legacy: require ('../ lib / legacy'). fromZigbee};
const tz = require ('../ converters / toZigbee');
const globalStore = require ('../ lib / store');
const ota = require ('../ lib / ota');
const tuya = require ('../ lib / tuya');
const reporting = require ('../ lib / reporting');
const extend = require ('../ lib / extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
        zigbeeModel: ['kud7u2l'],
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_b6wax7g0'}, {modelID: 'TS0601', manufacturerName: '_TZE200_ywdxldoj'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_cwnjrr72'}],
        model: 'TS0601_thermostat',
        vendor: 'TuYa',
        description: 'Radiator valve with thermostat',
        whiteLabel: [{vendor: 'Moes', model: 'HY368'}, {vendor: 'Moes', model: 'HY369RT'}, {vendor: 'SHOJZJ', model: '378RT'}],
        meta: {tuyaThermostatPreset: tuya.thermostatPresets, tuyaThermostatSystemMode: tuya.thermostatSystemModes3},
        ota: ota.zigbeeOTA,
        onEvent: tuya.onEventSetLocalTime,
        fromZigbee: [fz.tuya_thermostat, fz.ignore_basic_report, fz.ignore_tuya_set_time],
        toZigbee: [tz.tuya_thermostat_child_lock, tz.tuya_thermostat_window_detection, tz.tuya_thermostat_valve_detection,
            tz.tuya_thermostat_current_heating_setpoint, tz.tuya_thermostat_auto_lock,
            tz.tuya_thermostat_calibration, tz.tuya_thermostat_min_temp, tz.tuya_thermostat_max_temp,
            tz.tuya_thermostat_boost_time, tz.tuya_thermostat_comfort_temp, tz.tuya_thermostat_eco_temp,
            tz.tuya_thermostat_force_to_mode, tz.tuya_thermostat_force, tz.tuya_thermostat_preset, tz.tuya_thermostat_away_mode,
            tz.tuya_thermostat_window_detect, tz.tuya_thermostat_schedule, tz.tuya_thermostat_week, tz.tuya_thermostat_away_preset],
        exposes: [
            e.child_lock (), e.window_detection (), e.battery_low (), e.valve_detection (), e.position (),
            exposes.climate (). withSetpoint ('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
                .withLocalTemperature (ea.STATE) .withSystemMode (['heat', 'auto', 'off'], ea.STATE_SET)
                .withLocalTemperatureCalibration (ea.STATE_SET)
                .withAwayMode (). withPreset (['schedule', 'manual', 'boost', 'complex', 'comfort', 'eco']),
            e.auto_lock (), e.away_mode (), e.away_preset_days (), e.boost_time (), e.comfort_temperature (), e.eco_temperature (), e.force (),
            e.max_temperature (), e.min_temperature (), e.week (), e.away_preset_temperature ()],
    },

After restarting Zigbee2mqtt it still does not support the device

jmarfer, 2021-05-20:
Is there anything else I need to do?

yarosman, 2021-05-20:
Does it exist any valid instructions for adding new devices in Hassio? Because nothing work

jmarfer, 2021-05-21:
I don't know

jmarfer, 2021-05-29:
Hello again

Can someone tell me where exactly the .js file is created? I don't really understand "next to" configuration.yaml

2. Adding your device #
The next step is to create an external converter file. This file has to be created next to the configuration.yaml

robertalexa, 2021-06-01:
@jmarfer That means in the same folder as your zigbee2mqtt configuration.yaml file
[image]

So in config/zigbee2mqtt folder create a new file for your device.

Also make sure to read the Tuya specific instructions
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_tuya_devices.html

You could also set zigbee2mqtt locally with a spare antenna and have a play with it and support straight in devices/tuya.js

I suspect that the thermostat supports all the same features as the other models, so it might be a case of just adding this TRV model to this code 
```
    {
        zigbeeModel: ['kud7u2l'],
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_ckud7u2l'}, {modelID: 'TS0601', manufacturerName: '_TZE200_ywdxldoj'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_cwnjrr72'}],
        model: 'TS0601_thermostat',
        vendor: 'TuYa',
        description: 'Radiator valve with thermostat',
        whiteLabel: [{vendor: 'Moes', model: 'HY368'}, {vendor: 'Moes', model: 'HY369RT'}, {vendor: 'SHOJZJ', model: '378RT'}],
        meta: {tuyaThermostatPreset: tuya.thermostatPresets, tuyaThermostatSystemMode: tuya.thermostatSystemModes3},
        ota: ota.zigbeeOTA,
        onEvent: tuya.onEventSetLocalTime,
        fromZigbee: [fz.tuya_thermostat, fz.ignore_basic_report, fz.ignore_tuya_set_time],
        toZigbee: [tz.tuya_thermostat_child_lock, tz.tuya_thermostat_window_detection, tz.tuya_thermostat_valve_detection,
            tz.tuya_thermostat_current_heating_setpoint, tz.tuya_thermostat_auto_lock,
            tz.tuya_thermostat_calibration, tz.tuya_thermostat_min_temp, tz.tuya_thermostat_max_temp,
            tz.tuya_thermostat_boost_time, tz.tuya_thermostat_comfort_temp, tz.tuya_thermostat_eco_temp,
            tz.tuya_thermostat_force_to_mode, tz.tuya_thermostat_force, tz.tuya_thermostat_preset, tz.tuya_thermostat_away_mode,
            tz.tuya_thermostat_window_detect, tz.tuya_thermostat_schedule, tz.tuya_thermostat_week, tz.tuya_thermostat_away_preset],
        exposes: [
            e.child_lock(), e.window_detection(), e.battery_low(), e.valve_detection(), e.position(),
            exposes.climate().withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
                .withLocalTemperature(ea.STATE).withSystemMode(['heat', 'auto', 'off'], ea.STATE_SET)
                .withLocalTemperatureCalibration(ea.STATE_SET)
                .withAwayMode().withPreset(['schedule', 'manual', 'boost', 'complex', 'comfort', 'eco']),
            e.auto_lock(), e.away_mode(), e.away_preset_days(), e.boost_time(), e.comfort_temperature(), e.eco_temperature(), e.force(),
            e.max_temperature(), e.min_temperature(), e.week(), e.away_preset_temperature()],
    },
```

I do not have this device so I am unable to test this myself.

I hope this helps you a bit. You could try the external converter and use the bit from my quoted code to handle what it exposes. Same for toZigbee and fromZigbee sections.

jmarfer, 2021-06-01:
> Thank you very much for your help!!

One more question, please. I leave all the configuration in case there were any errors, because it does not work for me yet. I have put the specific name of my valve (_TZE200_b6wax7g0).

This would have to change it? zigbeeModel: ['kud7u2l']. I use as coordinator zzh

Thanks again!!

All the best!

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = 

    {
        zigbeeModel: ['kud7u2l'],
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_b6wax7g0'}, {modelID: 'TS0601', manufacturerName: '_TZE200_ywdxldoj'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_cwnjrr72'}],
        model: 'TS0601_thermostat',
        vendor: 'TuYa',
        description: 'Radiator valve with thermostat',
        whiteLabel: [{vendor: 'Moes', model: 'HY368'}, {vendor: 'Moes', model: 'HY369RT'}, {vendor: 'SHOJZJ', model: '378RT'}],
        meta: {tuyaThermostatPreset: tuya.thermostatPresets, tuyaThermostatSystemMode: tuya.thermostatSystemModes3},
        ota: ota.zigbeeOTA,
        onEvent: tuya.onEventSetLocalTime,
        fromZigbee: [fz.tuya_thermostat, fz.ignore_basic_report, fz.ignore_tuya_set_time],
        toZigbee: [tz.tuya_thermostat_child_lock, tz.tuya_thermostat_window_detection, tz.tuya_thermostat_valve_detection,
            tz.tuya_thermostat_current_heating_setpoint, tz.tuya_thermostat_auto_lock,
            tz.tuya_thermostat_calibration, tz.tuya_thermostat_min_temp, tz.tuya_thermostat_max_temp,
            tz.tuya_thermostat_boost_time, tz.tuya_thermostat_comfort_temp, tz.tuya_thermostat_eco_temp,
            tz.tuya_thermostat_force_to_mode, tz.tuya_thermostat_force, tz.tuya_thermostat_preset, tz.tuya_thermostat_away_mode,
            tz.tuya_thermostat_window_detect, tz.tuya_thermostat_schedule, tz.tuya_thermostat_week, tz.tuya_thermostat_away_preset],
        exposes: [
            e.child_lock(), e.window_detection(), e.battery_low(), e.valve_detection(), e.position(),
            exposes.climate().withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
                .withLocalTemperature(ea.STATE).withSystemMode(['heat', 'auto', 'off'], ea.STATE_SET)
                .withLocalTemperatureCalibration(ea.STATE_SET)
                .withAwayMode().withPreset(['schedule', 'manual', 'boost', 'complex', 'comfort', 'eco']),
            e.auto_lock(), e.away_mode(), e.away_preset_days(), e.boost_time(), e.comfort_temperature(), e.eco_temperature(), e.force(),
            e.max_temperature(), e.min_temperature(), e.week(), e.away_preset_temperature()],
    },

robertalexa, 2021-06-01:
@jmarfer 
Did you do this too?
[image]

Could you please provide the entry for this device from database.db file? It is in the same folder as what I have shown you above. Find the entry for your device and send it over.

Tomorrow I will try to put some code together if i have the time for you to try as an external converter.

jmarfer, 2021-06-01:
Hello again!

When changing these configuration parameters I cannot enter zigbee2mqtt (502: Bad Gateway)

In the database I get this:
{"id": 11, "type": "Group", "groupID": 901, "members": [], "meta": {}}

finally this is the configuration:

const exposes = require ('../ lib / exposes');
const fz = {... require ('../ converters / fromZigbee'), legacy: require ('../ lib / legacy'). fromZigbee};
const tz = require ('../ converters / toZigbee');
const globalStore = require ('../ lib / store');
const ota = require ('../ lib / ota');
const tuya = require ('../ lib / tuya');
const reporting = require ('../ lib / reporting');
const extend = require ('../ lib / extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [

{
        zigbeeModel: ['TS0601'],
        fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_b6wax7g0'}, {modelID: 'TS0601', manufacturerName: '_TZE200_ywdxldoj'},
            {modelID: 'TS0601', manufacturerName: '_TZE200_cwnjrr72'}],
        model: 'TS0601_thermostat',
        vendor: 'TuYa',
        description: 'Radiator valve with thermostat',
        whiteLabel: [{vendor: 'Moes', model: 'HY368'}, {vendor: 'Moes', model: 'HY369RT'}, {vendor: 'SHOJZJ', model: '378RT'}],
        meta: {tuyaThermostatPreset: tuya.thermostatPresets, tuyaThermostatSystemMode: tuya.thermostatSystemModes3},
        ota: ota.zigbeeOTA,
        onEvent: tuya.onEventSetLocalTime,
        fromZigbee: [fz.tuya_thermostat, fz.ignore_basic_report, fz.ignore_tuya_set_time],
        toZigbee: [tz.tuya_thermostat_child_lock, tz.tuya_thermostat_window_detection, tz.tuya_thermostat_valve_detection,
            tz.tuya_thermostat_current_heating_setpoint, tz.tuya_thermostat_auto_lock,
            tz.tuya_thermostat_calibration, tz.tuya_thermostat_min_temp, tz.tuya_thermostat_max_temp,
            tz.tuya_thermostat_boost_time, tz.tuya_thermostat_comfort_temp, tz.tuya_thermostat_eco_temp,
            tz.tuya_thermostat_force_to_mode, tz.tuya_thermostat_force, tz.tuya_thermostat_preset, tz.tuya_thermostat_away_mode,
            tz.tuya_thermostat_window_detect, tz.tuya_thermostat_schedule, tz.tuya_thermostat_week, tz.tuya_thermostat_away_preset],
        exposes: [
            e.child_lock (), e.window_detection (), e.battery_low (), e.valve_detection (), e.position (),
            exposes.climate (). withSetpoint ('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
                .withLocalTemperature (ea.STATE) .withSystemMode (['heat', 'auto', 'off'], ea.STATE_SET)
                .withLocalTemperatureCalibration (ea.STATE_SET)
                .withAwayMode (). withPreset (['schedule', 'manual', 'boost', 'complex', 'comfort', 'eco']),
            e.auto_lock (), e.away_mode (), e.away_preset_days (), e.boost_time (), e.comfort_temperature (), e.eco_temperature (), e.force (),
            e.max_temperature (), e.min_temperature (), e.week (), e.away_preset_temperature ()],
    },

wizu, 2021-06-02:
https://github.com/Koenkk/zigbee2mqtt/issues/3821#issuecomment-852808488

jmarfer, 2021-06-02:
Good news!!

I have managed to link the valve. In HA I have all these entities:

[image]

I don't know if the entities can be seen. Now I would have to test an automation to see that everything is going well

Something to try?

Thank you

Ahh, In database I have this entry:

{"id": 9, "type": "EndDevice", "ieeeAddr": "0xcc86ecfffeb80185", "nwkAddr": 61802, "manufId": 4098, "manufName": "_ TZE200_b6wax7g0", "powerSource": "Battery" , "modelId": "TS0601", "epList": [1], "endpoints": {"1": {"profId": 260, "epId": 1, "devId": 81, "inClusterList": [ 0,4,5,61184], "outClusterList": [25,10], "clusters": {"genBasic": {"attributes": {"65503": "} bI (\ u0013�bI (\ u0012� bI (\ u0012 "," 65506 ": 31," 65508 ": 0," modelId ":" TS0601 "," manufacturerName ":" _ TZE200_b6wax7g0 "," appVersion ": 64," powerSource ": 3," zclVersion ": 3, "stackVersion": 0, "hwVersion": 1, "dateCode": ""}}}, "binds": [], "configuredReportings": [], "meta": {}}}, "appVersion" : 64, "stackVersion": 0, "hwVersion": 1, "dateCode": "", "zclVersion": 3, "interviewCompleted": true, "meta": {}, "lastSeen": 1622659009752}

jmarfer, 2021-06-02:
Another thing. What is this error of the triangle due to?

[image]

jmarfer, 2021-06-02:
[image]

closed 2021-07-10 by github-actions[bot]