Koenkk/zigbee2mqtt #7526: Hue Dimmer v2 - unspecified / empty action

alexw1982, 2021-05-19:
I got some [hue v2 dimmers 929002398602](https://www.zigbee2mqtt.io/devices/929002398602.html#philips-929002398602) and I am having an issue with zigbee2mqtt. 
All button pushes / events are recognized, however sometimes button pushes / actions are recognized without classifiying them, resulting in `"action":""`

```
info MQTT publish: topic 'zigbee2mqtt/dimmer.marie', payload '{"action":"","battery":100,"linkquality":97}'
info MQTT publish: topic 'zigbee2mqtt/dimmer.marie', payload '{"action":"on_press_release","battery":100,"linkquality":97}'
info MQTT publish: topic 'zigbee2mqtt/dimmer.marie/action', payload 'on_press_release'
info MQTT publish: topic 'zigbee2mqtt/dimmer.marie', payload '{"action":"on_press","battery":100,"linkquality":97}'
info MQTT publish: topic 'zigbee2mqtt/dimmer.marie/action', payload 'on_press'
info MQTT publish: topic 'zigbee2mqtt/dimmer.marie', payload '{"action":"on_press_release","battery":100,"linkquality":97}'
info MQTT publish: topic 'zigbee2mqtt/dimmer.marie/action', payload 'on_press_release'
info MQTT publish: topic 'zigbee2mqtt/dimmer.marie', payload '{"action":"","battery":100,"linkquality":99}'
info MQTT publish: topic 'zigbee2mqtt/dimmer.marie', payload '{"action":"on_press_release","battery":100,"linkquality":97}'
info MQTT publish: topic 'zigbee2mqtt/dimmer.marie/action', payload 'on_press_release'
info MQTT publish: topic 'zigbee2mqtt/dimmer.marie', payload '{"action":"on_press","battery":100,"linkquality":97}'
info MQTT publish: topic 'zigbee2mqtt/dimmer.marie/action', payload 'on_press'
info MQTT publish: topic 'zigbee2mqtt/dimmer.marie', payload '{"action":"on_press_release","battery":100,"linkquality":97}'
info MQTT publish: topic 'zigbee2mqtt/dimmer.marie/action', payload 'on_press_release'
info MQTT publish: topic 'zigbee2mqtt/dimmer.marie', payload '{"action":"","battery":100,"linkquality":99}'
info MQTT publish: topic 'zigbee2mqtt/dimmer.marie', payload '{"action":"on_press_release","battery":100,"linkquality":97}'
info MQTT publish: topic 'zigbee2mqtt/dimmer.marie/action', payload 'on_press_release'
info MQTT publish: topic 'zigbee2mqtt/dimmer.marie', payload '{"action":"","battery":100,"linkquality":97}'
info MQTT publish: topic 'zigbee2mqtt/dimmer.marie', payload '{"action":"on_press_release","battery":100,"linkquality":97}'
info MQTT publish: topic 'zigbee2mqtt/dimmer.marie/action', payload 'on_press_release'
```

## Debug info
Zigbee2MQTT version: docker:latest
Adapter hardware: CC2531

Koenkk, 2021-05-19:
Do you have `homeassistant: true` in your configuration.yaml? If yes, then close this issue since this is required for HA to respond to the actions.

alexw1982, 2021-05-19:
Yes I have `homeassistant: true` in my z2m configuration.yaml. HA responds to the action if the action inside the payload does not equal "". However whenever it equals only "" there is obviously no way to set up an automation. Also the "" action occurs across all buttons on the dimmer switch occasionaly. I could hit the on/off button five times and out of these five times one time may be "" instead of "on_press" (and the automation obviosuly is not triggered).

Koenkk, 2021-05-20:
It should occur after every  `action`, can you provide the debug log? 

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

alexw1982, 2021-05-21:
the above log excerpt includes all entries related to the dimmer.
I will provide some debug logs over the weekend.

Lastly a rather important side note: Thanks for all the hard work you are putting into this!

closed 2021-06-28 by github-actions[bot]

smnk, 2023-09-20:
Is it possible I'm still seeing this issue? I have a V2 dimmer that shows an empty action on each press. It's correctly recognized as far as I know:

```
Device type: EndDevice
Zigbee model: RWL022
Firmware build date: 20201125
Firmwareversie: 2.44.0_hBB3C188
Model: [929002398602](https://www.zigbee2mqtt.io/devices/929002398602.html#philips-929002398602)
```

On the Exposed tab, each time I press I can see the LQI updating and the action is empty:

`action: Lege waarde("")`

(Lege waarde = empty value, in Dutch)

Here is a part of the log after some presses:

```
Info 2023-09-20 19:18:25MQTT publish: topic 'zigbee2mqtt/Kamer Nijs Remote', payload '{"action":"on_press","battery":66.5,"linkquality":87,"update":{"installed_version":33565696,"latest_version":33565954,"state":"available"},"update_available":null}'
Info 2023-09-20 19:18:25MQTT publish: topic 'zigbee2mqtt/Kamer Nijs Remote', payload '{"action":"","battery":66.5,"linkquality":87,"update":{"installed_version":33565696,"latest_version":33565954,"state":"available"},"update_available":null}'
Info 2023-09-20 19:18:25MQTT publish: topic 'zigbee2mqtt/Kamer Nijs Remote/action', payload 'on_press'
Info 2023-09-20 19:18:25MQTT publish: topic 'zigbee2mqtt/Kamer Nijs Remote', payload '{"action":"on_press_release","battery":66.5,"linkquality":91,"update":{"installed_version":33565696,"latest_version":33565954,"state":"available"},"update_available":null}'
Info 2023-09-20 19:18:25MQTT publish: topic 'zigbee2mqtt/Kamer Nijs Remote', payload '{"action":"","battery":66.5,"linkquality":91,"update":{"installed_version":33565696,"latest_version":33565954,"state":"available"},"update_available":null}'
Info 2023-09-20 19:18:25MQTT publish: topic 'zigbee2mqtt/Kamer Nijs Remote/action', payload 'on_press_release'
```

It looks like there are actions being registered, but they're getting overwritten by an empty value?

I have some V1 dimmers still in use and had to enable legacy mode in Z2M to get them to work. I'm in the process of replacing all Hue dimmers with EnOcean buttons, but I still would like these to work. Any help is greatly appreciated. I already tried repairing, updated the firmware, and restarted the device by holding all buttons. I have another V2 dimmer and they both produce the same result.

waldopree, 2023-11-12:
Does anyone have a solution for this problem? It got it too out of nowhere just like @smnk. It seems the value of action is overwritten. It might have something to do with the last update?

nilsacht, 2023-12-30:
I am new to ZigBee2Mqtt and added my smartbutton. I have the same issue. When I press my button I get "String ohne Inhalt ("")" (it means empty string). After some time nothing is pressed I get N/A in the action topic. 

However home assistant reacts to the press in an easy automation (button is pressed toggle light).

bplatypus, 2024-05-23:
same for me, but is it normal that no log is shown in Z2M??

horenzloerner, 2024-10-02:
Any updates / solutions already found? I just wanted to install my first 2 hue wall switch modules, but i also have the problem that the action is always empty "" (Empty string(""))... 

Am i doing something wrong?

hvalev, 2024-10-24:
I am in the same boat. After upgrading my coordinator to the latest firmware and recreating my zigbee network, the remote is connected to the network, but there is no light indicator on key-presses on the remote and the actions value is not updated on the broker either. Firmware version of the remote is 1107324829

waldopree, 2024-11-03:
I switched this off and it worked! 
[image]

saschabrockel, 2025-01-10:
> Any updates / solutions already found? I just wanted to install my first 2 hue wall switch modules, but i also have the problem that the action is always empty "" (Empty string(""))...
> 
> Am i doing something wrong?

Same here out of nowhere. Did you find a solution?