Koenkk/zigbee2mqtt #7528: Manufacturer specific attributes and configureReporting

TheStigh, 2021-05-19:
Hi,

My new device got some manufacturer specific attributes.
Everything works fine as long as I control the device from Z2M, but if I change the attribute 0x4005 from the device itself, Z2M do not receive any data from the device indicating this change. So I assume I have to make some reporting and tried with

```
    configure: async (device, coordinatorEndpoint, logger) => {

        const endpoint = device.getEndpoint(1);
        const options = {manufacturerCode: 4919};
        await reporting.bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
        await reporting.lockState(endpoint);
        await reporting.batteryPercentageRemaining(endpoint);
        await endpoint.read('closuresDoorLock', ['lockState', 'soundVolume', 'doorState']);
        await endpoint.read('closuresDoorLock', [0x4000, 0x4001, 0x4004, 0x4005], options);

        await endpoint.configureReporting('closuresDoorLock', [{
            attribute: {ID: 0x4005, type: 0x10},
            minimumReportInterval: 0,
            maximumReportInterval: repInterval.MINUTE,
            reportableChange: 0,
        }], options);

    },
```
I get no errors during pairing. I can still change the attribute 0x4005 from Z2M and states get updated. Still, if I change the attribute 0x4005 from the device itself, Z2M still don't get updated. There are nothing in the Z2M log indicating an updated state from the device.

What am I missing ?

Koenkk, 2021-05-19:
I would indeed expect that this can be fixed by configuring reporting on this attribute, but note that the device itself is responsible for doing to actual reporting. The best way to debug this is to check if this works with the original gateway (to figure out if the device even supports this)

TheStigh, 2021-05-19:
I've spoke with the manufacturer today and they do indeed support this (it was actually them who made me aware of this 'bug' in my integration). Never configured reporting, so the code above are obviously not enough ?

Koenkk, 2021-05-19:
No, could you ask them how they do this? If they could provide a sniff of the gateway sending this command we should be able to send the same.

TheStigh, 2021-05-19:
See PM on Telegram

TheStigh, 2021-05-22:
@Koenkk 

My code actually worked - a buggy CC2531 was the issue. Closing the topic.

closed 2021-05-22 by TheStigh