Koenkk/zigbee2mqtt #7535: Unsupported device ready to add (TS0222 / _TYZB01_kvwjujy9)

AlexKay88, 2021-05-19:
## Information about the device + link
I got a unsupported device working with Z2M:
**fingerprint**: [{modelID: 'TS0222', manufacturerName: '_TYZB01_kvwjujy9'}]
**Vendor**: Moes
**Model**: ZSS-ZK-THL
**Description**: Smart Brightness Thermometer
**Bought at**: https://de.aliexpress.com/item/1005002549320064.html
**Picture**: 
[image]
## Pairing with Tuya gateway sniffed with CC2531
[DeviceID_0x06bc_TS0222_TYZB01_kvwjujy9.pcapng.gz](https://github.com/Koenkk/zigbee2mqtt/files/6511696/DeviceID_0x06bc_TS0222_TYZB01_kvwjujy9.pcapng.gz)

## Converter for the device
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const definition = {
    fingerprint: [{modelID: 'TS0222', manufacturerName: '_TYZB01_kvwjujy9'}],
    model: 'ZSS-ZK-THL',
    vendor: 'Moes',
    description: 'Smart Brightness Thermometer',
    fromZigbee: [fz.battery, fz.illuminance, fz.humidity, fz.temperature],
    toZigbee: [],
    exposes: [e.battery(), e.illuminance(), e.illuminance_lux().withUnit('lx'), e.humidity(), e.temperature()],
};
module.exports = definition;
```
## data/database.db entry of the device
`{"id":55,"type":"EndDevice","ieeeAddr":"0x5c0272fffe0a490b","nwkAddr":52115,"manufId":4098,"manufName":"_TYZB01_kvwjujy9","powerSource":"Battery","modelId":"TS0222","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":262,"inClusterList":[0,1,1024],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0222","manufacturerName":"_TYZB01_kvwjujy9","powerSource":3,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":10415}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":770,"inClusterList":[1026,1029],"outClusterList":[],"clusters":{"msIlluminanceMeasurement":{"attributes":{"61441":20}},"msTemperatureMeasurement":{"attributes":{"measuredValue":1904}},"msRelativeHumidity":{"attributes":{"measuredValue":4685}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1621459608001}
`

freitas666, 2021-05-20:
Well, I have 5 of that and nothing works... I hope they fix it soon possible! :(

AlexKay88, 2021-05-20:
@freitas666
if you don't want to wait, you can follow "2. Adding your device" at https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
The converter you need to copy&paste into a .js file, is shown above.

freitas666, 2021-05-20:
@AlexKay88 Hey friend... well I was reading about that.... but I did not know how to start with procedure... could you help me?

freitas666, 2021-05-20:
Hey bro.. I start some troubleshottings here and now looks like perfectly! :)

freitas666, 2021-05-20:
I found a strange bug.... not show the battery level....!LOL,,, have you some idea?

AlexKay88, 2021-05-20:
@freitas666 Battery powered devices sleep most of the time to save battery, that causes delay on updating some values. You can wait or wake up device by pressing orange button on right side of it.
My two devices update Illuminance, humidity, temperature and battery without problems.

freitas666, 2021-05-20:
@AlexKay88 hey friend thank you for your feedback.... after some restarts in my z2m... they start working well...! :D

I have one doubt .... how do u sniff this connections to create a "template"?? Example actually I bought some door sensor (TS0203) but I have the same errors...
Could you help me?

AlexKay88, 2021-05-20:
@freitas666 Glad to hear thats it's also working on your end. Thanks for confirmation. For your TS0203 pls. check if there is already an open issue and if not, pls open your own new issue and mention me. I will see what I can do.

@Koenkk Would be great if you could look at my new converter, check if I did not make any mistakes and add it as a new supported device. I am unexperienced, but I am learning every day.

freitas666, 2021-05-20:
@AlexKay88 thank you! @Koenkk actually I am using this converter and works good in 4 devices.

Koenkk, 2021-05-20:
@AlexKay88 looks good, thanks! Added it.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-05-20 by Koenkk

blakadder, 2021-05-20:
Any idea why this sensor works normally with Z2M but when I try it in ZHA and Tasmota it reannounces itself on screen refresh and some values get stuck on 0?

DavidFFerreira, 2021-05-24:
it stills dont work, I have the edge in my HA. I did a force remote. Take out the battery and wait a few minuts. After try to pair again, I got this log:
Info Successfully interviewed '0x5c0272fffe0a438d', device has successfully been paired Warning Device '0x5c0272fffe0a438d' with Zigbee model 'TS0222' and manufacturer name '_TYZB01_kvwjujy9' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x5c0272fffe0a438d","ieee_address":"0x5c0272fffe0a438d","status":"successful","supported":false},"type":"device_interview"}' Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x5c0272fffe0a438d","supported":false},"type":"pairing"}' Warning Received message from unsupported device with Zigbee model 'TS0222' and manufacturer name '_TYZB01_kvwjujy9' Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

freitas666, 2021-05-25:
@DavidFFerreira 

hello friend...
Actually i have 5 devices working good in my Z2M.. 
Could you put the steps here?

simulot, 2021-09-28:
Hi, I have a batch of 6 ZSS-ZK-THL devices. Nice devices perfect for room temperature monitoring with a screen for humans and ZigBee connection for automation...

As @blakadder and @DavidFFerreira, my devices are behaving badly. After a variable moment, devices don't display or report temperature and humidity. Strangely light sensor is still working both on display and on zigbee connection. I must remove briefly the battery to reset it. Then it works for about one week. 

I have contacted Moes.  They consider that the device must be used with a Tuya hub. 

I suspect a product defect. Otherwise why lux sensor is still working? 
Or mayby Z2M is not gentle with them, and make them fautlling. 

@freitas666, are your devices still working? Are them connected to an Tuya hub? 

My configuration:
USB ConBee2 stick plugged on a Pi4 running the latest docker image

AlexKay88, 2021-09-28:
Same here.
After some time temperature and humidity is stuck, while light sensor is still working. Pressing the side button sometimes helps. If not, removing and reinserting battery helps.

kkossev, 2021-09-28:
@simulot @AlexKay88 can you make an experiment, pairing the TS0222 very close to a Zigbee repeating device that is also connected to the same z2m coordinator?  It may work also if you simply remove the batteries for 1 minute and insert the battery cell again, while the sensor is very close to the repeater. Zigbee repeating devices are any mains powered zigbee devices, such as a power plugs, zigbee bulbs, etc... This is to ensure that the communication between the sensor and the zigbee coordinator is not direct, but through a repeating device. You can also verify on the Zigbee2MQTT Map whether the link is via repeater or direct. For the experiment, leave the TS0222 close to the repeater if possible. You can monitor the performance in the HA History for a day or two, when the device resets there are rapid drops of the temperature and humidity values on the graph.
The above setup has worked for me on several different zigbee platforms.. When TS0222 communicates directly to the zigbee coordinator, it will self-reset periodically. When the communication goes through a zigbee repeater, the device is working stable without any resets for days!  
Can someone else confirm this finding?

AlexKay88, 2021-09-29:
My two devices are currently connected via repeaters and have also been connected via repeaters in the past. After a few days they don't show temperature and humidity any longer.

simulot, 2021-09-30:
Same here. Some of thermometers are located upstairs, using repeater devices.

To be sure, I have placed one thermometer close to a smart plug, and I have re-paired it. 12 hours later, it is still running. However, it takes usually about one week to fail.

kkossev, 2021-10-01:
Thank you @AlexKay88 and @simulot ! Same observations here, I had an almost 3 weeks period where TS0222 paired through a repeater worked without restarts, sending true measurements data when temperature, humidity or illuminance changes.. But at the end, probably something has changed in the zigbee network and the restarts began again.

This TS0222 device does not work reliably with any zigbee hub, other than Tuya.

simulot, 2021-10-01:
@kkossev, the tested thermometer has crashed this morning. Since then, no more temperature / humidity measure. It has ran during 18h. The display shows "E1" as temp. Lux is still functioning.  It is relayed by a TuYa TS0121_plug

In the next room, another thermometer is running since 26 days. This one is relayed by a Ikea ICPSHC24-30EU-IL-1. 
Another thermometer relayed by another TuYa TS0121_plug. This one is running since more than 30 days. 

@Koenkk  It time to remove this device from the compatibility list, or at least add big warnings

maweki, 2022-02-01:
I have the same device but it advertises as Zigbee model 'TS0222' and manufacturer name '_TYZB01_ftdkanlj'

I am using zigbee2mqtt as HomeAssistant addon though and I am unsure about how to add proceed. Can this other manufacturer name just be added to the existing definition? Where would I find that?

Koenkk, 2022-02-01:
@maweki `_TYZB01_ftdkanlj` will be supported in todays new release (few hours from now)

freitas666, 2022-02-02:
Man…

This sensor (ZSS-ZK-THL) its a bad project of moes……..

Unfortunately the product stop to work without explanation!

And its not just for you or me… many friends have the same issues… works for little days and stop!
I really give up this product of (Moes).
And for me its not a problem between sensor and zigbee dongle…
Because in normal hub tuya happen the same issue.

Thank you

Enviado do meu iPhone

kkossev, 2022-02-02:
I can still confirm the same problem. I have 2 different ZSS-ZK-THL devices, paired to two different Tuya Zigbee gateways - both devices stop updating the temperature and the humidity readings after a few days or a week. Not only the reported values, the digits on the e-Ink display freeze! Only the illuminance continues working. Seems like a firmware bug.

I have no idea why some customers report that the device is working... The only explanation for me is that these are very early reports, without actually checking whether the T/H readings are correct for a long time enough.

jhron, 2022-02-02:
I can confirm, that this device is pretty "useless". Although it seems that it is somehow working for a longer time now.  Mine is not connected directly to GW but over a router. So maybe it can be true that over router it is "working". 

I write "working" because although it measure something, values are sometimes weird. See attached image.

[image]

I use it in my garage where is temperature pretty stable - it is closed all the time, nobody goes in or out. Yet there are time to time some "temperature" drops.

The report interval is another story.... 1x per hour is pretty useless for this type of device. Only what is reported almost right away is illuminance, but not temperature or humidity. It does not matter if temperature around will change significantly.  It will just wait for its timer (on display too)..

kkossev, 2022-02-02:
> Yet there are time to time some "temperature" drops
These spikes are due to the zero temperature and humidity readings that follow immediately after the device self-resets / reboots.

Aggtaa, 2022-02-03:
Got mine 3 pcs from aliexpress yesterday, all are _TYZB01_ftdkanlj. Connected them directly to a coordinator (SONOFF usb dongle), so far so good.

From immediate observations:
pros:
- cheap
- slick and compact
- epaper display looks nice and is good for battery life
- reports T and H changes not on intervals but on actual changes. So in stable environment it can report nothing for a veeery long period of time (up to an hour of my patience). Actually both a con and a pro.
- No extreme temperature spikes or drops reported

cons:
- 3 devices from same batch (according to their serial numbers), all are off by up to 0.7C from a calibrated thermometer with total variance of 1.2C between them. Humidity is measured 2%-5% lower than it should be.
- (all the same) internal temperature debouncing or device mechanical structure is very eager to react to heating, almost an immediate reaction to hand touching warming it up yet it takes almost an hour to smooth down a natural 1C temperature drop.
- screen numbers are rounded to floor, not to closest number, so zigbee2mqtt reports numbers not maching the on-screen ones
- device screen is capable of displaying up to 999lux or "E1" error otherwise, which is not enough for a cloudy day (indoors, on a wall ~60cm from a window), although zigbee reports over 1000 lux are fine, so just a screen issue
- light sensor debouncing math is weird. Sitting still in a room on a wall shows around 50 lux. Cover the sensor by hand => goes to 0, fine. Take a step back, immediately goes to 80 lux in exactly same env. 30 lux or 60% is way too much for a sensor precision error.

overall, IMO, even not taking into consideration the zigbee connection instability without a repeater issue reported by many,
this device seems to be a nice-looking toy rather than a reliable home sensor.

rbrouwer, 2022-02-09:
Just wondering based on my experience so far (< 1 day with _TYZB01_ftdkanlj ).
My sensor actually have no issues connecting to my Zigate (with [Domoticz-Zigbee](https://github.com/zigbeefordomoticz/Domoticz-Zigbee)). I did notice it worked well for a few hours, but after a while it would shortly report 0 degrees C and 0% humidity and a few tenth of a sec later report a proper temperature. Even the sensor reporting 0 lux after plenty of hours later.

That issue seemed gone after replacing the factory included battery with another CR2032 cell. Did others try to replace the CR2032 battery? Maybe that could at least explain why this sensor behaving really badly for some, while others are pretty fine with it.

kkossev, 2022-02-10:
Yes, putting a new fresh battery cell seems to stop the cyclic restarts.... for a while. Unfortunately, after several days, or a week my 2 devices stop updating the temperature and humidity readings on both the e-Ink display and in the Zigbee reporting data.

Aggtaa, 2022-02-11:
Indeed, the device came with a stale battery, 2.8V on air, drops to 2.1V immediately under load when inserted. Replaced with a 3.6V LIR2032, connection is stable for a week now.

gacopl, 2022-02-13:
@Aggtaa also if possible upgrade firmware through tuya zigbee gateway once i done that i don't have any issues, on the values discrepancies you can adjust differences, but yeah luminance meter is kinda flaky it needs to be in the right spot but at the same time not direct sunlight for temps, so generally i would wish you could switch the view and show big temp instead of lux

NeuerUser, 2022-02-17:
Any chance to get a firmware update then without purchasing a Tuya gateway? Would be great if we could fix the issues with this.

AlexKay88, 2022-02-17:
My two displays are unreliable, too.
A few days ago I put new batteries inside, paired them to Tuja gateway and updated to V1.0.11, but  one of them already started to reboot again and show just 0 temperature and 0 humidity on e-ink display.

evlo, 2022-09-03:
> My two displays are unreliable, too. A few days ago I put new batteries inside, paired them to Tuja gateway and updated to V1.0.11, but one of them already started to reboot again and show just 0 temperature and 0 humidity on e-ink display.
Happened to me too with bad battery, so try changing for another ones maybe?

igorsantos07, 2022-11-02:
I had the same rebooting issues many people have reported (and, obviously, I failed to research before).
Then, I talked to AliExpress seller and they [1] asked if it's a Tuya gateway (which seems to be common but someone above complained about issues with the "correct" gateway) and then [2] asked me to replace the battery. Conclusion: I lost two batteries, the factory one and a brand new. In a few days with it on (and I'm out of home) it went dead, blinking black/white (which either seems to mean "I'm out of juice" or it's trying to startup and fails?). It seems the frequent reboots ate the whole battery, and then I replaced without understanding what was going on.

I should stress it was working wonders _before_ pairing with my Sonoff Dongle-P. As soon as I paired it with HA (which was a pain, because it was simply never recognized by my ZB-Minis on the wall), it started those reboots.

That's very sad since Moes seems to be a good brand, it's a rare device with TH+Lux+Zigbee, and the device is really pretty :man_shrugging:

alexebner, 2023-02-17:
Hi !
I have the same device but it reports as 
Zigbee Modell
    TS0222
Zigbee Hersteller
    _TYZB01_o0cx3zca

And it shows as unsuported.
What can i do ?