Koenkk/zigbee2mqtt #7538: Unsupported device TS011F

kimet83, 2021-05-20:
## Information about the device + link
..
tuya zibee 3.0 plug 
https://ko.aliexpress.com/item/1005002344798281.html?spm=a2g0s.9042311.0.0.2edc4c4dMTSgWB
Zigbee model 'TS011F' and manufacturer name '_TZ3000_cphmq0q7'
## data/database.db entry of the device
..
{"id":8,"type":"Router","ieeeAddr":"0xb4e3f9fffe19f30d","nwkAddr":3919,"manufId":4098,"manufName":"_TZ3000_cphmq0q7","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_cphmq0q7","powerSource":1,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1621506452613}

jellewestenberger, 2021-05-20:
Support for this device [has been added recently](https://github.com/Koenkk/zigbee-herdsman-converters/commit/9ea7675c542867abada40eeb64145363dd148716) but is currently only available on the dev branches. I received these plugs in the mail today and can confirm that they work in version 1.18.3-dev.

Koenkk, 2021-05-20:
Already added indeed (in the dev branch).  (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-05-20 by Koenkk

pcornill, 2022-03-31:
Hello, I know this issue is closed but I guess I have similar issues with a new TS011F _TZ3000_typdpbpg which has Appversion: 74

{"id":15,"type":"Router","ieeeAddr":"0xa4c1386de9c10da4","nwkAddr":56002,"manufId":4417,"manufName":"_TZ3000_typdpbpg","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57345,57344,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1648743576390,"defaultSendRequestWhen":"immediate"}

Is there a quick local fix?

rossiluis22, 2022-05-03:
@pcornill did you find any solution? I'm facing the same issue

pcornill, 2022-05-03:
I didn't do anything outside of delete/pair again.

rossiluis22, 2022-05-03:
> I didn't do anything outside of delete/pair again.

Thanks!, i’ve solved by updating zigbee2mqtt to the latest version

DieterKoblenz, 2023-05-28:
Is this device still only in dev z2m available? I'm running Zigbee2MQTT version [1.18.1] and it shows unsupported. Is it possible to add support manually without switching to dev?

Koenkk, 2023-05-28:
@DieterKoblenz 
If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

DieterKoblenz, 2023-05-28:
> @DieterKoblenz If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

Oh. My. 

Never mind. Works flawlessly.

teargas666, 2023-09-11:
Hi. I have the same situation. I have Modemix MOD037, which should be support OOB by Zigbee2MQQT, but when I tried to pair it with my HA, I receive information that it's not supported. 

My Z2M version is 1.33.0-1