Koenkk/zigbee2mqtt #7548: ZM-CG206 - Door/window sensor

frankiha, 2021-05-21:
Unsupported

## Information about the device + link
https://www.aliexpress.com/item/1005002439611647.html?spm=a2g0s.9042311.0.0.42024c4d1GJCUj

## data/database.db entry of the device
{"id":4,"type":"EndDevice","ieeeAddr":"0x00124b001857afb2","nwkAddr":45861,"manufId":0,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":200}}},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1621604110578}

robert-pajak, 2021-05-29:
Could anyone smarter/more experienced help with adding support for this device?

Steefph, 2021-06-14:
```
//       Converter for ZM-RT201 ZM-35ZH-Q with battery now.        https://www.youtube.com/channel/UCJRaKbgv8u1O4OwG-0D3uRw
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const motionSensorDetector = {
    cluster: 'ssIasZone',
    type: ['attributeReport'],
    convert: (model, msg, publish, options, meta) => {
/*             if (msg.data.zoneStatus !== 1) {
                // In case of 0 no occupancy is reported.
                // https://github.com/Koenkk/zigbee2mqtt/issues/467
                return;
            }

            // The occupancy sensor only sends a message when motion detected.
            // Therefore we need to publish the no_motion detected by ourselves.
            /* const timeout = options && options.hasOwnProperty('occupancy_timeout') ?
                options.occupancy_timeout : 15;   // occupancy delay in seconds

            // Stop existing timers because motion is detected and set a new one.
            globalStore.getValue(msg.endpoint, 'timers', []).forEach((t) => clearTimeout(t));
            globalStore.putValue(msg.endpoint, 'timers', []);

            if (timeout !== 0) {
                const timer = setTimeout(() => {
                    publish({occupancy: false});
                }, timeout * 1000);

                globalStore.getValue(msg.endpoint, 'timers').push(timer);
            } */

            // No occupancy since
           /*  if (options && options.no_occupancy_since) {
                options.no_occupancy_since.forEach((since) => {
                    const timer = setTimeout(() => {
                        publish({no_occupancy_since: since});
                    }, since * 1000);
                    globalStore.getValue(msg.endpoint, 'timers').push(timer);
                });
            } */

            if (msg.data.zoneStatus !== 1) {
                return {contact: true};
            } else {
                return {contact: false};
            }
        
            }
};

 const definition = {
//   fingerprint: [{modelID: 'TS0202', manufacturerName: '_TYZB01_dl7cejts'}],
    fingerprint: [{type: 'EndDevice', ieee_address: '0x00124b000f'}],   // WARNING! Change this ieee_address for yours ieee_address!!!!
    zigbeeModel: ['ZM-CG206'],
    model: 'HG06336',
    vendor: 'TuYa',
    description: 'magnetic contact sensor',
    fromZigbee: [ motionSensorDetector, fz.battery ],
    //fromZigbee: [ motionSensorDetector, fz.battery, fz.ignore_basic_report ],
    toZigbee: [],
    exposes: [e.contact(), e.battery_low(), e.battery()],
};

module.exports = definition;
```

I'm not an expert but, the above code does work in my case. Note that the ieee_address needs to be in your range of devices (or you could complete put your IEEE there "0x00124b001857afb2"

robert-pajak, 2021-06-14:
Thanks for sharing - will give it a try in a few minutes

Creating definition for multiple devices (multiple ieee_adresses) is not ideal but should work for now :)
BTW it looks like you are reading my mind - I was also interested in purchasing ZM-35ZH-Q :)

robert-pajak, 2021-06-14:
Thanks again - it kinds of works but zigbee-frontend reports these devices as not paired (but they are paired and appropriate MQTT messages are being produced) - could anyone smarter/more experienced advice please?
[image]

I took a freedom and tweak a bit your code to remove dead code/comments and not necessary dependencies
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const contactSensorDetector = {
    cluster: 'ssIasZone',
    type: ['attributeReport'],
    convert: (model, msg, publish, options, meta) => {
        if (msg.data.zoneStatus !== 1) {
            return {contact: true};
        } else {
            return {contact: false};
        }
    }
};

 const definition = {
    fingerprint: [{type: 'EndDevice', ieee_address: '0x00124b000f'}],   // WARNING! Change this ieee_address for yours ieee_address!!!!
    zigbeeModel: ['ZM-CG206'],
    model: 'HG06336',
    vendor: 'TuYa',
    description: 'Magnetic Contact Sensor',
    fromZigbee: [ contactSensorDetector, fz.battery ],
    toZigbee: [],
    exposes: [e.contact(), e.battery_low(), e.battery()],
};

module.exports = definition;
```

Steefph, 2021-06-15:
Just keep an eye on this one:
https://github.com/Koenkk/zigbee2mqtt/issues/7564

As it is the same "sort" of device with the same issues.

robert-pajak, 2021-06-15:
Thanks for sharing the link to 7564
I dived a bit deeper and zigbee-frontend reports interview was never completed for these ZM-CG206 devices
Also they never report battery_low values (for me it is always null) and it looks like z2m is not properly detecting these devices are battery devices
[image]
But all other properties are reported correctly (contact, battery %, linkquality)
Looks like we are almost there with adding support for these devices :)

Steefph, 2021-06-15:
Well not really, I've now the ZMRT201 and ZMCG206 converters fighting with each other. For some, for me unknown, reason the RT201 are recognized as CG206 while I've the fingerprint set specifically to the IEEE.

I do think that the empty-battery only gets triggered once it is send and it is only send when the event occurs (that's an assumption)

robert-pajak, 2021-06-15:
You might be right regarding `battery_low`
Thinking out loud it seems like device power source is not properly handled and this could be the reason for not completing 100% the interview.
I am a newbie to Zigbee protocol but it seems like ZM-CG206 devices have genBasic, genIdentify and genPowerCfg clusters
Maybe we can query them and mitigate IEEE issue and/or improve interview issue?

robert-pajak, 2021-06-16:
I managed to dive deeper (using Docker's DEBUG=zigbee-herdsman* setting)
and this is the result of generated log
https://pastebin.com/tZ8d1t2h
Am I on the right track with my learning?
Is this caused because we don't have IdentifyQuery implemented for this device
```
zigbee-herdsman:controller:log Skipping command 'identifyQuery' because it is missing from the lookup +0ms
```
Which wasn't a biggie for other devices as these other devices was able to get required information using other calls (and these calls fail for our device "at hand")?

 ```
 zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x00124b000f7b7f33:35418,240,4) +7s
  zigbee-herdsman:controller:endpoint Read 0x00124b000f7b7f33/1 genBasic(["modelId"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240)) +54s
  zigbee-herdsman:controller:device:log Interview - failed to read attribute 'modelID' from endpoint '1' (Error: Read 0x00124b000f7b7f33/1 genBasic(["modelId"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240))) +54s
  zigbee-herdsman:controller:endpoint Read 0x00124b000f7b7f33/1 genBasic(["manufacturerName"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +44s
  zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x00124b000f7b7f33:35418/1 (0,0,1) +1ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":35418,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":12,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,5,0,4,0]}} +7s
```
and then 
```
 zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x00124b000f7b7f33:35418,240,4) +7s
  zigbee-herdsman:controller:endpoint Read 0x00124b000f7b7f33/1 genBasic(["manufacturerName"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240)) +44s
  zigbee-herdsman:controller:device:log Interview - first manufacturerName retrieval attempt failed, retrying after 10 seconds... +44s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,23,68,129,0,0,3,0,90,138,1,47,1,120,0,152,210,217,0,0,3,17,11,1,90,138,29,16] +10s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,23,68,129,0,0,3,0,90,138,1,47,1,120,0,152,210,217,0,0,3,17,11,1,90,138,29,16] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 23 - 2 - 4 - 129 - [0,0,3,0,90,138,1,47,1,120,0,152,210,217,0,0,3,17,11,1,90,138,29] - 16 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":3,"srcaddr":35418,"srcendpoint":1,"dstendpoint":47,"wasbroadcast":1,"linkquality":120,"securityuse":0,"timestamp":14275224,"transseqnumber":0,"len":3,"data":{"type":"Buffer","data":[17,11,1]}} +10s
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":11,"manufacturerCode":null,"commandIdentifier":1},"Payload":{},"Command":{"ID":1,"parameters":[],"name":"identifyQuery"}},"address":35418,"endpoint":1,"linkquality":120,"groupID":0,"wasBroadcast":true,"destinationEndpoint":47}' +15s
```
and then 
```
  zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x00124b000f7b7f33:35418,240,4) +7s
  zigbee-herdsman:controller:endpoint Read 0x00124b000f7b7f33/1 genBasic(["powerSource"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC transaction expired' (240)) +43s
  zigbee-herdsman:controller:device:log Interview - failed to read attribute 'powerSource' from endpoint '1' (TypeError: Cannot read property 'powerSource' of undefined) +43s
  zigbee-herdsman:controller:endpoint Read 0x00124b000f7b7f33/1 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) +43s
```
and a few similar errors, for example;
```
  zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x00124b000f7b7f33:35418,240,4) +7s
  zigbee-herdsman:controller:endpoint Read 0x00124b000f7b7f33/1 genBasic(["zclVersion"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null [... cut ...]
Zigbee2MQTT:info  2021-06-16 13:52:36: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b000f7b7f33","ieee_address":"0x00124b000f7b7f33","status":"failed"},"type":"device_interview"}'
Zigbee2MQTT:info  2021-06-16 13:52:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x00124b000f7b7f33"},"type":"pairing"}'
  zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0} +3m
```

dessone, 2021-06-26:
> Well not really, I've now the ZMRT201 and ZMCG206 converters fighting with each other. For some, for me unknown, reason the RT201 are recognized as CG206 while I've the fingerprint set specifically [...]

Same problem here, i used the code you posted in other post.
I'm newbee so i can not help.
I stay here to see if there is a solution.
Thanks for the work

mastertape, 2021-07-09:
Some news for this Door Sensor?

Koenkk, 2021-07-09:
Looks to be the same device as https://github.com/Koenkk/zigbee2mqtt/issues/7553, I've ordered it and will debug further once I receive it, please keep an eye on that issue.

closed 2021-07-09 by Koenkk

hossface, 2021-08-04:
The ZMCG206 still has the same issue. it hasnt been resolved the same as #7553

phillprice, 2021-09-24:
I too have a problem with interviewing the CG206 device ```Zigbee2MQTT:error 2021-09-24 13:45:18: Failed to interview '0x00124b000fce4668', device has not successfully been paired
Zigbee2MQTT:error 2021-09-24 13:49:10: Failed to interview '0x00124b000fce4668', device has not successfully been paired```

I'm happy to help debug where I can :)

bkarlson, 2021-11-08:
Same here with the same device... It is listed as Tuya on the seller's website, but clearly it is some Chinese no-name. Unable to pair, so i'd have to return them...

phillprice, 2021-11-08:
The devices do pair with the latest version of this and the extra js file. Make sure you amend the ieee id. I’d prefer not to have one js file per device but it works!