Koenkk/zigbee2mqtt #7549: JZ-ZB-004 - Multiprise with 4 AC outlets and 2 USB super charging ports (WP33-EU)

eldelapura, 2021-05-21:
Hi!!

I just received a LELLKI JZ-ZB-004 - Multiprise with 4 AC outlets and 2 USB super charging ports (WP33-EU). I think that is the same TUYA 4AC outlet (TuYa E220-KR4N0Z0-HA - [https://www.zigbee2mqtt.io/devices/E220-KR4N0Z0-HA.html](https://www.zigbee2mqtt.io/devices/E220-KR4N0Z0-HA.html) ) but this one is nor recognized by Zigbee2mqtt.

The exact model in tuya version is E220-KR4N0Z0-HA and in my case zigbee2mqtt detects JZ-ZB-004 model.

I've tried to include a .js file using [https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js) with the parameters of E220-KR4N0Z0-HA but I'm doing something wrong.. :(:(

## Information about the device + link
[https://es.aliexpress.com/item/4001352735475.html?spm=a2g0o.cart.0.0.49f13c00ADpeMq&mp=1](https://es.aliexpress.com/item/4001352735475.html?spm=a2g0o.cart.0.0.49f13c00ADpeMq&mp=1)

LOG:
Zigbee2MQTT:warn  2021-05-21 16:21:28: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2021-05-21 16:21:28: Received message from unsupported device with Zigbee model 'JZ-ZB-004' and manufacturer name 'LELLKI'
..

## data/database.db entry of the device

{"id":34,"type":"Router","ieeeAddr":"0x00124b0022ed2743","nwkAddr":57320,"manufId":0,"manufName":"LELLKI","powerSource":"Mains (single phase)","modelId":"JZ-ZB-004","epList":[1,2,3,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{"modelId":"JZ-ZB-004","manufacturerName":"LELLKI","powerSource":1,"zclVersion":1,"appVersion":5,"hwVersion":1,"dateCode":"20201207","swBuildId":"1.0.5"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":5,"hwVersion":1,"dateCode":"20201207","swBuildId":"1.0.5","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1621606888892}
..

Thanks in advance!

jonathanathe, 2021-05-24:
Hi, I also bought the same and there is no way a question where did you find the devices.js file as I did not find it and I'm giving up already....

eldelapura, 2021-05-25:
> 
> 
> Hi, I also bought the same and there is no way a question where did you find the devices.js file as I did not find it and I'm giving up already....

Hi!
I read some information about this file, but I never saw this it. I'm using Home Assistant add-on and I'm following this guide: [https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html). 

In my case  after add an external converter (_external_converters:  - my_own_file.js_) I'm allways getting erros and zigbee2mqtt service never starts :(

jonathanathe, 2021-05-25:
> > Hola, también compré lo mismo y no hay duda de dónde encontraste el archivo devices.js ya que no lo encontré y ya me rindo ....
> 
> ¡Hola!
> Leí algo de información sobre este archivo, pero nunca lo vi. Estoy usando el complemento Home Assistant y sigo esta guía: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html .
> 
> En mi caso, después de agregar un convertidor externo ( _external_converters: - my_own_file.js_ ) siempre _recibo_ errores y el servicio zigbee2mqtt nunca se inicia :(

Me too, I hope someone will add it and it will be compatible for everyone...

Koenkk, 2021-05-25:
Added. 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

eldelapura, 2021-05-25:
> 
> 
> Added.
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Thanks!!!!!

jonathanathe, 2021-05-25:
Thank you, it's working! I have been waiting for a long time!

jonathanathe, 2021-05-25:
> Adicional.
> Los cambios estarán disponibles en la rama de desarrollo en unas pocas horas a partir de ahora. ( https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html )

Thank you, it's working! I have been waiting for a long time!