Koenkk/zigbee2mqtt #7553: Converter for TS0203 - DOOR SENSOR

freitas666, 2021-05-22:
Hello Guys,

Actually I am having problems to add in my Z2M this sensor door (TS0203)....

I follow some steps in website of zigbee2mqtt and "how to add new devices" ... and both not works for me....

So I tried create a file called "door.js" and put this informations above:

 const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const definition = {​​​​​​​
    fingerprint: [{​​​​​​​modelID: 'TS0203', manufacturerName: 'TS0203'}​​​​​​​],
    model: 'TS0203',
    vendor: 'CR Smart Home',
    description: 'Door sensor',
    fromZigbee: [fz.ias_contact_alarm_1, fz.battery, fz.ignore_basic_report, fz.ias_contact_alarm_1_report],
    toZigbee: [],
    exposes: [e.contact(), e.battery_low(), e.tamper(), e.battery()],
}​​​​​​​;
module.exports = definition;

But does not work too.... its funny because I bought 8 sensors of that... and just one works, but its have a signal of exclamation on icon...

Could you help me?

Thanks!

freitas666, 2021-05-23:
@AlexKay88

AlexKay88, 2021-05-24:
@freitas666 Pls. enable debug logging, remove TS0203 device, pair it again and wait for a minute. We need to see what the device communicates...
Enable debug logging:
```
advanced:
  log_level: debug
   
```
If you paste config or logs in a new comment, use "<>" icons to mark code!

freitas666, 2021-05-26:
`debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from 'Router Quarto', type 'attributeReport', cluster 'genBinaryValue', data '{"description":"24616/0x00124B00184AD290","inactiveText":"CHILD_RFD","minimumOffTime":0,"presentValue":1,"relinquishDefault":2}' from endpoint 8 with groupID 0
info MQTT publish: topic 'zigbee2mqtt/Router Quarto', payload '{"description":"24616/0x00124B00184AD290","led":false,"linkquality":118,"rssi":1,"type":"CHILD_RFD"}'
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
[... cut ...]
info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x00124b00184ad290"},"type":"pairing"}'
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0`

@AlexKay88  sorry for me delay.... this information help you?

Thank you.

AlexKay88, 2021-05-26:
To mention at first: Your device does not pair successfully. I don't know if such devices can be supported.
`error Failed to interview '0x00124b00184ad290', device has not successfully been paired`

Did you catch whole pairing from the beginning in debug log? There are no log lines like:
```
Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join'
Zigbee: allowing new devices to join.
```
Did you try to open and close door sensor several times? There is only `zoneStatus:1` and nothing like `zoneStatus:0`
`debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0`

It would also help to post `data/database.db` entry for device 0x00124b00184ad290 (one line starting with `{"id":xx,"type":"yy","ieeeAddr":"0x14b457fffe0e490b", ... ... ...`)

freitas666, 2021-05-26:
Thank you for ur reply...

Its strange for example, exactly now i went to collect information for you... and works... but works like others "with a exclamation"  and error "Failed to interview

 '0x00124b00184ad290', device has not successfully been paired" (?!?!?!?!) 

`error Failed to interview '0x00124b00184ad290', device has not successfully been paired
info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00184ad290","ieee_address":"0x00124b00184ad290","status":"failed"},"type":"device_interview"}'
info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x00124b00184ad290"},"type":"pairing"}'
info MQTT publish: topic 'homeassistant/binary_sensor/0x00124b00184ad290/contact/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00124b00184ad290"],"manufacturer":"CR Smart Home","model":"Door sensor (TS0203)","name":"0x00124b00184ad290","sw_version":"Zigbee2MQTT 1.18.3-dev"},"device_class":"door","json_attributes_topic":"zigbee2mqtt/0x00124b00184ad290","name":"0x00124b00184ad290 contact","payload_off":true,"payload_on":false,"state_topic":"zigbee2mqtt/0x00124b00184ad290","unique_id":"0x00124b00184ad290_contact_zigbee2mqtt","value_template":"{{ value_json.contact }}"}'
info MQTT publish: topic 'homeassistant/binary_sensor/0x00124b00184ad290/battery_low/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00124b00184ad290"],"manufacturer":"CR Smart Home","model":"Door sensor (TS0203)","name":"0x00124b00184ad290","sw_version":"Zigbee2MQTT 1.18.3-dev"},"device_class":"battery","json_attributes_topic":"zigbee2mqtt/0x00124b00184ad290","name":"0x00124b00184ad290 battery low","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x00124b00184ad290","unique_id":"0x00124b00184ad290_battery_low_zigbee2mqtt","value_template":"{{ value_json.battery_low }}"}'
info MQTT publish: topic 'homeassistant/binary_sensor/0x00124b00184ad290/tamper/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00124b00184ad290"],"manufacturer":"CR Smart Home","model":"Door sensor (TS0203)","name":"0x00124b00184ad290","sw_version":"Zigbee2MQTT 1.18.3-dev"},"json_attributes_topic":"zigbee2mqtt/0x00124b00184ad290","name":"0x00124b00184ad290 tamper","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x00124b00184ad290","unique_id":"0x00124b00184ad290_tamper_zigbee2mqtt","value_template":"{{ value_json.tamper }}"}'
info MQTT publish: topic 'homeassistant/sensor/0x00124b00184ad290/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00124b00184ad290"],"manufacturer":"CR Smart Home","model":"Door sensor (TS0203)","name":"0x00124b00184ad290","sw_version":"Zigbee2MQTT 1.18.3-dev"},"device_class":"battery","json_attributes_topic":"zigbee2mqtt/0x00124b00184ad290","name":"0x00124b00184ad290 battery","state_topic":"zigbee2mqtt/0x00124b00184ad290","unique_id":"0x00124b00184ad290_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
info MQTT publish: topic 'homeassistant/sensor/0x00124b00184ad290/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00124b00184ad290"],"manufacturer":"CR Smart Home","model":"Door sensor (TS0203)","name":"0x00124b00184ad290","sw_version":"Zigbee2MQTT 1.18.3-dev"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x00124b00184ad290","name":"0x00124b00184ad290 linkquality","state_topic":"zigbee2mqtt/0x00124b00184ad290","unique_id":"0x00124b00184ad290_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
debug Received MQTT message on 'homeassistant/binary_sensor/0x00124b00184ad290/contact/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00124b00184ad290"],"manufacturer":"CR Smart Home","model":"Door sensor (TS0203)","name":"0x00124b00184ad290","sw_version":"Zigbee2MQTT 1.18.3-dev"},"device_class":"door","json_attributes_topic":"zigbee2mqtt/0x00124b00184ad290","name":"0x00124b00184ad290 contact","payload_off":true,"payload_on":false,"state_topic":"zigbee2mqtt/0x00124b00184ad290","unique_id":"0x00124b00184ad290_contact_zigbee2mqtt","value_template":"{{ value_json.contact }}"}'
debug Received MQTT message on 'homeassistant/binary_sensor/0x00124b00184ad290/battery_low/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00124b00184ad290"],"manufacturer":"CR Smart Home","model":"Door sensor (TS0203)","name":"0x00124b00184ad290","sw_version":"Zigbee2MQTT 1.18.3-dev"},"device_class":"battery","json_attributes_topic":"zigbee2mqtt/0x00124b00184ad290","name":"0x00124b00184ad290 battery low","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x00124b00184ad290","unique_id":"0x00124b00184ad290_battery_low_zigbee2mqtt","value_template":"{{ value_json.battery_low }}"}'
debug Received MQTT message on 'homeassistant/sensor/0x00124b00184ad290/battery/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00124b00184ad290"],"manufacturer":"CR Smart Home","model":"Door sensor (TS0203)","name":"0x00124b00184ad29 [... cut ...] nit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'`

DB LINE:

{"id":14,"type":"EndDevice","ieeeAddr":"0x00124b00184ad290","nwkAddr":26609,"manufId":0,"modelId":"TS0203","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280],"outClusterList":[0,3,1,1280],"clusters":{"genBasic":{"attributes":{"modelId":"TS0203"}}},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1622039455486}

[image]

AlexKay88, 2021-05-27:
You mentioned "works now". What exactly works and what not?

To avoid interview failed and not successfully paired:
Have you tried pairing a few centimeters close to your Zigbee Coordinator?
If door/window contact has a wake up button, try pressing it regularly during interview/pairing.
Does it help to complete interview and successfully pair?

Befor re-pairing: Remove your own external converter for `fingerprint: [{​​​​​​​modelID: 'TS0203', manufacturerName: 'TS0203'}​​​​​​​]`, as it does not match your device. Your TS0203 is already included in Z2M.

freitas666, 2021-05-27:
Hello, when i said " works now"  its because the sensor works good and exposes show for z2m BUT I really dont understand why show this "exclamation" any way i will delete the device and paring again (without converter)...
And in the same space of coordinator i have a router (curtain zigbee) ...
i will try now and send the logs if not have success.

thank and wait :D

freitas666, 2021-05-27:
Hey friend not works:

info Device '0x00124b00184ad290' joined
info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00184ad290","ieee_address":"0x00124b00184ad290"},"type":"device_joined"}'
info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00184ad290","ieee_address":"0x00124b00184ad290","status":"started"},"type":"device_interview"}'
info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b00184ad290"},"type":"pairing"}'
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryVoltage":30}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryVoltage":30}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":120}' from endpoint 1 with groupID 0
debug Device '0x00124b00184ad290' announced itself
info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00184ad290","ieee_address":"0x00124b00184ad290"},"type":"device_announce"}'
info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00124b00184ad290"},"type":"device_announced"}'
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
debug Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0
error Failed to interview '0x00124b00184ad290', device has not successfully been paired
info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00184ad290","ieee_address":"0x00124b00184ad290","status":"failed"},"type":"device_interview"}'
info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x00124b00184ad290"},"type":"pairing"}'

Koenkk, 2021-05-27:
@AlexKay88 regarding the failed to interview, could you provide the herdsman debug log when pairing the device and get a failed to interview? 

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

freitas666, 2021-05-27:
@Koenkk I will provide the logs from herdsman logs... i had to enable it ...

Give me few minutes...

thank you

freitas666, 2021-05-27:
@Koenkk 

root@matrix:/usr/share/hassio/homeassistant/zigbee2mqtt/log# tail -f 2021-05-27.14-38-35/log.txt | grep 0x00124b00184ad290
debug 2021-05-27 14:39:13: Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}'                                                      from endpoint 1 with groupID 0
info  2021-05-27 14:39:44: Starting interview of '0x00124b00184ad290'
info  2021-05-27 14:39:44: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00184ad290","ieee_address":"                                                     0x00124b00184ad290","status":"started"},"type":"device_interview"}'
info  2021-05-27 14:39:44: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b                                                     00184ad290"},"type":"pairing"}'
debug 2021-05-27 14:39:44: Device '0x00124b00184ad290' announced itself
info  2021-05-27 14:39:44: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00184ad290","ieee_address":"                                                     0x00124b00184ad290"},"type":"device_announce"}'
info  2021-05-27 14:39:44: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00124b00184ad29                                                     0"},"type":"device_announced"}'
debug 2021-05-27 14:39:44: Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryVoltag                                                     e":30}' from endpoint 1 with groupID 0
debug 2021-05-27 14:39:44: Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercen                                                     tageRemaining":160}' from endpoint 1 with groupID 0
debug 2021-05-27 14:39:44: Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryVoltag                                                     e":30}' from endpoint 1 with groupID 0
debug 2021-05-27 14:39:44: Received Zigbee message from '0x00124b00184ad290', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercen                                                     tageRemaining":160}' from endpoint 1 with groupID 0
error 2021-05-27 14:40:05: Failed to interview '0x00124b00184ad290', device has not successfully been paired
info  2021-05-27 14:40:05: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00184ad290","ieee_address":"0x00124b00184ad290","status":"faile                  d"},"type":"device_interview"}'
info  2021-05-27 14:40:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x00124b00184ad290"},"type":"pairing"}'
`

AlexKay88, 2021-05-28:
Hi @freitas666 , never did it myself up to now, but this does not look like herdsman debug log. You should see something like this:
```
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,131,0,0,169] +15ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,4,97,8,0,2,98,26,23] +6ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,4,97,8,0,2,98,26,23] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 4 - 3 - 1 - 8 - [0,2,98,26] - 23 +0ms

```

Koenkk, 2021-05-28:
I need the complete log, starting from the point where you put the device into pairing mode until you get interview failed. Please post it on pastebin.com and link it here.

freitas666, 2021-05-28:
@Koenkk 

https://pastebin.com/sSnpgTk0

Logs since interview start till failed...

Thank you!

AlexKay88, 2021-05-29:
I was curious to dig into herdsman debug log...
Perhaps this is the interesting part at line 1464?
```
2021-05-28T20:41:48.481Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x00124b00184ad290', attempt 6
2021-05-28T20:41:48.483Z zigbee-herdsman:controller:device:log Interview - failed for device '0x00124b00184ad290' with error 'Error: Interview failed because can not get node descriptor ('0x00124b00184ad290')
    at Device.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/device.js:355:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/device.js:25:65)'
2021-05-28T20:41:48.495Z zigbee-herdsman:controller:error Interview failed for '0x00124b00184ad290 with error 'Error: Interview failed because can not get node descriptor ('0x00124b00184ad290')'
```

freitas666, 2021-05-29:
@AlexKay88 its very strange right?

Koenkk, 2021-05-29:
If the node descriptor request fails (for 6 attempts) it means the device did not connect to the network at all. Pairing it closer to the coordinator might help.

freitas666, 2021-05-29:
@Koenkk strange... because i was very close of coodinator like 50cm +-.....

Anyway i will try again, collect the logs since start interview till the end to see if we have some change...

Thank you.

freitas666, 2021-06-04:
@Koenkk :

https://pastebin.com/sSnpgTk0

AlexKay88, 2021-06-04:
Same log as 7 days ago:
`2021-05-28T20:41
Interview failed because can not get node descriptor / device has not successfully been paired`

freitas666, 2021-06-04:
@AlexKay88 hey brother....

Well... i dont know what do anymore... 

I reinstall z2m, create some converters... change usb port of coordinator... try to use anothers router in differents sides of home... and nothing work.
If you have some magic let me know <3

Koenkk, 2021-06-04:
@freitas666 can you provide a link (aliexpress?) where I can buy this device? I need to do some more investigation on this.

freitas666, 2021-07-06:
> @freitas666 can you provide a link (aliexpress?) where I can buy this device? I need to do some more investigation on this.

Sorry my delay bro...

I bought in this link:

https://a.aliexpress.com/_mqIptwz

Koenkk, 2021-07-07:
@freitas666 thanks, ordered it.

Koenkk, 2021-07-24:
Update: received my sensor but also failed to pair it. Ordered the TuYa gateway now to investigate further.

freitas666, 2021-07-24:
Ok bro! Mine works with converter but still showing the “red exclamation”
Its possible sniffer the connection between sensor and hub tuya, using a dongle with sniffer hex?

Enviado do meu iPhone

freitas666, 2021-07-27:
@Koenkk do you solve?

Koenkk, 2021-07-27:
Got my TuYa gateway today, fixed the issue.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

freitas666, 2021-07-27:
Great!!!

Thank you 🥰

Enviado do meu iPhone

freitas666, 2021-07-28:
@Koenkk yesterday i did some tests and now works without converter!
For me its ok!

Thank you for help us :) ❤️

dessone, 2021-08-01:
Mines does not work in z2mqtt edge without converter, what am i doing wrong?

elgerg, 2021-08-01:
Ditto:
[image]

Vorlex, 2021-08-01:
@dessone, @elgerg
AFAIK, you need to use Zigbee2mqtt Edge
[image]

dessone, 2021-08-01:
yes i was using it, it works the second time! thanks

elgerg, 2021-08-01:
> @dessone, @elgerg
> AFAIK, you need to use Zigbee2mqtt Edge
> [image]
> 
> edit: I have checked Release Notes for [1.21.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.21.0) and there is no info about TS0203 in New supported devices

Hi, i am not using HAssio.
I assumed it was working due to:
[image]

Is it still not supported?

elgerg, 2021-08-01:
Did i miss the dev to prod branch?

hossface, 2021-08-02:
> Did i miss the dev to prod branch?

Not working for me on the prod branch 
Thanks for all the work thou

Koenkk, 2021-08-02:
If it still fails on 1.21.0, please provide the herdsman debug log when trying to pair the device close to the coordinator.

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

adrianofox, 2021-08-02:
My four TS203 door sensors stopped working when the z2m was updated to 1.21.0. 

Everyone was like this:

[image]

Any tips?
Thanks

Koenkk, 2021-08-02:
@adrianofox added it back.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

elgerg, 2021-08-03:
I am using a CC2531 with this firmware:
CC2531_DEFAULT_20201127.zip
(I think)

Koenkk, 2021-08-03:
@elgerg how many times have you tried to pair it? Can you try it multiple times, does it always fail with this error? Looking at your logs the device immediately stops to respond which makes me clueless on how to solve it.

elgerg, 2021-08-03:
Once in this instance. I'm starting to think it might be battery? I'll get a new one tomorrow and try again....

robert2s, 2021-08-06:
All TS0203 door sensors stopped working a few days ago, possibly due to the new zigbee2mqtt software version 1.21.0. They worked flawlessly before. There are no problems with other devices. Now the support status of TS0203 is displayed as "not supported", as reported before by @adrianofox. I use a ConBee II coordinator, and I have updated Home Assistant to version 2021.8.1.

When I try to pair a TS0203 sensor again, the log shows the following:
```
2021-08-06T06:39:27.171Z zigbee-herdsman:deconz:driver query aps data indication
2021-08-06T06:39:27.171Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 119
2021-08-06T06:39:27.175Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 119 srcAddr: 0xe60f destAddr: 0x0 profile id: 0x104 cluster id: 0x500 lqi: 255
2021-08-06T06:39:27.175Z zigbee-herdsman:deconz:frameParser response payload: 9,58,0,0,0,0,23,0,0
2021-08-06T06:39:27.175Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":58,"manufacturerCode":null,"commandIdentifier":0},"Payload":{"zonestatus":0,"extendedstatus":0},"Command":{"ID":0,"parameters":[{"name":"zonestatus","type":33},{"name":"extendedstatus","type":32}],"name":"statusChangeNotification"}},"address":58895,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2021-08-06 08:39:27: Received Zigbee message from 'Dakraam', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":0}' from endpoint 1 with groupID null
2021-08-06T06:39:27.176Z zigbee-herdsman:controller:endpoint DefaultResponse 0x847127fffe11ed9c/1 1280(0, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-08-06T06:39:27.176Z zigbee-herdsman:deconz:adapter zclFrame.payload:
2021-08-06T06:39:27.176Z zigbee-herdsman:deconz:adapter { cmdId: 0, statusCode: 0 }
2021-08-06T06:39:27.176Z zigbee-herdsman:deconz:adapter no response expected
2021-08-06T06:39:27.176Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
Zigbee2MQTT:warn  2021-08-06 08:39:27: Received message from unsupported device with Zigbee model 'TS0203' and manufacturer name '_TYZB01_ncdapbwy'
Zigbee2MQTT:warn  2021-08-06 08:39:27: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
2021-08-06T06:39:27.178Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xe60f EP:1 SeqNr. 120 request id: 164
2021-08-06T06:39:27.181Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 164 status: 0
2021-08-06T06:39:27.181Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-08-06T06:39:28.182Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2021-08-06T06:39:28.182Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 
```

magicmatt007, 2021-08-06:
Same problem for me: My TS0203 sensor stopped working. Now showing as "not supported". Hope that the fixes discussed here will solve the problem soon :-)

Koenkk, 2021-08-06:
TS0203  showing as unsupported is already fixed in the dev branch. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

elgerg, 2021-08-06:
Hi @Koenkk,

Please find attached a log with 2x attempts to join.
[Door sensor 2 attempts.txt](https://github.com/Koenkk/zigbee2mqtt/files/6946796/Door.sensor.2.attempts.txt)

Regards

robert2s, 2021-08-08:
Is fixed in dev version (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html). I had the same problem, and all TS0203 reappeared instantly today (2021-8-8) after I switched to add-on zigbee2mqtt **Edge** in Home Assistant. This is what I did:

1. Disable add-on zigbee2mqtt: In the Information page of zigbee2mqtt in Home Assistant, stop the add-on, and put the "start on boot" (Dutch: start bij opstarten) to Off.
2. Install add-on zigbee2mqtt **Edge**: in the add-on Store of Home Assistant, select zigbee2mqtt **Edge**, and install it.
3. Copy the configuration from zigbee2mqtt to zigbee2mqtt **Edge**:  in the Configuration page of zigbee2mqtt, copy the complete yaml text (ctrl-C). Go to the Configuration page of zigbee2mqtt **Edge**, and paste it there.
4. Start add-on zigbee2mqtt **Edge**, and switch on "start on boot" and "show in sidebar".

As soon as a new version of zigbee2mqtt is available (i.e. version 1.22.0) I recommend to reverse the above procedure, so you use the normal zigbee2mqtt add-on again.
Note: alternatively, you may decide to wait until version 1.22.0 is availble, because that will automatically solve the problems.

Koenkk, 2021-08-08:
@elgerg I've further improved the interview procedure. First attempt will still fail but it should succeed after the second attempt now. 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html). If it does not work please provide the logs again.

seygi, 2021-08-08:
I'm still with the same problem in interview step

BCTANBC, 2021-08-11:
> Is fixed in dev version (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html). I had the same problem, and all TS0203 reappeared instantly today (2021-8-8) after I switched to add-on [...]

Thanks. This is working for me. At least for now. But not really sure when is the best timing to switch back the"non-edge" version?

Koenkk, 2021-08-11:
@BCTANBC 

>    But not really sure when is the best timing to switch back the"non-edge" version?

With the next release (on 1 September)

seygi, 2021-09-03:
After update from release on 1 sept, i'm still with problem with this device, the error occurs on pairing step, the error show unsupported device

Koenkk, 2021-09-04:
Please provide the herdsman debug log when pairing the device

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

bonjour81, 2021-09-27:
Hi, 
In the middle of the discussion, @elgerg mentionned ZM-CG205...is this sensor supposed to be same as TS0203 ? (I'm a bit confused as the discussion started with TS0203.

closed 2021-11-04 by github-actions[bot]