Koenkk/zigbee2mqtt #7555: Zigbee2MQTT:error 2021-05-22 10:08:08: Error: AREQ - ZDO - stateChangeInd after 60000ms

hyzgit, 2021-05-22:
,Using '/app/data' as data directory
,Zigbee2MQTT:info  2021-05-22 10:07:06: Logging to console and directory: '/app/data/log/2021-05-22.10-07-04' filename: log.txt
,Zigbee2MQTT:info  2021-05-22 10:07:06: Starting Zigbee2MQTT version 1.18.3 (commit #f2e39af)
,Zigbee2MQTT:info  2021-05-22 10:07:06: Starting zigbee-herdsman (0.13.92)
,Zigbee2MQTT:error 2021-05-22 10:08:08: Error while starting zigbee-herdsman
,Zigbee2MQTT:error 2021-05-22 10:08:08: Failed to start zigbee
,Zigbee2MQTT:error 2021-05-22 10:08:08: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
,Zigbee2MQTT:error 2021-05-22 10:08:08: Exiting...
,Zigbee2MQTT:error 2021-05-22 10:08:08: Error: AREQ - ZDO - stateChangeInd after 60000ms
,    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
,    at listOnTimeout (internal/timers.js:554:17)
,    at processTimers (internal/timers.js:497:7)

rebelint, 2021-05-27:
I am having the same problem. Did you manage to solve it?

hyzgit, 2021-06-04:
> I am having the same problem. Did you manage to solve it?

no

malosaa, 2021-06-18:
I have the same problem now.. @Koenkk  you know why this is happening?

It worked for months in a row and today it stopped and get the same message as above..
Restarted server(windows10) inserted the stick in other ports.. nothing helps still this message.

Koenkk, 2021-06-18:
@malosaa  try to: reflash stick -> update z2m to 1.19.1 -> start again.

malosaa, 2021-07-09:
that did solve the issue for now, im on last version now.

closed 2021-08-17 by github-actions[bot]