Koenkk/zigbee2mqtt #7556: Gledopto GL-C-008-2ID fails to turn on/off after first action

magnuslarsen, 2021-05-22:
## What happened
Upon toggling (or just setting the state to on/off), on my [Gledopto GL-C-008-2ID](https://www.zigbee2mqtt.io/devices/GL-C-008-2ID.html#gledopto-gl-c-008-2id), zigbee2mqtt immediately "toggles" it off again.  
Doing the same action again (toggling x2, setting state to on x2, ..) registers it probably; but only after the second time. This goes for both toggle / setting directly to on and off.

This happens ONLY to the CCT part (RGB works fine):
[image]
(this also happens in the zigbee2mqtt webui)
 
Here is debug logs for this device, turning it (CCT) on twice, then turning it off twice: https://pastebin.com/9dmCGexn

## What did you expect to happen
When turning on/off or toggling the light, I expect it to behave probably after the first click :-)

## How to reproduce it (minimal and precise)
Have the device paired, and turn on/off the light. After the first action, the physical light turn on, but Zigbee2mqtt / Home Assistant still sees it as powered off.   
Toggling again does nothing to the physical lights, but Zigbee2mqtt / Home Assistant now sees it as on again.

## Debug info
Zigbee2MQTT version:  1.18.3 commit: f2e39af (using the latest docker tag)
Adapter hardware: CC2652R (zzh! stick)
Coordinator: zStack3x0
Coordinator revision: 20210120

This works in Zigbee2MQTT version 1.18.2 commit: abd8a09

Koenkk, 2021-05-22:
I think this bug has already been fixed in the z2m dev branch.  (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

magnuslarsen, 2021-05-22:
I just tried 1.18.3-dev commit: 792883c - and the error persist there as well, unfortunately

Koenkk, 2021-05-22:
Try setting the color via the color picker once.

magnuslarsen, 2021-05-22:
Huh - yeah that did it! Setting the color on the RGB made everything work :-) Thanks!

closed 2021-05-22 by magnuslarsen