Koenkk/zigbee2mqtt #7561: Cannot get TuYa compatible ZG-4CH12V working

dvandonkelaar, 2021-05-22:
## Information about the device + link
Hi,

A few days ago, I received a [ZG-4CH12V zigbee relay](https://aliexpress.com/item/1005002530124474.html?spm=a2g0s.9042311.0.0.f5394c4dDFIIBs) from Ali Express.
The device is recognized by Z2M as a TS0004. I followed the [how to support new tuya devices](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_tuya_devices.html) guide and created an external converter for this device. So far, so good.

As an external converter, I created the external converter based on the [supplied example](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_tuya_devices.html#2-adding-your-device).
After starting the pairing, I get te following log. 
```
Zigbee2MQTT:info  2021-05-22 15:57:07: Device '0xcc86ecfffe5b927c' joined
Zigbee2MQTT:info  2021-05-22 15:57:07: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xcc86ecfffe5b927c","ieee_address":"0xcc86ecfffe5b927c"},"type":"device_joined"}'
Zigbee2MQTT:info  2021-05-22 15:57:07: Starting interview of '0xcc86ecfffe5b927c'
Zigbee2MQTT:info  2021-05-22 15:57:07: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xcc86ecfffe5b927c","ieee_address":"0xcc86ecfffe5b927c","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2021-05-22 15:57:07: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xcc86ecfffe5b927c"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-05-22 15:57:07: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xcc86ecfffe5b927c"},"type":"pairing"}'
Zigbee2MQTT:debug 2021-05-22 15:57:07: Device '0xcc86ecfffe5b927c' announced itself
Zigbee2MQTT:info  2021-05-22 15:57:07: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xcc86ecfffe5b927c","ieee_address":"0xcc86ecfffe5b927c"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-05-22 15:57:07: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xcc86ecfffe5b927c"},"type":"device_announced"}'
Zigbee2MQTT:debug 2021-05-22 15:57:07: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS0004"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-22 15:57:07: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZ3000_hdlpifbk"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-22 15:57:07: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-22 15:57:08: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-22 15:57:08: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"appVersion":71}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-22 15:57:08: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-22 15:57:08: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-22 15:57:08: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-22 15:57:08: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-05-22 15:57:08: Successfully interviewed '0xcc86ecfffe5b927c', device has successfully been paired
Zigbee2MQTT:info  2021-05-22 15:57:08: Device '0xcc86ecfffe5b927c' is supported, identified as: TuYa Zigbee 4CH relay switch (ZG-4CH12V)
Zigbee2MQTT:info  2021-05-22 15:57:08: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Zigbee 4CH relay switch","exposes":[{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"ZG-4CH12V","supports_ota":false,"vendor":"TuYa"},"friendly_name":"0xcc86ecfffe5b927c","ieee_address":"0xcc86ecfffe5b927c","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:info  2021-05-22 15:57:08: Configuring '0xcc86ecfffe5b927c'
Zigbee2MQTT:error 2021-05-22 15:57:09: Failed to configure '0xcc86ecfffe5b927c', attempt 1 (ReferenceError: bind is not defined
    at Object.configure (C:\Users\Dirk\Desktop\zigbee2mqtt\lib\util\externally-loaded.js:29:9)
    at Configure.configure (C:\Users\Dirk\Desktop\zigbee2mqtt\lib\extension\configure.js:137:45)
    at Configure.onZigbeeEvent (C:\Users\Dirk\Desktop\zigbee2mqtt\lib\extension\configure.js:119:18)
    at Controller.callExtensionMethod (C:\Users\Dirk\Desktop\zigbee2mqtt\lib\controller.js:384:44))
```
The device is paired, but the device could not be configured and gives me the ```ReferenceError: bind is not defined```.
A Z2M restart also triggers this error.

As stated in the guide, a ```data/tuya.dump.txt``` file should be created. This file would not show anywhere in my Z2M (data) directory (this was tested on a Windows 10 laptop and docker image, both with the latest version from dev).

**Question 1**: how can I get the ```data/tuya.dump.txt``` file to be created?

After some fiddling with the external converter, I got some logging when I push some switches when pairing. Below these logs entries. 
```
Zigbee2MQTT:info  2021-05-22 16:32:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Zigbee 4CH relay switch","friendly_name":"0xcc86ecfffe5b927c","model":"ZG-4CH12V","supported":true,"vendor":"TuYa"},"type":"pairing"}'
Zigbee2MQTT:debug 2021-05-22 16:32:42: Received Zigbee message from '0xcc86ecfffe5b927c', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-22 16:32:42: No converter available for 'ZG-4CH12V' with cluster 'genOnOff' and type 'attributeReport' and data '{"onOff":1}'
Zigbee2MQTT:debug 2021-05-22 16:32:45: Received Zigbee message from '0xcc86ecfffe5b927c', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-22 16:32:45: No converter available for 'ZG-4CH12V' with cluster 'genOnOff' and type 'attributeReport' and data '{"onOff":0}'
Zigbee2MQTT:debug 2021-05-22 16:36:01: Received MQTT message on 'zigbee2mqtt/bridge/request/restart' with data '{"transaction":"xz5jz-1"}'
```

**Question 2**: how can I figure out how to configure the external converter?

## data/database.db entry of the device
```
{
    "id": 5,
    "type": "Router",
    "ieeeAddr": "0xcc86ecfffe5b927c",
    "nwkAddr": 64193,
    "manufId": 4098,
    "manufName": "_TZ3000_hdlpifbk",
    "powerSource": "Mains (single phase)",
    "modelId": "TS0004",
    "epList": [1, 2, 242, 3, 4],
    "endpoints": {
        "1": {
[... 53 lines cut ...]
            "configuredReportings": [],
            "meta": {}
        },
        "242": {
            "profId": 41440,
            "epId": 242,
            "devId": 97,
            "inClusterList": [],
            "outClusterList": [33],
            "clusters": {},
            "binds": [],
            "configuredReportings": [],
            "meta": {}
        }
    },
    "appVersion": 71,
    "stackVersion": 0,
    "hwVersion": 1,
    "dateCode": "",
    "zclVersion": 3,
    "interviewCompleted": true,
    "meta": {},
    "lastSeen": null
}
```

Koenkk, 2021-05-22:
The `TS0004` is already supported and I would expect that this device works out-of-the-box. (https://github.com/Koenkk/zigbee-herdsman-converters/blob/d0fb06c2429171f327950484ea3dec80864637cc/devices/tuya.js#L827)

dvandonkelaar, 2021-05-22:
Thanks for the reply.
If I try to pair without the custom converter, I also get an error.
The log entries below are from the pairing process.
```
Zigbee2MQTT:info  2021-05-22 20:03:42: Device '0xcc86ecfffe5b927c' joined
Zigbee2MQTT:info  2021-05-22 20:03:42: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xcc86ecfffe5b927c","ieee_address":"0xcc86ecfffe5b927c"},"type":"device_joined"}'
Zigbee2MQTT:info  2021-05-22 20:03:42: Starting interview of '0xcc86ecfffe5b927c'
Zigbee2MQTT:info  2021-05-22 20:03:42: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xcc86ecfffe5b927c","ieee_address":"0xcc86ecfffe5b927c","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2021-05-22 20:03:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xcc86ecfffe5b927c"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-05-22 20:03:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xcc86ecfffe5b927c"},"type":"pairing"}'
Zigbee2MQTT:debug 2021-05-22 20:03:42: Device '0xcc86ecfffe5b927c' announced itself
Zigbee2MQTT:info  2021-05-22 20:03:42: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xcc86ecfffe5b927c","ieee_address":"0xcc86ecfffe5b927c"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-05-22 20:03:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xcc86ecfffe5b927c"},"type":"device_announced"}'
Zigbee2MQTT:debug 2021-05-22 20:03:42: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS0004"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-22 20:03:43: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZ3000_hdlpifbk"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-22 20:03:43: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-22 20:03:43: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-22 20:03:43: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"appVersion":71}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-22 20:03:43: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-22 20:03:43: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-22 20:03:43: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-22 20:03:43: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-05-22 20:03:44: Successfully interviewed '0xcc86ecfffe5b927c', device has successfully been paired
Zigbee2MQTT:info  2021-05-22 20:03:44: Device '0xcc86ecfffe5b927c' is supported, identified as: TuYa Smart light switch - 4 gang with neutral wire (TS0004)
Zigbee2MQTT:info  2021-05-22 20:03:44: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart light switch - 4 gang with neutral wire","exposes":[{"endpoint":"l1","features":[{"access":7,"description":"On/off state of the switch","endpoint":"l1","name":"state","property":"state_l1","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"l2","features":[{"access":7,"description":"On/off state of the switch","endpoint":"l2","name":"state","property":"state_l2","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"l3","features":[{"access":7,"description":"On/off state of the switch","endpoint":"l3","name":"state","property":"state_l3","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"l4","features":[{"access":7,"description":"On/off state of the switch","endpoint":"l4","name":"state","property":"state_l4","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0004","supports_ota":false,"vendor":"TuYa"},"friendly_name":"0xcc86ecfffe5b927c","ieee_address":"0xcc86ecfffe5b927c","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:info  2021-05-22 20:03:44: Configuring '0xcc86ecfffe5b927c'
Zigbee2MQTT:info  2021-05-22 20:03:44: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Smart light switch - 4 gang with neutral wire","friendly_name":"0xcc86ecfffe5b927c","model":"TS0004","supported":true,"vendor":"TuYa"},"type":"pairing"}'
[... cut ...]
Zigbee2MQTT:info  2021-05-22 20:09:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0xcc86ecfffe5b927c' failed: 'Error: Command 0xcc86ecfffe5b927c/1 genOnOff.on({}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 54065 - 1 - 52 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"0xcc86ecfffe5b927c"},"type":"zigbee_publish_error"}'
```
If the buttons are triggered on the device itself, Z2M does not log it.

Koenkk, 2021-05-23:
I don't think you need an external converter for this device to work since I don't think there is anything wrong with the existing converter.

Can you try to pair the device close to the coordinator and see if it works? If not please provide a sniff when controlling this device: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

dvandonkelaar, 2021-05-24:
Pairing the device next to the coordinator gives me the logs below (like before).
```
Zigbee2MQTT:info  2021-05-24 09:55:21: Device '0xcc86ecfffe5b927c' joined
Zigbee2MQTT:info  2021-05-24 09:55:21: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xcc86ecfffe5b927c","ieee_address":"0xcc86ecfffe5b927c"},"type":"device_joined"}'
Zigbee2MQTT:info  2021-05-24 09:55:21: Starting interview of '0xcc86ecfffe5b927c'
Zigbee2MQTT:info  2021-05-24 09:55:21: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xcc86ecfffe5b927c","ieee_address":"0xcc86ecfffe5b927c","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2021-05-24 09:55:21: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xcc86ecfffe5b927c"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-05-24 09:55:21: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xcc86ecfffe5b927c"},"type":"pairing"}'
Zigbee2MQTT:debug 2021-05-24 09:55:22: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS0004"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-24 09:55:22: Device '0xcc86ecfffe5b927c' announced itself
Zigbee2MQTT:info  2021-05-24 09:55:22: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xcc86ecfffe5b927c","ieee_address":"0xcc86ecfffe5b927c"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-05-24 09:55:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xcc86ecfffe5b927c"},"type":"device_announced"}'
Zigbee2MQTT:debug 2021-05-24 09:55:22: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZ3000_hdlpifbk"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-24 09:55:22: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-24 09:55:22: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-24 09:55:22: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"appVersion":71}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-24 09:55:22: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-24 09:55:22: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-24 09:55:22: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-24 09:55:22: Received Zigbee message from '0xcc86ecfffe5b927c', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-05-24 09:55:23: Successfully interviewed '0xcc86ecfffe5b927c', device has successfully been paired
Zigbee2MQTT:info  2021-05-24 09:55:23: Device '0xcc86ecfffe5b927c' is supported, identified as: TuYa Smart light switch - 4 gang with neutral wire (TS0004)
Zigbee2MQTT:info  2021-05-24 09:55:23: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart light switch - 4 gang with neutral wire","exposes":[{"endpoint":"l1","features":[{"access":7,"description":"On/off state of the switch","endpoint":"l1","name":"state","property":"state_l1","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"l2","features":[{"access":7,"description":"On/off state of the switch","endpoint":"l2","name":"state","property":"state_l2","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"l3","features":[{"access":7,"description":"On/off state of the switch","endpoint":"l3","name":"state","property":"state_l3","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"l4","features":[{"access":7,"description":"On/off state of the switch","endpoint":"l4","name":"state","property":"state_l4","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0004","supports_ota":false,"vendor":"TuYa"},"friendly_name":"0xcc86ecfffe5b927c","ieee_address":"0xcc86ecfffe5b927c","status":"successful","supported":true},"type":"device_interview"}'
Zigbee2MQTT:info  2021-05-24 09:55:23: Configuring '0xcc86ecfffe5b927c'
Zigbee2MQTT:info  2021-05-24 09:55:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Smart light switch - 4 gang with neutral wire","friendly_name":"0xcc86ecfffe5b927c","model":"TS0004","supported":true,"vendor":"TuYa"},"type":"pairing"}'
[... cut ...]
Zigbee2MQTT:info  2021-05-24 10:03:09: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0xcc86ecfffe5b927c' failed: 'Error: Command 0xcc86ecfffe5b927c/3 genOnOff.on({}, {\"sendWhenActive\":false,\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Timeout - 65248 - 3 - 39 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"0xcc86ecfffe5b927c"},"type":"zigbee_publish_error"}'
```

The traffic sniff logs will come later today.

dvandonkelaar, 2021-05-24:
Apparently, my CC2351 (my testing coordinator) is not working correct. I paired the device with my CC2652RB (my production coordinator) and it all works out of the box. I think I need to check my CC2351 firmware.

closed 2021-05-24 by dvandonkelaar